MichaIng/DietPi #4265: Forum Registration not possible: "The solution you provided was incorrect"

Hannlora, 2021-04-12:
Sorry if this does not belong here. I did not find another way to contact you.

#### Required Information
- URL: https://dietpi.com/phpbb/ucp.php?mode=register
- I tried Firefox & Chrome (clean setup) on Linux & MacOS on home Wifi & Mobile network
- I tried different names, passwords and email addresses

#### Steps to reproduce
1. Open URL: https://dietpi.com/phpbb/ucp.php?mode=register
2. Accept the terms
3. Enter the four data fields (name, pw, pw repeat, email)
4. Hit submit

#### Expected behaviour
- Registration completed and confirmation email sent

#### Actual behaviour
- Error message: "The solution you provided was incorrect"

ravenclaw900, 2021-04-12:
This is probably a better fit for the DietPi-Website repository, so I'll transfer it there.

Joulinar, 2021-04-12:
Hi,

We will have a look. In meantime you can post your issue on GitHub as well 😉

ravenclaw900, 2021-04-12:
This is probably a better fit for the DietPi-Website repository, but someone else will have to transfer it, as I don't have write access.

MichaIng, 2021-04-12:
Many thanks for your report. Did you send me an email yesterday? Regardless, I'll copy & paste the content:

The problem is the Google reCAPTCHAv3 "challenge" that protects phpBB registrations from bots. The problem with v3 compared to v2 is that there is no chance to pass the test by solving a puzzle or hitting the "I'm no bot" button that v2 provided. A rating is automatically generated, based on IP and some other data Google has, and if that is below a limit value, registration is blocked.

You're the 3rd (or now 4th) report of this issue already, so I'm considering to revert to reCAPTCHAv2. You should be able to raise your IPs rating by using the Google search or logging into a Google server, if you have an account, so that your IP is recognised as human user. IMO it's not great to force using Google in any way, hence my consideration for the downgrade, when the situation does not become better. Sadly reCAPTCHA is by far the best working bot protection that phpBB supports, so to add burden onto our moderators, for now it needs to stay with either v2 or v3.

Let me know if the workaround works for you, otherwise I'll temporarily disable reCAPTCHA or downgrade now.

RobLeighton67, 2021-04-12:
Hi,

I have tried the workaround - logging into my Google account whilst in Chrome (I am a heavy Google search user but via DuckDuckGo)... But no success as yet.
Ironically the issue I wanted raise on the forum has now been resolved (mosquitto.org was down to everything but pings - so couldn't APT Upgrade )

MichaIng, 2021-04-12:
Please try again now. Lol the issue was actually not reCAPTCHAv3 but a missing exception in the CSP rules 😅. So we can keep it for now until getting more actual issue reports with it.

Hannlora, 2021-04-12:
Worked like a charm, thank you all!

closed 2021-04-12 by Hannlora