MichaIng/DietPi #4266: swapfile not working on external usb drive

champymarty, 2021-04-12:
### Creating a bug report/issue

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version 10.9
- Kernel version | `uname -a`
- SBC model Linux DietPi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux

- Power supply used 5V 3A Official raspberry pi 4 power supply
- SDcard used  SanDisk ultra 64 Gb

#### Additional Information (if applicable)
- Software title swapfile in drive_manager
- Was the software title installed freshly or updated/migrated?
I just install dietpi and after I install Pihole + unbound with owncloud + certbot

#### Steps to reproduce

1. ... Go to drive_manager
2. ... Select swapfile
3.  Write 1 for automatic swapfile size

#### Expected behaviour

- it should activate the swapfile on the external USB drive

#### Actual behaviour

-  I does nothing, when I try to redo the valeu is still 0 (deactivated)

#### Extra details
[image]
[image]
After update:
[image]

Joulinar, 2021-04-12:
Hi,

option 1 for auto-choice assuring min 2048 MiB total memory. Means if your system has more than 2GB ram, no swap file will be added. However you can force creating the swap file by running 

`/boot/dietpi/func/dietpi-set_swapfile 1024 /mnt/<your_mount_point>/swap`

In this example, a 1GB swap file will be created

champymarty, 2021-04-12:
I see can I force it to a bigger swapfile size ? Because when I dowload big file via owncloud (like a 6Gb zip file for exemple) it run out of memory

Joulinar, 2021-04-12:
yes you can choose 4096 to add 4GB SWAP

champymarty, 2021-04-12:
thank you ! Do you know what is the max of SWAP ?

Joulinar, 2021-04-12:
just check your 2nd screen print you added on the initial post 😉

champymarty, 2021-04-13:
ahaahah wow thanks you sooo much :D

MichaIng, 2021-04-13:
> Do you know what is the max of SWAP ?

I'm not aware of any hard limit for swap file sizes, aside of disk space of course. At least our scripts to no limit it.

Joulinar, 2021-04-13:
At least we do a statement inside drive manager if you select swap to be created.

Joulinar, 2021-04-14:
I will go to close the topic. Feel free to reopen if needed.

closed 2021-04-14 by Joulinar