MichaIng/DietPi #4273: How to flash dietpi to emmc NanoPi NEO Plus2

gamerthundertv, 2021-04-15:
I don't know how to flash new OS dietpi to emmc NanoPi NEO Plus2?

MichaIng, 2021-04-15:
Not sure if there is some kind of connector that can be used to flash an image from an external host system directly to the internal eMMC?

Otherwise:
- Flash the DietPi image onto an SD card and boot that.
- Download and extract again the DietPi image within the booted DietPi from SD card.
- Flash it via `dd if=/path/to/DietPi.img of=/dev/mmcblk1 bs=1M`, expecting the SD card device to be `mmcblk0` and the eMMC `mmcblk1`, which should be verified.

Installer images are in planning, which can be flashed to USB drive or SD card to install DietPi to an internal drive. But no ETA possible.

StephanStS, 2021-04-24:
Take into account that the size of the SD card used should be smaller than the eMMC size (8 GB), i.e. at best use a 4 GB SD Card to be sure. Otherwise the copy with dd may run into problems.

closed 2021-05-07 by MichaIng