MichaIng/DietPi #4274: dietpi-software

BilSheckler, 2021-04-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.0.2
- Distro version | 
- Kernel version |
- SBC model | RPi4B
- Power supply used | CANApower (EG: 5.1V 2.5A)
- SDcard used | (EG: SanDisk ultra 32 GB)

#### Additional Information (if applicable)
- Software title |never got that far...
- Was the software title installed freshly or updated/migrated? Fresh
- Can this issue be replicated on a fresh installation of DietPi? Yes
 Not sure how
- Bug report ID 

#### Steps to reproduce

1. ...fresh hardware, fresh software. After the first boot, I was having trouble with log-in as the keyboard mapping was all screwy. Figured that out and ssh'd in and fixed the keyboard mapping. Ran DietPi-Launcher and DietPi-Config and got everything configured and loaded optimally (I think).
2. ran DietPi-Software command. Dietpi replied that it was initializing database, then reading, then it crashed, pretty hard.

#### Expected behaviour

#### Actual behaviour

Very repeatable, local or SSH.

I have some PC experience, and a little linux experience, but this is the first time that I have used DietPi.

When this crashes the pi, it's locked-up tight enough that I can not use the mouse to highlight the fault text and copy it to a clipboard to paste it to a bug report, or at least I have not figured out how.

Hopefully it's easier than the keyboard, that one really had me scratching my head.

Thank You for your help,

Bil Sheckler

Joulinar, 2021-04-15:
Hi,

If you use SSH, you should be able to copy/past the error from SSH client. Or just do a picture with your mobile phone. Same for local connection. I guess you have a screen attached and could do a picture as well.

MichaIng, 2021-04-15:
Since pinging the publicly available IP address 9.9.9.9 fails, the system seems to have internet connectivity issues. When connected via SSH and PuTTY, you can easily select the error message in PuTTY (which stores all selected text in paste buffer automatically) and past it here on GitHub 🙂.

Can you try it from console manually (allowing some more retries)?
```
ping -c 5 9.9.9.9
```
It's a bit strange since SSH works, hence local network connectivity is fine. Does pinging other known hosts work, e.g.
```
ping -c 5 dietpi.com
```

BilSheckler, 2021-04-16:
[image]

Michalng,  I enabled logging SSH packet data in PuTTY, but not until after the screenshot above, which was from trying to send a bug report.

Manually pinging -c 5 9.9.9.9 locally is successful with no data loss in 10 ms.

Manually pinging -c 5 dietpi.com locally is successful with no data loss in 11 ms.

I was also surprised that that it was apparently failing to connect as it was fresh and I had just run dietpi-launcher and dietpi-config, and I thought that those used the internet...

Thank You again for your help,

Bil

Joulinar, 2021-04-16:
You can simply mark the content in PuTTY and copy/past the lines into GitHub.

Interesting, the bug report could be uploaded but ping to Quad9 is not working?

BilSheckler, 2021-04-16:
Michalng, I used Etcher to burn the same image to a Samsung EVO Plus 32 GB SD card and it work fine.

I used Etcher to burn the Sandisk Ultra card too...

Moving on.  Closed.

Thank You for the great work that you do !

closed 2021-04-16 by BilSheckler

MichaIng, 2021-04-16:
Probably some temporary connectivity issue. Indeed strange as also the MOTD in the banner got downloaded successfully etc.