MichaIng/DietPi #4275: External USB drive slow write RPi4 4GB

bbsixzz, 2021-04-15:
DietPi v7.0.2, latest apt updates as of today.

I've noticed rTorrent putting high load on the machine last couple of weeks even though CPU usage is shown as low, I've been doing fine for years and could easily achieve 30 MiB writes on the external WD Elements 4 TB USB 3.0 (purchased November 2018).

Benchmark reads fine 100 MiB/s but writes only ~10-15 MiB/s, drive is 57% full.

I might have filled and emptied the drive a couple times during the years but nothing extreme.

Are there any commands I can run to check the health of the drive? Is the drive just old?

Please help, I'm out of ideas!

MichaIng, 2021-04-15:
Many thanks for your report.

Run `dmesg -l emerg,alert,crit,err` to check for kernel errors.

And you can use `dietpi-drive_manager` (or `fsck` after `umount`) to check the file system for errors.

There is a defragmentation tool for ext4 file systems as well: `e4defrag /mnt/<mount_point>`.

You did a raw file system benchmark, e.g. via `dietpi-drive_manager`? Not sure if decreasing write speed is a typical sign of degrading/ageing drives.

bbsixzz, 2021-04-15:
Thanks!

I first ran `e4defrag -c /mnt/server`

```
Total/best extents                             114438/24229
 Average size per extent                        24490 KB
 Fragmentation score                            0
 [0-30 no problem: 31-55 a little bit fragmented: 56- needs defrag]
 This directory (/mnt/server) does not need defragmentation.
 Done.
```

Then `dmesg -l emerg,alert,crit,err`

```
[    0.553956] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[    2.492632] sd 0:0:0:0: [sda] No Caching mode page found
[    2.492642] sd 0:0:0:0: [sda] Assuming drive cache: write through
[    2.578036] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[    2.578055] [vc_sm_connected_init]: failed to initialize shared memory service
[    2.586976] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
```

I also did a repair with drive_manager but nothing significant came up.

MichaIng, 2021-04-16:
Which I/O scheduler is active on that drive?
```
cat /sys/block/sda/queue/scheduler
```
By default it's `mq-deadline`, AFAIK, so you could test `none`.

Also is UAS active? Usually it should enhance USB write speeds but causes issues with some drives:
```
lsmod | grep uas
dmesg | grep uas
```
In case it could be tried to blacklist it:
```
echo 'blacklist uas' > /etc/modprobe.d/disable-uas.conf
```

Since no obvious errors shows up, it's just trying out a few things which can affect performance, but I have no real idea, sadly. What write speed did you have in the past with that drive, actually?

bbsixzz, 2021-04-16:
Well I have a cloned machine at my parents with exact same specs and HDD:

[image]
[image]

I tried no scheduler with no effect, same with blacklist.

I don't have another drive to test unfortunately, maybe I can find one in the trash downstairs later…

MichaIng, 2021-04-16:
> Well I have a cloned machine at my parents

Cloned really means same SD card content?

> same with blacklist

It requires a reboot to be effective btw.

> maybe I can find one in the trash downstairs later…

😄

One last idea I have is updating the EEPROM manually: `rpi-eeprom-update -u`
It should be done automatically at boot, but who knows.

bbsixzz, 2021-04-16:
Not cloned software although 99% same settings.

I tried blacklist now after a reboot, problem persists and results seem very uneven.

[image]

EEPROM is up to date on both machines, same version.

This is what both machines are running:

```
[  OK  ] DietPi-Services | start : proftpd
[  OK  ] DietPi-Services | start : php7.3-fpm
[  OK  ] DietPi-Services | start : lighttpd
[  OK  ] DietPi-Services | start : rtorrent
[  OK  ] DietPi-Services | start : sonarr
[  OK  ] DietPi-Services | start : cron
```

MichaIng, 2021-04-16:
A larger test file would probably make sense on a HDD.

The other applications should not play a role since services are stopped while the benchmark is running.

Otherwise, I'm also out of ideas currently. You could check the Raspberry Pi forum or GitHub repository, if you find similar reports or ideas how to further debug it. But probably it's really the drive itself which has some mechanical ageing that makes it spin slower or slows the write head (since read speeds seem okay).

bbsixzz, 2021-04-16:
The fast drive is almost empty and the slow drive is 57% full (with lots of data removed in the last month when it was approaching 95%).

Working theory is that the fuller a spinning drive is the slower it gets as the writing arm approaches the inner rings. 

Cheers

MichaIng, 2021-04-17:
> Working theory is that the fuller a spinning drive is the slower it gets as the writing arm approaches the inner rings.

Ah, true, although I wouldn't expect the 57% to be physical laid out strictly in the inner area of the disk, especially since there are multiple disks in one drive, usually. So in my experience, the effect is rarely that visible, and, it should effect the read benchmark as well, as the same data is read that was written before, hence on the same physical position of the disk 🤔. But could be tested by moving some data to a backup disk, of formatting it freshly 🙂.

bbsixzz, 2021-04-17:
I'll get back to this once I have a second drive, no luck with the trash :[

bbsixzz, 2021-04-19:
I'm checking drive health using smartmontools after finding this thread https://dietpi.com/phpbb/viewtopic.php?t=6025

I passed the 2 min test with no errors and I'm now running the 688 min test to see if anything shows up.

bbsixzz, 2021-04-19:
I did some further testing with iotop, nothing is writing to the disk unless I'm downloading something with rTorrent so that rules out any extra load. I set the download speed to 4 MiB/s just to keep the load under 1 but after a while it slowly trickles up and I'm back to 3-4 which is not ideal if I want to stream some media from the server at the same time.

Two years seems way too soon for a drive to fail like this, I got unlucky I guess.

MichaIng, 2021-04-20:
Good idea with `smartctl`, I forgot about that. This means that electrical and mechanical properties seem to be fine and every bit of the drive can be read successfully. AFAIK it does not include any test that involves writing to the disk.

> Two years seems way too soon for a drive to fail like this

Close to usual warranty times. Probably the manufacturer support can suggest another test or provide a replacement. I still would test a clean format first, although the tests done basically rule out any filesystem-related issue.

Re-creating the partition table works fine?
```
sfdisk --no-reread --no-tell-kernel -fN1 /dev/sda <<< ',+'
```
This maximises the size of the first partition of the /dev/sda drive. Should have no effect, but it rewrites the partition table. Although if we'd assume any issue here, it's probably dangerous to do that, as without partition table, the file system cannot be accessed either 🤔.

MichaIng, 2021-05-07:
I mark this as closed. Feel free to reopen if required.

closed 2021-05-07 by MichaIng

bbsixzz, 2021-05-23:
So I connected another hard drive (320GB, formatted ext4) and received the same abysmal speeds.

When formatted to APFS and connected to a mac it went fast according to USB3 speeds.

I poked around on the fast machine and it looks like this:

[image][image]

The slow machine looks like this:

[image][image]

The only difference I could notice was Filesystem autofs ext4 on the fast machine vs. ext4 on the slow one.

I don't even know why it's different, they were setup almost exactly the same; someone please educate me.

Is there a way to change to autofs ext4 without wiping the drive?

MichaIng, 2021-05-23:
That "autofs" is shown, is a bug in dietpi-drive_manager. Run the following two commands to have it fixed:
```
sed -i '/findmnt -no FSTYPE/c\\t\t\t\taDRIVE_FSTYPE[$index]=$(blkid -s TYPE -o value -c /dev/null "${aDRIVE_MOUNT_SOURCE[$index]}")' /boot/dietpi/dietpi-drive_manager
sed -Ei 's/findmnt -r?no /findmnt -Ufnro /' /boot/dietpi/dietpi-drive_manager
```

When having a drive added in `/etc/fstab` with `x-systemd.automount` (default for external drives), `findmnt -no FSTYPE` will show two lines for it, one for the actual drive with the actual filesystem and one for the `x-systemd.automount` (autofs) wrapper. But "autofs" is of course no actual filesystem. It is visual only in the first place, but finally "autofs" will be stored in `/etc/fstab` as well which makes the mount fail on next boot. To get the actual correct filesystem type we changed the commands to skip those wrapper lines and only print a single line for the actual filesystem, which is required to store and use to be able to mount it.

However, then the drive is mounted, it does not affect R/W speed. It can only be mounted with the correct type, `ext4` in your case.

Is it intended that there is a second FAT partition on that drive?

bbsixzz, 2021-05-23:
Ok thx, the second FAT partition is not intended I didn't dare to touch it however I just noticed it.

bbsixzz, 2021-06-22:
I have purchased a secondary drive of the same make and model. 

What would be the best way to clone the existing drive using macOS so that I can just shut the Pi down, replace the drive and start back up again with no headaches?

The filesystem is EXF4.

I can't clone on the Pi because it can't power both drives at the same time using the 3 amp official charger.

MichaIng, 2021-06-22:
The macOS native disk utility seems to be able to do this, using the "Restore" feature to "restore" from the current RPi drive to the backup drive: https://www.lifewire.com/use-disk-utility-to-clone-macs-drive-4042367

I didn't test it, but method looks good. Alternatively, AFAIK `dd` is as well available on macOS, so from the console:
```sh
dd if=/dev/rpi of=/dev/backup bs=1M conv=sparse
```
replacing "rpi" and "backup" with the actual drive identifiers (the drives, not the partitions!) should work. `conv=sparse` is to try skipping NUL blocks (of min 1 MiB size as of `bs=1M`) instead of writing them to the target drive, so we try to save some time and disk writes here. `zerofree` or `fstrim` could be additionally used to overwrite empty space with NUL, which makes the sparse option more effective, but performs the disk writes on the source drive instead, which is worse when this is an SD card while the target is an HDD or SSD (in terms of wear). But `dd` will most likely still be slower than a backup/restore/clone utility that is meant to clone drives 😉.

bbsixzz, 2021-06-22:
Thanks for the quick reply, I will get back to you with the results.

bbsixzz, 2021-06-25:
Ok so Disk Utility doesn't seem to like EXF4 (it can't even mount it).

[image]

Joulinar, 2021-06-25:
MacOS do not support ext4

bbsixzz, 2021-06-25:
Yeah I'm installing Ubuntu on VirtualBox and then going to try Clonezilla; I'll report the results.

MichaIng, 2021-06-25:
Ah, the disk utility then seems to not recover the raw disk, but tries to read/write the partition table and filesystems. `dd`, if present, should work, as it copies the raw disk content byte by byte and doesn't care about partition table and filesystems. But Clonezilla might be the more efficient way, good idea.

bbsixzz, 2021-06-26:
Alright, the journey is finally over! 

[image]

What I don't understand is this benchmark in Ubuntu inside VirtualBox using the old slow drive:

[image]

The performance is much better, DietPi choked on 4 MB/s using ruTorrent with the load passing 4.

Doesn't it look fishy to you? 

Could there be some inefficiencies in how DietPi handles external USB HDDs?

I'm going to reformat the old drive and run the benchmarks again in DietPi to see what happens.

I'll be back.

MichaIng, 2021-06-26:
> Could there be some inefficiencies in how DietPi handles external USB HDDs?

DietPi doesn't handle USB HDDs, it's all the official kernel, firmware, drivers etc, exactly like on Raspberry Pi OS, so both should behave the same. Ubuntu uses a different kernel and drivers, so if it turns out to be a Raspberry Pi OS / kernel issue, then we could check back with the RPi developers.

But you cannot compare ruTorrent with a benchmark, of course. So if you want to compare this, you'd need to run ruTorrent on Ubuntu as well 😉.

bbsixzz, 2021-06-27:
Yes thanks for guiding me through all of this, after formatting the old drive and doing more tests the performance was still abysmal so I conclude that there is nothing strange going on here other than a moribund drive taking its last breaths.

I'm just glad I managed to move all my data over before everything went haywire.

Cheers

GvY85, 2021-08-07:
I have had slow performance and 'lag' in the system for a while on my x86_64 Intel Atom mini pc with external 2.5inch mechanical HDD's. 
qBittorrent would choke wit a few downloads open and I would get interupted streams when streaming with MiniDLNA to my TV. 
Did some reading and found out that mq-deadline may not be the best default scheduler for mechanical drives. So I changed it to BFQ and now the disk can handle IO much better. Also, I changed the read ahead buffer a bit and that solved streaming hickups. 
@MichaIng perhaps something to integrate as an option in DietPi when installing an external drive or when installing NAS/Dowload software?

Improving IO perfomance on external mechanical drive:
`sudo modprobe bfq` if theis show up no error bfq is supported
`sudo cat /sys/block/sda/queue/scheduler` in my case bfq was not listen
`sudo echo bfq > /etc/modules-load.d/bfq.con`f enable bfq support
`dietpi-drive_manager `-> select scheduler bfq for the drive
`sudo cat /sys/block/sda/queue/scheduler` should now show `[bfq]`

Improving secuential reads (f.i when streaming video)
`cat /sys/block/sda/queue/read_ahead_kb` shows the amount of read ahead data, in my case it was 256kB
`echo 32768 > /sys/block/sda/queue/read_ahead_kb` to change it to 32MB solved all my streaming problems

MichaIng, 2021-08-11:
Good to know that `bfq` may be hidden behind a kernel module. Btw, once applied it via `dietpi-drive_manager`, you don't need `/etc/modules-load.d/bfq.conf` anymore. The module is loaded automatically when the scheduler is applied, via `echo bfq > /sys/block/sda/queue/scheduler` or via udev rules. But since it is not shown in the first place, I'll add a `modprobe bfq` to the I/O scheduler menu on x86_64 systems.

Raising the read ahead data may stabilise streaming indeed, but it breaks performance of small random access, e.g. databases. There are many ways to tune I/O for a specific use case, but the default should be okay for most cases. I'm not sure if it makes sense to add this one single tunable to a DietPi script while not adding all the others. To me this is something to manually tune, as you did. We may create a documentation page about I/O scheduling and tuning in general, and/or about how to permanently apply such settings via udev rules.