MichaIng/DietPi #4277: Nanopi Neo serial not working

redex-2, 2021-04-16:
#### Required Information
- DietPi version 
    G_DIETPI_VERSION_CORE=7
    G_DIETPI_VERSION_SUB=0
    G_DIETPI_VERSION_RC=2
    G_GITBRANCH='master'
    G_GITOWNER='MichaIng'

- Distro version | 10.9
- Kernel version | Linux DietPi 5.10.21-sunxi #21.02.3 SMP Mon Mar 8 00:28:04 UTC 2021 armv7l GNU/Linux
- SBC model | NanoPi NEO (armv7l)
- Power supply used | 5V 2A USB
- SDcard used | n/a

#### Steps to reproduce
  c lang:
  open("/dev/ttyS0", O_RDWR);
  tcgetattr(fd_serial, &tty);

#### Expected behaviour
  correct tcgetattr for ttySX

#### Actual behaviour
  In errno Input/output error

#### Extra details
  Similar problem was report on forum: https://dietpi.com/phpbb/viewtopic.php?t=1226

MichaIng, 2021-04-16:
Many thanks for your report.

Does the serial device actually exist?
```
ls -l /dev/ttyS0
```

I see by default a console is active as well on:
```
ls -l /dev/ttyGS0
```

There are device tree overlays available to add additional UART devices, but I think these are `/dev/ttyS1-3`. However to test:
```
sed -Ei 's/^(overlays=.*)/\1 uart1/' /boot/armbianEnv.txt
reboot
```
That should be actually the solution for the forum thread you linked 😄.

What are you trying to do with that serial device, actually? Connecting a serial terminal?

redex-2, 2021-04-16:
Hi! 
Yes, devices exist and in dietpi-config ttyS1 was enable. 
This sed resolve problem and serial1 working well.
Currently I am not able to do full test with another serial ports, but when add uart2 to armbianEnv.txt also i can use it. 
ttyS1 is first serial on main gpio and i want use this to communication with another devices by rs485.
Thank you so much and have a nice day 😃

MichaIng, 2021-04-17:
`/dev/ttyS0` cannot be used as console? It seems to be the dedicated debug port: https://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO#Layout
Swapped with the audio pins in revision 1.3.

redex-2, 2021-04-18:
I have first version of NanoPi Neo and i can't use ttyS0(debug port) as serial(i try use minicom and simple c program-nothing on out) and during boot i have garbage and some data on the bus.

MichaIng, 2021-04-20:
I have not much experience with serial consoles/devices. Probably the baudrate is wrong 🤔. But good that the other UARTs work fine.

redex-2, 2021-04-20:
I think ttyS0 is blocked or something similar. I use logic analizer for test and when i try use ttyS0 i don't have any signal. But this is not problem, in normal applications i use uart1-2(main GPIO). But i think dietpi-config when chose enable/disable for Serial/UART should add/remove automaticlly to overlays uartX.

MichaIng, 2021-04-20:
On RPi we have a dedicated option to enable/disable the UART device itself, while enabling disabling the console (+boot/kernel log to it) is the separate switch. Since UART devices can be used for different things, this makes sense.

So we'd need to add such general UART device toggles for Armbian-based images as well, based on the available device tree overlays.