MichaIng/DietPi #4278: Wiki software list not in sync with actual software

miku86, 2021-04-17:
Hey folks,

I just want to setup an automatic install of DietPi v7.0.2 64bit.

That's why I want to add some software by using the `AUTO_SETUP_INSTALL_SOFTWARE_ID` in `dietpi.txt`.

There is a link to the wiki, but some of the software I could install is not listed in the wiki.

#### Steps to reproduce

1. run `dietpi-software`
2. look at the available list of software
3. go to: https://github.com/MichaIng/DietPi/wiki/DietPi-Software-list
4. some available software, e.g. docker-compose, is missing in the wiki

#### Expected behaviour

I expect the wiki software list is in sync with the actual list of software I could install.

Greetings
Michael

Joulinar, 2021-04-17:
Hi,

The wiki might be not update. Pls have a look to our online docs. https://dietpi.com/docs/software/

miku86, 2021-04-17:
Thanks!

It's more about the values to add to `AUTO_SETUP_INSTALL_SOFTWARE_ID`.
As far as I can see, they are only in the wiki (and the source code).

Joulinar, 2021-04-17:
You are right. I could help with the ID. Which software you are looking for.

miku86, 2021-04-18:
Awesome, thanks a lot!

closed 2021-04-18 by miku86