MichaIng/DietPi #4280: trouble adding music to Kodi library

BilSheckler, 2021-04-17:
I searched for this problem in the WIKI and FAQ, but did not see it.
I am a relative NOOB, and I appreciate your help.
HW is new RPi4B and Samsung 860EVO SSD on Sabrent USB adapter cable, formatted ext4.
OS is DietPi 7.0 Allo Gui
SW is Kodi 18.7 from DietPi distribution.
PS is Canakit.
SD card is Samsung EVO Plus 32GB.

If I run the latest Raspian buster OS, it shows this SSD at:  /media/pi/Tunes, and reads it fine.

When I run DietPi/Kodi and go to the Add Music to Library screen, and try to browse for the SSD, it doesn't show up.

When I enter the address  /media/pi/Tunes  manually, I get an "unable to connect" message.

When I enter the address /media/pi/sda1  manually,  I get an "unable to connect" message.

I have used this SSD/cable combo successfully with RPi4B and Kodi 18.5 on LibreElec 9.2, but I believe that one was formatted NTFS.

Unfortunately, my notes from the last go round are not very good, and my memory is worse.

Please let me know if I have posted in the wrong forum.

Thanks for your help.

Bil Sheckler

Joulinar, 2021-04-17:
Hi, 

What is the place you mounted the SSD on? This would need to be done in DietPi drive manager.

BilSheckler, 2021-04-18:
Joulinar, 
Thank You for your reply.
I mounted the ssd with DietPi drive manager at:  /mnt/d95ed7-11d7-4495-821d-a4a3f8207846
Best Regards,
Bil Sheckler

ravenclaw900, 2021-04-18:
Hello, you would need to direct Kodi to use that mount point, or change the mount point on DietPi Drive Manager to `/mnt/iTunes` or similar.

BilSheckler, 2021-04-18:
ravenclaw900, 
That does it. Thank You Very Much.
Best Regards,
Bil Sheckler

closed 2021-04-18 by BilSheckler