MichaIng/DietPi #4281: META | Brave Creators

ravenclaw900, 2021-04-18:
It might be a good idea to sign up for [Brave Creators](https://creators.brave.com).  It's a way to earn optional 'ad revenue' from users.

MichaIng, 2021-04-20:
This does not depend on actual Brave token generation (i.e. advertising) on our websites? Otherwise sounds good 👍.

ravenclaw900, 2021-04-20:
No, not at all (though I think that is an option).

MichaIng, 2021-04-22:
I just created an account "DietPi" and added your domain + GitHub + Twitter. There was some info that it may take 24 - 48h. If I understand it right, when browsing any of these sites with Brave browser, one should see some hint that Brave tokens can be donated?

Payout is only done to Gemini or Upload crypto wallets. Any suggestions which one to use? I have zero experience with cryptocurrencies in general 😅.

ravenclaw900, 2021-04-22:
[image]
It shows up for all of them.  The best part is, if people have Auto-contribute turned on (like me), you'll automatically be paid depending on how much people are on one of the websites.

I don't have any personal experience with either Uphold or Gemini, but people seem to prefer Gemini.  It also has no withdrawal fees, so it seems like the better option.

MichaIng, 2021-04-22:
> It looks like Gemini is not available in Germany

Strange. I wonder how it is bound to a location/country, I guess it cannot receive from or payout to a German bank account then? I'll have a deeper look if even for only receiving Brave tokens and e.g. paying out to PayPa (or holding it for now)l isn't possible either. Else Uphold, okay.

Joulinar, 2021-04-22:
probably we would need to migrate to `Cayman Islands` 🤣

MichaIng, 2021-04-22:
I could offer `Indonesia`, but that isn't supported either 😄.

MichaIng, 2021-04-29:
To follow about Gemini in Germany: https://www.reddit.com/r/Gemini/comments/m3fz3f/when_will_gemini_be_available_in_germany_i_bought/

Checked Uphold and funding via BAT + withdrawal via SEPA at least is free of charge on Uphold side: https://support.uphold.com/hc/en-us/articles/206118653-Account-funding-withdrawal-costs-and-limits-Classic-Web-Wallet-

But no need to hurry with this, as BAT value is generally rising mid-long-term, I'd say. I move this issue over to the website repository to add it to the contribution page.