MichaIng/DietPi #4282: Upgrade to v6.35 failed

markus-hannover, 2021-04-18:
#### Details:
- Date           | Sun Apr 18 17:19:09 BST 2021
- Bug report     | 0322d14e-d616-4c32-ba28-0b2b4b58e338
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi B+ (armv6l) (ID=1)
- Kernel version | Linux DietPi 5.4.83+ #1379 Mon Dec 14 13:06:05 GMT 2020 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 127
- Software title | DietPi-Update
#### Steps to reproduce:
Starting dietpi-update (with or without making backup
#### Expected behaviour:
Finishing Update
#### Actual behaviour:
An error occurs and the update stops
#### Extra details:
I don't know what to post here
#### Additional logs:
```
apt-get: error while loading shared libraries: /usr/lib/arm-linux-gnueabihf/libapt-pkg.so.5.0: nonzero padding in e_ident
```

Joulinar, 2021-04-18:
Hi,

Pls can you post output of `apt update && apt upgrade`

markus-hannover, 2021-04-18:
Hi,

of course - here is the output:

` apt update
apt: error while loading shared libraries: /usr/lib/arm-linux-gnueabihf/libapt-pkg.so.5.0: nonzero padding in e_ident`
` apt upgrade
apt: error while loading shared libraries: /usr/lib/arm-linux-gnueabihf/libapt-pkg.so.5.0: nonzero padding in e_ident`

Before trying to upgrade to v6.35 I already did several other updates without problems (just for information)

ravenclaw900, 2021-04-18:
Could you run `curl -sSfLO http://ftp.debian.org/debian/pool/main/a/apt/libapt-pkg5.0_1.8.2.2_armhf.deb && dpkg -i libapt-pkg5.0_1.8.2.2_armhf.deb && rm libapt-pkg5.0_1.8.2.2_armhf.deb` and try `apt update && apt upgrade` again?

Joulinar, 2021-04-18:
Be careful, it's a Raspberry OS. Debian `apt` pool might not be correct 😉

markus-hannover, 2021-04-18:
It worked but now something is defect

`root@DietPi:~# curl -sSfLO http://ftp.debian.org/debian/pool/main/a/apt/libapt-pkg5.0_1.8.2.2_armhf.deb
root@DietPi:~# dpkg -i libapt-pkg5.0_1.8.2.2_armhf.deb
(Lese Datenbank ... 77382 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von libapt-pkg5.0_1.8.2.2_armhf.deb ...
Entpacken von libapt-pkg5.0:armhf (1.8.2.2) über (1.8.2.2) ...
libapt-pkg5.0:armhf (1.8.2.2) wird eingerichtet ...
Trigger für libc-bin (2.28-10+rpi1) werden verarbeitet ...
root@DietPi:~# rm libapt-pkg5.0_1.8.2.2_armhf.deb
 

root@DietPi:~# apt update
Ungültiger Maschinenbefehl
root@DietPi:~# apt upgrade
Ungültiger Maschinenbefehl
`

Joulinar, 2021-04-18:
I guess it was the incorrect package was installed. The one for RPi OS would be needed and not Debian

markus-hannover, 2021-04-18:
Can you tell me an ftp adress?

markus-hannover, 2021-04-18:
I found a mirror and did the download again. Actually upt update && apt upgrade is working

Joulinar, 2021-04-18:
Ok once done you can repeat `dietpi-update`

markus-hannover, 2021-04-18:
Thanks! `apt update && apt upgrade` finished - Now `dietpi-update` is working

Joulinar, 2021-04-18:
ok that's good, the issue seems to be fixed as it was related to the `apt upgrade` which already finished 😃

markus-hannover, 2021-04-18:
Thanks for help!

markus-hannover, 2021-04-18:
dietpi-update finished with success

closed 2021-04-18 by markus-hannover