MichaIng/DietPi #4283: Keep getting "[FAILED] Unknown install state/First run setup failed'

jspiers1231, 2021-04-19:
I've recently got a new pack of 32gb SD cards, and put the diet pi image on one of them, and when I try to setup dietpi, the resizing fails and when I try to finish the install process, for example, when I get to the dietpi-software screen and choose the software I want, and click install, I get the first run setup failed error, and I've tried hard drives and many different storage solutions, and I couldn't get dietpi to work for a couple days.(I'm using a pi 4)

Joulinar, 2021-04-19:
Hi,

pls can you share following 3 files

```
dietpi-firstboot.log
dietpi-firstrun-setup.log
fs_partition_resize.log
```

they are located at `/var/tmp/dietpi/logs`

jspiers1231, 2021-04-19:
[dietpi-firstboot.log](https://github.com/MichaIng/DietPi/files/6336590/dietpi-firstboot.log)

Joulinar, 2021-04-19:
What about the other one. `fs_partition_resize.log`

jspiers1231, 2021-04-19:
i copied it to /boot/ and im not seeing it

Joulinar, 2021-04-19:
Hmm that's the important one to see what the issue was during resize

jspiers1231, 2021-04-19:
on the raspberry pi, its a file, but when i cp it to /boot/ and open it on my computer, fs_partition_resize.log is not there

Joulinar, 2021-04-19:
You could try to open the file using `cat` and just copy the output from SSH terminal

jspiers1231, 2021-04-19:
root@DietPi:~# cat /var/tmp/dietpi/logs/fs_partition_resize.log
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
Disk /dev/mmcblk0: 29.1 GiB, 31266439168 bytes, 61067264 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xe8af6eb2

Old situation:

Device         Boot  Start     End Sectors   Size Id Type
/dev/mmcblk0p1        8192  532479  524288   256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 2092431 1559952 761.7M 83 Linux

/dev/mmcblk0p2:
New situation:
Disklabel type: dos
Disk identifier: 0xe8af6eb2

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 61067263 60534784 28.9G 83 Linux

The partition table has been altered.

jspiers1231, 2021-04-19:
it does this for every single hard drive and SD card no matter if used or new

Joulinar, 2021-04-19:
This always will be done on first initial boot as part of the initial setup

jspiers1231, 2021-04-19:
but it just stops after there and I just unplug the power and plug it back in to continue the process

Joulinar, 2021-04-19:
Usually it should not happen again as the trigger got removed at first

`Removed /etc/systemd/system/local-fs.target.wants/`

@MichaIng 
Can you have a look why it is not continuing. The resize seems to be fine according the log.

Joulinar, 2021-04-19:
What I don't understand, why are you going to remove power. This should not be needed and indeed interrupt first initial setup

jspiers1231, 2021-04-19:
Because it doesn't continue after:
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
Disk /dev/mmcblk0: 29.1 GiB, 31266439168 bytes, 61067264 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xe8af6eb2

Old situation:

Device         Boot  Start     End Sectors   Size Id Type
/dev/mmcblk0p1        8192  532479  524288   256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 2092431 1559952 761.7M 83 Linux

/dev/mmcblk0p2:
New situation:
Disklabel type: dos
Disk identifier: 0xe8af6eb2

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 61067263 60534784 28.9G 83 Linux

The partition table has been altered.
^^^^^^^^
After this line, it just sits there, doesn't do anything. The process does not continue until i plug the power back in

ravenclaw900, 2021-04-19:
Note that it can take a while after it shows that to finish.  How long did you wait?

jspiers1231, 2021-04-19:
5-10 minutes

Joulinar, 2021-04-19:
It depends on the size. Let's wait on @MichaIng

jspiers1231, 2021-04-19:
> Note that it can take a while after it shows that to finish. How long did you wait?

Ok so i waited a little bit longer, and it worked. So it was not the software that was the problem, it was I that was the problem. It was me pulling the power too early.

ravenclaw900, 2021-04-19:
So is the issue resolved?

jspiers1231, 2021-04-19:
yes

closed 2021-04-19 by ravenclaw900