MichaIng/DietPi #4285: Navidrome lightweight alternative to airsonic

bbsixzz, 2021-04-20:
https://feathub.com/MichaIng/DietPi/+224

https://www.navidrome.org/

Up-and-coming airsonic killer, using less resources; modern GoLang instead of Java.

Supports subsonic API!

Dylstructor, 2021-04-20:
is this an advertisement or a feature request?

MichaIng, 2021-04-20:
Many thanks for your request.

Looks reasonable, as the massive startup delay and memory consumption of the Java runtime is really the reason I'd personally currently never use Subsonic nor Airsonic.

bbsixzz, 2021-04-20:
Great, I'm holding off for the same reason. I'm using the free iSub app on iOS but have heard great things about play:sub (not free).

bbsixzz, 2021-04-27:
Still only one vote on feathub :/

Come on everyone with airsonic should want this!

bbsixzz, 2021-07-31:
I just deployed this and it works excellent following the official install guide https://www.navidrome.org/docs/installation/linux/

I configured everything like this to fit DietPi:

```
++++++ Install prerequisites

sudo apt install ffmpeg

++++++ Create directory structure

sudo install -d -o dietpi -g dietpi /opt/navidrome
sudo install -d -o dietpi -g dietpi /var/lib/navidrome

++++++ Get Navidrome

wget https://github.com/navidrome/navidrome/releases/download/v0.44.1/navidrome_0.44.1_Linux_armv6.tar.gz -O Navidrome.tar.gz
sudo tar -xvzf Navidrome.tar.gz -C /opt/navidrome/
sudo chown -R dietpi:dietpi /opt/navidrome

++++++ Create configuration file navidrome.toml in the working directory /var/lib/navidrome

DataFolder = '/var/lib/navidrome/data'
MusicFolder = '/mnt/dietpi_userdata/Music'
ImageCacheSize = '100MiB'

LogLevel = 'info'
SessionTimeout = '24h'
AuthRequestLimit = '5'
AuthWindowLength = '20s'
ScanSchedule = '@every 1h'
EnableLogRedacting = 'true'
Scanner.Extractor = 'taglib'

CoverJpegQuality = '75'
CoverArtPriority = 'embedded'

++++++ Create a systemd unit under /etc/systemd/system/ named navidrome.service

[Unit]
Description=Navidrome Music Server and Streamer compatible with Subsonic/Airsonic
After=remote-fs.target network.target
AssertPathExists=/var/lib/navidrome

[Install]
WantedBy=multi-user.target

[Service]
User=dietpi
Group=dietpi
Type=simple
ExecStart=/opt/navidrome/navidrome --configfile "/var/lib/navidrome/navidrome.toml"
WorkingDirectory=/var/lib/navidrome
TimeoutStopSec=20
KillMode=process
Restart=on-failure

# See https://www.freedesktop.org/software/systemd/man/systemd.exec.html
DevicePolicy=closed
NoNewPrivileges=yes
PrivateTmp=yes
PrivateUsers=yes
ProtectControlGroups=yes
ProtectKernelModules=yes
ProtectKernelTunables=yes
RestrictAddressFamilies=AF_UNIX AF_INET AF_INET6
RestrictNamespaces=yes
RestrictRealtime=yes
SystemCallFilter=~@clock @debug @module @mount @obsolete @reboot @setuid @swap
ReadWritePaths=/var/lib/navidrome

# You can uncomment the following line if you're not using the jukebox This
# will prevent navidrome from accessing any real (physical) devices
#PrivateDevices=yes

# You can change the following line to `strict` instead of `full` if you don't
# want navidrome to be able to write anything on your filesystem outside of
# /var/lib/navidrome.
ProtectSystem=full

# You can uncomment the following line if you don't have any media in /home/*.
# This will prevent navidrome from ever reading/writing anything there.
#ProtectHome=true

# You can customize some Navidrome config options by setting environment variables here. Ex:
#Environment=ND_BASEURL="/navidrome"

++++++ Start the Navidrome Service

sudo systemctl daemon-reload
sudo systemctl start navidrome.service
sudo systemctl status navidrome.service

++++++ Start Navidrome on Startup

sudo systemctl enable navidrome.service
```

Works great without ffmpeg right out of the box, but you'll need it if you want transcoding.

Here's hoping for inclusion in DietPi 7.5 :]

bbsixzz, 2021-08-19:
I updated Navidrome to v0.45.0 using the following procedure:

```
sudo systemctl stop navidrome.service

wget https://github.com/navidrome/navidrome/releases/download/v0.45.0/navidrome_0.45.0_Linux_armv6.tar.gz  -O Navidrome.tar.gz

sudo tar -xvzf Navidrome.tar.gz -C /opt/navidrome/

sudo systemctl start navidrome.service
```

Everything is working fine, is there any way to streamline this?

jake-g, 2021-12-18:
was looking into this as a non-java alt for airsonic (which ive been using for years), then found this thread. I'll try to install manually, but would be great to see this officially supported (assuming it works well for me)

MichaIng, 2021-12-18:
I'll move this over to discussions, so you guys can vote for it.