MichaIng/DietPi #4291: Steam install not working

Cloudftw, 2021-04-21:
### Creating a bug report/issue

#### Required Information
- DietPi version = 7
- Distro version = 10.9
- Kernel version = 5.10.17-v7l+
- SBC model = RPi4-4GB
- Power supply used = Canakit 5V3A
- SDcard used = Sandisk 32gb

#### Additional Information (if applicable)
- Software title = Steam/box86
- Was the software title installed freshly or updated/migrated? Yes
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Install steam from dietpi-software
2. launch steam using the desktop shortcut

#### Expected behaviour

- Steam should launch 

#### Actual behaviour

- Steam closes out and hangs

#### Extra details

- [message (4).txt](https://github.com/MichaIng/DietPi/files/6347306/message.4.txt)
- command line output after running /usr/games/steam since the desktop shortcut stops working after the first few times it runs

MichaIng, 2021-04-21:
Many thanks for your report.

> after the first few times it runs

This means it did run before?

Box86 is improving daily, so an update/reinstall could be at least tested:
```
dietpi-software reinstall 62
```

I'll also run some tests once DietPi v7.1 beta is up.

Cloudftw, 2021-04-21:
the first few times it at least opened the updating window for steam, but nothing after that. still waiting for a potential answer in the box86 discord.

MichaIng, 2021-04-21:
The `libappindicator1` package is installed, isn't it?
```
dpkg-query -s libappindicator1
```

Cloudftw, 2021-04-21:
yes it's installed

MichaIng, 2022-01-30:
Sorry for the eternal late reply. Tested Steam on RPi ARMv6 Bullseye image now and works fine when compiling current Box86 master. The latest release indeed fails due to a newer bug which has been solved upstream but is not yet contained in latest release: https://github.com/ptitSeb/box86/issues/558