MichaIng/DietPi #4293: Reinstall tool!

cocoflan, 2021-04-21:
An option to reinstall to 0 (clean) a system when your userdata is on an other disk and you can simple restore all settings and data via installing the apps again and data en settings are set via the data location on other disk.

MichaIng, 2021-04-21:
Many thanks for your request.

However, wouldn't be the solution to simply flash a new DietPi image? Every approach to somehow remove all additional files and revert all changes can only lead to an unclean or even broken system. Either re-flashing the image or going forward and use the PREP script are the only viable solutions I can think of.

cocoflan, 2021-04-21:
dietpi_userdata is ok to restore your apps setting, 
when reinstalling your system if the dietpi_userdata is installed on an other disk.
But it is not possible to save the system settings in such a way?

Joulinar, 2021-04-21:
not all application settings are stored in dietpi_userdata. Quite a lot of apps store their config data on an individual place. Keeping track on all of them is nearly impossible.

MichaIng, 2021-04-21:
`dietpi_userdata` contains most data and configs for `dietpi-software` installs, but Debian/GNU/Linux packages store data mostly in `/var/lib/<name>` and settings in `/etc/<name>`. `/boot/dietpi.txt` and `/boot/dietpi/` contain DietPi script-related settings. Custom compiled software store settings sometimes in `/usr/local/etc/<name>` and many programs store data and configs in the users home directory.

So a "fresh" system with all config and data preserved is naturally hand-work, depending on what you installed, how you use it and your manual changes of course. And in cases config and data of software needs to match the software version, so restoring the system below or copying config+data into a new one sometimes needs manual adjustments to make that software work.

I'm no big fan of scattered configs and data either, but that is something we need to live with in the Linux/UNIX world. We try to store those parts we can control combined in dietpi_userdata, which covers not everything, but the most important stuff usually, e.g. when the system drive breaks, your cloud and media files are still there (when on a dedicated drive).
____________
What is actually the issue you want to solve with a cleaned system?

cocoflan, 2021-04-21:
I just want to quick recover a broken system, in limited time.

Joulinar, 2021-04-21:
best would be to create backups on regular basis or before preforming big changes. DietPi already offering a feature to restore a backup automatically on first initial boot from a USB drive.