MichaIng/DietPi #4294: Open Beta v7.1 | Please help testing and hardening the upcoming release

MichaIng, 2021-04-21:
RC version | v7.1.0
---------- | -------
v7.1.0 PR | #4288
Changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release planned | **2021-04-28**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av7.1
____
### Beta v7.1.0
_(2021-04-21)_

#### Supported SBC changes
- For historical reasons, DietPi had internal hardware identifiers for SBCs which are not officially supported anymore for a long time, namely variouos Orange Pi and Banana Pi models, BeagleBone Black and LeMaker Guitar. Since we do not provide a reference image and do not test features on those boards, we have no chance to reasonably use those hardware IDs code-wise to apply tweaks or enable features, which renders them meaningless. To avoid wrong assumptions, we hence removed the identifies for the named SBC models. As a replacement, we added a set of generic SoC identifiers, which can as well be selected from within the DietPi-PREP system converter script. This means that one is e.g. able to select "Generic Allwinner H3" and have an SoC identifier set by this, which is used within the code for conditionals based on the SoC, e.g. regarding GPU acceleration and other SoC-internal hardware features. The removed SBC models are converted automatically to those generic identifiers, so that e.g. after the update an OrangePi PC will be shown in our scripts and banner as "Generic Allwinner H3". While it looks like a degradation, it is a hardening, as it assures that no false assumptions and conditionals are triggered in our scripts but only those which we can know for sure being true, regardless of the initial Debian image you used.
- We removed the Firefly RK3399 from our supported devices list, as no a single user is reported via DietPi-Survey (https://dietpi.com/survey/#device). If case there were actually Firefly RK3399 DietPi systems, not participating DietPi-Survey, those would be migrated to the "Generic Rockchip RK3399" identifier, which code-wise maintains the exact same support.

#### Changes
- DietPi-LetsEncrypt | It is now possible to obtain certificates for multiple domains, given as comma-separated list, with the first domain used as primary webserver name.
- DietPi-LetsEncrypt | It is now possible to obtain certificates without having a webserver installed. Certbot can start an own internal webserver for such case and the certificate can then used to manually enable HTTPS for other web applications, running on other ports than 80/443.
- DietPi-LetsEncrypt | It is now possible to enable OCSP (Online Certificate Status Protocol) stapling for the obtained certificate. This allows clients to check if the certificate has been revoked, e.g. when the server has been compromised or the private key stolen.
- DietPi-LetsEncrypt | Lighttpd: HTTPS is now enabled for IPv6 requests and the deprecated TLSv1.0 and TLSv1.1 are disabled from Debian Buster on. With the Lighttpd version shipped by Debian Stretch, those TLS versions cannot be disabled.
- DietPi-NordVPN | The script reveived a major rework to allow being setup as ProtonVPN client and even with a custom OVPN config. For this reason it has been renamed to "DietPi-VPN", with the script moved to /boot/dietpi/dietpi-vpn and the console command alias changed to "dietpi-vpn" accordingly. Additionally a killswitch feature has been added which forcefully drops all WAN packets not sent through the VPN, when the connection got lost, until the VPN is disconnected manually/intentionally. Further is IPv6 now automatically disabled when the VPN connection is established. This is required to prevent IPv6 leaks as most publish VPN providers/servers do not support IPv6. When it has not been disabled before, IPv6 is re-enabled automatically once the VPN connection stops. Many thanks to @ravenclaw900 for doing this major rework: https://github.com/MichaIng/DietPi/pull/4180
- DietPi-FS_partition_resize | Added support to automatically resize F2FS and Btrfs filesystems on first boot.
- DietPi-Drive_Manager | Added support for resizing F2FS and Btrfs filesystems as well as format- and filesystem check & repair support for XFS filesystems.
- DietPi-Drive_Manager | When adding Samba mounts, credentials are not added in plain text to /etc/fstab anymore, but stored instead in a separate per-mount credential file with strict root-only read permissions. Many thanks to @TheOriginalMrWolf for doing this suggestion: https://github.com/MichaIng/DietPi/issues/4082
- DietPi-Config | Added a safe overclocking profile for RPi 3+ models. Many thanks to `@lone` for doing long-term stability tests and reporting back the result: https://dietpi.com/phpbb/viewtopic.php?p=32285#p32285
- DietPi-Config | When disabling the RPi camera feature, the bcm2835_isp kernel module is now additionally blacklisted. Since kernel 5.X it is otherwise loaded automatically and pulls in the whole camera modules stack as dependency, adding some additional memory usage and boot overhead. Many thanks to @ferbar for making us aware of this: https://github.com/MichaIng/DietPi/issues/4203
- DietPi-Software | The "optimised" and "additional" software menus have been merged. The separation did not follow consistent rules and this change allows us to have development tools and platforms in a more prominent position. Also it simplifies the software selection and addresses possible confusion about that separation among our users.
- DietPi-Software | Since multiple software titles were listeing on the ports 5000 and 8000, the default ports of those titles have been changed to avoid conflicts on concurrent installs. No change is done during the DietPi update, even a software title reinstall won't change it, in most cases. For details see the individual software changes below.
- DietPi-Software | Mosquitto: Since v2, by default remote connections and no unauthenticated requests are possible anymore. On fresh installs and reinstalls, we'll enable remote connections, but create a password file, so that MQTT clients need to authenticate with username "mosquitto" and the global software password by default. Many thanks to @mattsmithuk for reporting this change: https://github.com/MichaIng/DietPi/issues/4133
- DietPi-Software | IceCast: The streaming server can now be installed on virtual machines as well and the default web UI password will now be the global software password instead of a random one. Since the DarkIce config file /etc/darkice.cfg contains the global software password in plain text, its permission mode is changed to 600 to limit read access to the root user only.
- DietPi-Software | OctoPrint: On fresh installs, the default listening port has been changed from 5000 to 5001 to avoid conflicts with Shairport Sync.
- DietPi-Software | Firefox Sync Server: On fresh installs, the default listening port has been changed from 5000 to 5002 to avoid conflicts with Shairport Sync.
- DietPi-Software | rTorrent: For performance reasons, and to avoid conflicts with Shairport Sync, the default SCGI/RPC access has been changed from TCP port 5000 to UNIX socket. This change is applied on fresh installs and reinstalls of rTorrent, as it requires a change on both all sides, rTorrent, ruTorrent and the webserver. HTTP authentication is now enforced for the rTorrent RPC access with all webservers, using the global software password by default, and the password hash algorithm has been hardened with Nginx.
- DietPi-Software | WebIOPi: On fresh installs, the default listening port has been changed from 8000 to 8002 to avoid conflicts with IceCast. Additionally the install process has been slightly sped up by skipping unnecessary installer-internal APT calls.
- DietPi-Software | Koel: On fresh installs and reinstalls, the listening port has been changed from 8000 to 8003 to avoid conflicts with IceCast. Additionally the install process has been simplified and dependencies reduces, since after Koel v5.0.0 pre-compiled frontend binaries are available. This also allows a completely unattended install now. As little security hardening, the Koel MariaDB database password has been changed to a long random character string and the config files permission mode has been changed to 600 to permit read access to the Koel service user only. Finally as little performance enhancement, Koel now connects via UNIX s [... cut ...]
- DietPi-Software | Koel: Resolved issues which caused a failing install, caused by changed archive directory names and changed Laravel dependencies.
- DietPi-Software | myMPD: Resolved an issue where the service fails to start because of a renamed setting. Many thanks to @sofad for reporting this issue: https://github.com/MichaIng/DietPi/issues/4256
- DietPi-Software | O!MPD: Resolved an issue where browsing the media directory from the web UI failed because of a missing slash in the local config file. Many thanks to `@pinkdot` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=8904

Joulinar, 2021-04-24:
Was testing migration of Sonarr without issues. At least on an empty system it was working well. Notification was given on DietPi v7.1 update and I could reinstall Sonarr afterwards. 

@MichaIng 
One question. Does Sonarr v3 still require `Mono`? Because it's still a dependency and will be setup during `dietpi-software reinstall 144`

Joulinar, 2021-04-24:
I guess I found an issue on `dietpi-vpn`. 

What I did:

1. run `dietpi-vpn`
2. select `Custom`
3. `OK`
4. don't do anything, just select `<Exit>`
5.  Exit DietPi-Explorer? `<Ok>`
6. getting the error dialog now as the system is trying top open `/etc/openvpn/client.ovpn` still
7. this is the error dialog
<details>
  <summary>Click to expand!</summary>

```
                ┌────────────────────────────────────────────────────┤ DietPi-VPN ├────────────────────────────────────────────────────┐
                │  - Command: cp  /etc/openvpn/client.ovpn                                                                             │
                │  - Exit code: 1                                                                                                      │
                │  - DietPi version: v7.1.0 (MichaIng/beta) | HW_MODEL: 3 | HW_ARCH: 2 | DISTRO: 5                                     │
                │  - Image creator: DietPi Core Team                                                                                   │
                │  - Pre-image: Raspbian Lite                                                                                          │
                │  - Error log:                                                                                                        │
                │ cp: cannot stat '': No such file or directory                                                                        │
                │                                                                                                                      │
                │                         Retry          : Re-run the last command that failed                                         │
                │                         DietPi-Config  : Edit network, APT/NTP mirror settings etc                                   │
                │                         Open subshell  : Open a subshell to investigate or solve the issue                           │
                │                         Send report    : Uploads bugreport containing system info to DietPi                          │
                │                                        ●─ Devs only ──────────────────────────────────────●                          │
                │                         Change command : Adjust and rerun the command                                                │
                │                                                                                                                      │
                │                                                                                                                      │
                │                                  <Ok>                                      <Exit>                                    │
                │                                                                                                                      │
                └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```
</details>

As well I see another error if I scroll up on my SSH session
`/boot/dietpi/dietpi-vpn: line 125: /tmp/.dietpi-explorer_selected_location: No such file or directory`

MichaIng, 2021-04-24:
> Was testing migration of Sonarr without issues.

Yes, the DEB package pre/postinst scripts do quite a good to to cleanly migrate Sonarr v2 instances. And the package can be/is pre-configured to use our custom data dir right from the start. A few settings cannot be migrated, e.g. removed or changed ones, like arguments for the "Custom Script" feature, but all important parts are migrated safely.

> One question. Does Sonarr v3 still require Mono?

Yes it does, same as Radarr v3. Only Jackett got a non-Mono .NET version.

Good find with DietPi-VPN. Looks like a return is missing when `dietpi-explorer` returns an exit code, if does it not return one?? I'll check.

Joulinar, 2021-04-24:
the error happen right after this box

```
┌──────────────┤ DietPi-Explorer ├───────────────┐
│                                                │
│ Exit DietPi-Explorer?                          │
│                                                │
│           <Ok>               <Cancel>          │
│                                                │
└────────────────────────────────────────────────┘
```

MichaIng, 2021-04-24:
Solved: https://github.com/MichaIng/DietPi/commit/8fe74a87610d4d7b63ac018537b403d54d97e6ee
The commit looks larger than it is. I replaced two:
```
if G_WHIP; then

	<many code lines>

fi
```
with
```
G_WHIP || return 0

<many code lines>
```
to reduce unnecessary code line indentation.

MichaIng, 2021-04-24:
Another little related enhancement: https://github.com/MichaIng/DietPi/commit/2c88d26e4faf568e6239495fc9eebe23a6253b52

When "Custom" was selected but the DietPi-Explorer exited, "Custom" was shown in the menu, like if it was selected successfully. Since the custom file is not shown anywhere in the menu it is not visible that no valid selection has been made.

Now, in that case no variable is touched, hence the previous provider is shown as if nothing had changed, which is the true.

MichaIng, 2021-04-25:
### Beta v7.1.1 has been merged: #4299

gstrauss, 2021-04-26:
> DietPi-LetsEncrypt | Lighttpd: HTTPS is now enabled for IPv6 requests and the deprecated TLSv1.0 and TLSv1.1 are disabled from Debian Buster on. With the Lighttpd version shipped by Debian Stretch, those TLS versions cannot be disabled.

Debian Buster has lighttpd 1.4.53 (+ some patches).  buster-backports has lighttpd 1.4.59.  Since lighttpd 1.4.48, lighttpd supports 
`ssl.openssl.ssl-conf-cmd = ("MinProtocol" => "TLSv1.2")`
If you are on an older system with very old openssl, then with openssl 1.0.2
`ssl.openssl.ssl-conf-cmd = ("Protocol" => "-ALL, TLSv1.2")`

lighttpd 1.4.56 and later disable old TLS versions (< TLSv1.2) by default.

MichaIng, 2021-04-26:
That is how it's done now: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-letsencrypt#L160-L170

The old system requires the `ssl.use-sslv3 = "disable"` syntax anyway, so no overlap of old OpenSSL and new Lighttpd syntax or the other way round.

> lighttpd 1.4.56 and later disable old TLS versions (< TLSv1.2) by default.

That is good to know, however no harm to keep the directive to disable it, e.g. if some old/custom config is in place that overrides the Lighttpd default. Also I like to have those settings exposed in general, so users see what is possible by only checking the config and could change it to `"MinProtocol" => "TLSv1.3"` e.g.

gstrauss, 2021-04-26:
`ssl.openssl.ssl-conf-cmd = ("MinProtocol" => "TLSv1.2", "Options" => "-SessionTicket")`

In lighttpd 1.4.56 and later, it is recommended that you remove `"Options" => "-SessionTicket"` since it is safe to use TLSv1.2 Session Tickets in lighttpd 1.4.56 and later, since lighttpd rotates the session ticket encryption keys automatically starting with lighttpd 1.4.56.  More details at [lighttpd TLS Session Tickets](https://redmine.lighttpd.net/projects/lighttpd/wiki/Docs_SSL#Session-Tickets)  This lighttpd feature is something that Apache and nginx do not do as well (https://github.com/mozilla/server-side-tls/issues/135)

Separately, `ssl.disable-client-renegotiation = "enable"` is the default, and it is recommended to keep the default for security reasons.  This is not something that someone should change on a whim, so it probably ought not to be listed in the config.

MichaIng, 2021-04-26:
Great news about session tickets in Lighttpd v1.4.56. I guess for the patched v1.4.53 from Debian Buster this is not true yet?
At least for Bullseye and up I'll have it removed. We cannot count on backports as RPi (Raspbian) has no backports. Probably I'll add a CLI-based version estimation with next release, to correctly treat backports and own source builds.

naddel91, 2021-04-26:
dietpi-ddns does not work for me. I tried domain and username to login, but it reports "DDNS update test failed, please check your input: unknown".
I only have login data (username and password). that cannot be the culprit of the error here since manually logging in (website dynu.net) works fine.

Joulinar, 2021-04-26:
Hi,

usually DDNS provider offering an update URL that would need to be added inside `dietpi-ddns`. There is a describtion according dynu.com

https://www.dynu.com/DynamicDNS/IPUpdateClient/Custom

naddel91, 2021-04-26:
it works using the update URL. but why is Dynu in the profiles then?

Joulinar, 2021-04-26:
according the script, update URL is exactly same as on dynu.com web side

https://github.com/MichaIng/DietPi/blob/29502ae182253876eca4f875c9651713fc4d022e/dietpi/dietpi-ddns#L130

Do you use any special character on your password? just a question.

naddel91, 2021-04-27:
strange. will try again after stable release. thanks for your help.

Joulinar, 2021-04-27:
You mean the custom profile? Nö that not the reason. The custom profile doesn't know what fields are required. Therefore it will aks all data. However in the custom profile it should be fine just to enter the update link according provider website and leave user + pw empty

naddel91, 2021-04-27:
I mean, the script did not ask me for username, only domain when using the pre-configured Dynu-profile.

Joulinar, 2021-04-27:
This is correct. As you can see on provider web site, this is not required. Only domain and password.

naddel91, 2021-04-27:
ok, good to know. will try again when 7.1(.1) is public.

Joulinar, 2021-04-27:
Actually there is no change planned for the ddns tool. Means it would be good to find out why it is failing in your case

naddel91, 2021-04-27:
test update failed when entering through pre-configured profile (dynu). working fine when updating manually (custom).

Joulinar, 2021-04-27:
can you do a screenshot from the error message pls

naddel91, 2021-04-27:
it is "nochg" that failed. when entering manually there is one more prompt to enter (username). and manually using this here works fine:

https://api.dynu.com/nic/update?hostname=hostdomain&password=mypassword

using the preconfigured script (dynu) with the same credentials fails telling me "NOCHG" as error.

naddel91, 2021-04-27:
https://abload.de/img/ddnsdietpifailssljfd.png

Joulinar, 2021-04-27:
the user name should not be required. As you see on the link you posted, there is no username value.

naddel91, 2021-04-27:
I know, but there might be something strange with the underlying URL. because custom works fine. same input into the Dynu-profile fails.

Joulinar, 2021-04-27:
Found what `nochg` means

https://www.dynu.com/en-US/Forum/ViewTopic/nochg/6911

MichaIng, 2021-04-27:
Many thanks for reporting. The response is checked wrong. Can you please test it with after:
```
curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/dev/dietpi/dietpi-ddns -o /boot/dietpi/dietpi-ddns
```

naddel91, 2021-04-27:
that did it! Micha FTW!
Joulinar, too. Of course!

MichaIng, 2021-04-27:
Great.

When using a custom provider, we do not know how the response is composed, so we just check for successful connection + HTTP 2** response (curl with "-f" option). That means that the test can succeed even if the update failed, when the server does not response with an HTTP error but an error body as body or a regular 200 response (like Dyno and also DuckDNS).

So adding further providers gives the test more meaning, but the response need to be checked correctly of course 😄.

naddel91, 2021-04-27:
So dietpi 7.1.1 is stable now? ;)

Joulinar, 2021-04-27:
well we already on 7.1.2 within development 😉 
https://github.com/MichaIng/DietPi/pull/4308

marcelloinfoweb, 2021-04-28:
The temperature display on the Home screen when logging in via ssh has been removed.

Joulinar, 2021-04-28:
what SBC you are running on?

marcelloinfoweb, 2021-04-28:
> ‎em que SBC você está executando?‎

Raspberry Zero W

Joulinar, 2021-04-28:
for me it is working well on my RPi3B+

```
─────────────────────────────────────────────────────
 DietPi v7.1.2 (dev) : 02:37 - Wed 04/28/21
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B+ (armv7l)
 - Uptime : up 0 minutes
 - CPU temp : 46'C : 114'F (Optimal temperature)
 - LAN IP : 192.168.0.12 (eth0)
 - Info Text : !!! DEMO 32bit !!!
 ─────────────────────────────────────────────────────
```

can you check inside `dietpi-banner` if `CPU temp` is checked

```
 ┌───────────────────────────────────────────────────────────────
 │ Please (de)select options via spacebar to be shown in the :   
 │                                                               
 │    [*] 0   Device model                                       
 │    [*] 1   Uptime                                             
 │    [*] 2   CPU temp                                           
 │    [ ] 3   FQDN/hostname                                      
 │    [ ] 4   NIS domainname                                     
 │    [*] 5   LAN IP                                             
 │    [ ] 6   WAN IP                                             
 │    [ ] 7   Freespace (RootFS)                                 
 │    [ ] 8   Freespace (userdata)                               
 │    [ ] 9   Weather (wttr.in)                                  
 │    [*] 10  Info Text                                          

```

marcelloinfoweb, 2021-04-28:
> for me it is working well on my RPi3B+
>
> ```
> ─────────────────────────────────────────────────────
>  DietPi v7.1.2 (dev) : 02:37 - Wed 04/28/21
>  ───────────────────────────────────────────────────── [...]

I didn't know this "dietpi-banner" command, thanks !!!

MichaIng, 2021-04-28:
But also I accidentally switched the defaults. On a VM, the temperature should have not been shown by default (since a VM cannot know it). Now it is correct: https://github.com/MichaIng/DietPi/commit/4513a8446178fd78c6f9132056f8ee0367f83b2c

MichaIng, 2021-04-28:
### Beta v7.1.2 has been merged: #4308

MichaIng, 2021-04-28:
### DietPi v7.1 has been released: #4305
Many thanks to all testers ❤️!

closed 2021-04-28 by MichaIng