MichaIng/DietPi #4295: Troubles getting touchscreen working

iamslan, 2021-04-22:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

I am not able to get my touchscreen working. Display is working perfectly through HDMI, only touch inputs are unavailable.
The product is [this one](https://www.amazon.fr/gp/product/B07PKLGMSY).
The driver which is working on Raspbian is [this one](https://github.com/goodtft/LCD-show/blob/master/LCD7C-show).
The script is messing with the installation, I read [this whole topic](https://github.com/MichaIng/DietPi/issues/2590).
What I did is copying only the needed files, modifying both boot txts, and activate UART/I2C/SPI.
No luck at this point, I'd take any suggestions to try them out !
Best,

MichaIng, 2021-04-22:
Many thanks for your report.

Please provide details about every setup step that you did, at best the complete config files:
- `/boot/config.txt`
- `/boot/cmdline.txt`
- All files in `/etc/X11/xorg.conf.d/`

**The scripts from https://github.com/goodtft/LCD-show are indeed nothing I would recommend on any system**, as they replace whole core system/kernel/boot configuration files as if the LCD was the only thing one is ever going to install. They may work on current plain Raspberry Pi OS at time of writing, may not work anymore when default Raspberry Pi OS changes, and when you do/did own changes, they break/override those or other parts of the system and can make it unbootable due to `cmdline.txt` replacement. It's sad that there is no clean documentation, which device tree overlay to use, how to use/enable the contained driver and resolution + calibration parameters, as nothing else should be required.

iamslan, 2021-04-22:
[config.txt](https://github.com/MichaIng/DietPi/files/6358369/config.txt)

[cmdline.txt](https://github.com/MichaIng/DietPi/files/6358370/cmdline.txt)

[image]

MichaIng, 2021-04-22:
Looks all good.

X server is installed, including the `xserver-xorg-input-libinput`?
Also could you try to move the other file in `/etc/X11/xorg.conf.d/` away? It's to disable screen power savings and should not interfere as there is no `InputClass` section, but just to rule it out...

iamslan, 2021-04-23:
`xserver-xorg-input-libinput` is installed (0.28.2-2)

I tried removing `/etc/X11/xorg.conf.d/98-dietip-disable-dpms.conf`, but it does not change anything.
Touchscreen is still not responding.

closed 2021-04-23 by iamslan

anandv85, 2025-02-06:
> `xserver-xorg-input-libinput` is installed (0.28.2-2)
> 
> I tried removing `/etc/X11/xorg.conf.d/98-dietip-disable-dpms.conf`, but it does not change anything. Touchscreen is still not responding.
> 
> EDIT: I tried to change the touchscreen input's USB port, and voila, it works. No clue why. We can close this one, thanks for your help!

Probably the touch screen is not getting enough power from the USB port. You can also use Y connector and that should work on any USB port.