MichaIng/DietPi #4296: Can't connect to Ampache server using any Andoid clients

jaum20, 2021-04-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
 G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.9
- Kernel version | `uname -a`
Linux DietPi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
-RPi 4 Model B (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- 5V 3A generic chinese power supply
- SDcard used | (EG: SanDisk ultra)
SanDisk ultra 16GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
-Ampache
- Was the software title installed freshly or updated/migrated?
- Installed via dietpi-software
- Can this issue be replicated on a fresh installation of DietPi?
-Yes

ddc07e6f-3fb3-471d-82fc-b3b9e01aa3b8
- Bug report ID | `echo $G_HW_UUID`
echo $G_HW_UUID
#### Steps to reproduce

1. Install Ampache via dietpi-software script
2. Access the server via browser (http://192.168.0.101/ampache) and configure it as described on the wiki 
3. Try to connect to the server using Power Ampache or another ampache android client
4. Can't connect to server 

#### Expected behaviour

- The client should be able to connect to the server

#### Actual behaviour

- I'm getting a 'password may be wrong' or 'server AUTH failed' errors. I'm sure that the password is correct because I can connect using browser

#### Extra details

- Tried serveral clients: Power Ampache, Amdroid, AMPlayer, Just Play...

MichaIng, 2021-04-23:
Many thanks for your report.

Please refer to the Ampache and/or client docs for this.

From what I found, by default Ampache denies requests to its RPC API form clients which are not allowed explicitly: https://github.com/ampache/ampache/blob/develop/config/ampache.cfg.php.dist#L179-L185

Here the Wiki page about this (which is outdated as the ALC defaults to "true" as can be seen in the default config file): https://github.com/ampache/ampache/wiki/ACL
I just could not find the place where to define those IP ranges. Probably it can be done from within the web interface?

Another Wiki page where this is included without saying where (file/interface) this has to be added: https://github.com/ampache/ampache/wiki/API#creating-the-apirpc-acl-entry

If I understood it correctly, when the ACL setting is disabled, the API is as well. So in case you want to access it from your mobile phone with unpredictable variable IP address, you need to use `0.0.0.0 - 255.255.255.255` as IP range to permit API access from all possible IPs.

jaum20, 2021-04-24:
@MichaIng Thank you for your response. I think the ALC is set already

https://imgur.com/a/ptE4pZD

MichaIng, 2021-04-24:
I guess you verified that for IPv6 requests, using `::` as start IP only is correct? Could you try to add an IPv6 RPC/API rule as well?

jaum20, 2021-04-24:
According to the wili page (https://github.com/ampache/ampache/wiki/ACL), only IPV4 is supported. Also, those  (with excelption of the last one) was already there, probabily by default.

My decision to post this here insted of Ampache's git is beacuse I think it has some to do with the lighttpd server config (installed by the dietpi script), since ampache recomends use ampache as server ant its documentation regarding lighttpd is scarce.

MichaIng, 2021-04-24:
Here Lighttpd is listed as well: https://github.com/ampache/ampache/wiki/Installation#prerequisites

The install instructions only show very minimal Apache configuration:
```
<Directory /var/www/html/>
Options Indexes FollowSymLinks
AllowOverride All
Require all granted
</Directory>
```
- Grand access
- Allow directory listing... why ever that is done, as it is definitely not required to get a list of raw files in any directory.
- Enable `.htaccess`

And now I think we found the issue. E.g. in `rest/` directory, the following rewrite rules are applied via `.htaccess`:
```
<IfModule mod_rewrite.c>
    RewriteEngine On
    RewriteCond %{REQUEST_FILENAME} !-d
    RewriteCond %{REQUEST_FILENAME} !-s
    RewriteRule ^(.+)\.view$ /rest/index.php?ssaction=$1 [PT,L,QSA]
    RewriteRule ^fake/(.+)$ /play/$1 [PT,L,QSA]
</IfModule>
```
This does not even work with Apache2, since `/rest/index.php` assumes Ampache being installed in the webroot while it's in a subdirectory and would hence need to be `/ampache/rest/index.php?ssaction=$1`. Same for `/play/` below and the rewrites in that directory.
Ah that is even explained here: https://github.com/ampache/ampache/wiki/Installation#apache
Scrolling down are the even Nginx and Lighttpd configs, containing all rewrite rules.

Please try the following:
```
lighty-enable-mod rewrite
cat << '_EOF_' > /etc/lighttpd/conf-available/98-dietpi-ampache.conf
# Based on: https://github.com/ampache/ampache/wiki/Installation#lighttpd
$HTTP["url"] =~ "^/ampache($|/)" {
    url.rewrite-if-not-file += (
        "^/(.*)\.(css|js|jpg|png|gif)$" => "$0",
        "^/ampache/rest/(.+)\.view$" => "/ampache/rest/index.php?ssaction=$1",
        "^/ampache/rest/fake/(.+)$" => "/ampache/play/$1",
        "^/ampache/play/art/([^/]+)/([^/]+)/([0-9]+)/thumb([0-9]*)\.([a-z]+)$" => "/ampache/image.php?object_type=$2&object_id=$3&auth=$1&thumb=$4&name=art.jpg",
        "^/ampache/play/([^/]+)/([^/]+)/([^/]+)/([^/]+)(/.*)?$" => "/ampache/play/$5?$1=$2&$3=$4",
        "^/ampache/play/([^/]+)/([^/]+)(/.*)?$" => "/ampache/play/$3?$1=$2",
        "^/ampache/play(/[^/]+|[^/]+/|/?)$" => "/ampache/play/index.php",
        "^/ampache/channel/([0-9]+)/(.*)$" => "/ampache/channel/index.php?channel=$1&target=$2"
    )
}
_EOF_
lighty-enable-mod dietpi-ampache
systemctl restart lighttpd
```

jaum20, 2021-04-24:
The clients still not able to connect to the server plus now when I try to play a song using browser, the song doesn't plays and instead the browser downloads an 'apache playlist.m3u' file

MichaIng, 2021-04-25:
Okay, I think we'll have to dig deeper into the API then. The above is copy&paste from the Ampache wiki, and the rewrite conditions and sources look correct, but obviously they overlap with what Ampache internally (web interface) does. The general problem I have with rewrite needs is that the exact query strings do always work and hence clients should be able to use them directly, rather then relying on an obviously fragile and complex rewrite structure.
And I think they do. E.g. the `/rest` redirects are for the Subsonic backend only, most `/play` redirects are for beautification, one for downloading single songs. `/channel` is required for the channels broadcast feature.

Btw, you did assure that the clients connect to the `http(s://your.domain/ampache` URL path, rather than the webroot, right? The AUTH errors also indicate that it does not fail at the API but at an earlier stage.

Please also check webserver and client logs and enable additional Ampache debug logging: https://github.com/ampache/ampache/wiki/Troubleshooting#Enable-Logging
The log path needs to be owned by the `www-data` user so that it can create and write contained log files.

jaum20, 2021-04-25:
> Btw, you did assure that the clients connect to the `http(s://your.domain/ampache` URL path, rather than the webroot, right? The AUTH errors also indicate that it does not fail at the API but at an earlier stage.

Checking the log I found this:

```
2021-04-25T11:45:14-03:00 [ampache] (Session) -> Session created: 257cd7f564224a0ba3ae4632220fac1d
2021-04-25T11:45:14-03:00 [ampache] (Api) -> Login Success, passphrase matched
2021-04-25T11:45:14-03:00 [ampache] (Dba) -> Error_query SQL: SELECT COUNT(`id`) FROM `label`
2021-04-25T11:45:14-03:00 [ampache] (Dba) -> Error_query MSG: ["42S02",1146,"Table 'ampache.label' doesn't exist"]

```

So looks like it is a database error

Joulinar, 2021-04-25:
@MichaIng 
something we would need to add to our install script or at least our docs?

MichaIng, 2021-04-25:
Obviously our old database is not migrated anymore successfully in every bit. I'll update the file on our server, based on a clean install.

MichaIng, 2021-05-30:
Database has been updated. It also is much cleaner now, shipping less junk but only what's necessary for default login and `/mnt/dietpi_userdata/Music` added as library.

closed 2021-05-30 by MichaIng