MichaIng/DietPi #4298: Unbound stuck in a weird state after 6.34-7.02 upgrade

farsider-uk, 2021-04-23:
- Date           | Fri Apr 23 19:06:31 BST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | PharCyder
- Pre-image      | Debian
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | Linux PharCloud 4.19.0-16-amd64 #1 SMP Debian 4.19.181-1 (2021-03-19) x86_64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | systemctl restart unbound
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. ... Installed Unbound in DietPi v6.34 which failed due to the known issues installing PiHole & Unbound together at the time through DietPI Software. Tried to uninstall Unbound via DietPi Software that errored too.
2. ... Upgrade DietPi to v7 where I can see a fix is mentioned. Tried to re-install Unbound but now get the error below.

#### Expected behaviour:

- ... I'd expect Unbound to re-install correcting what stopped it installing before.

#### Actual behaviour:

- ... The error in the logs below is shown when trying to re-install Unbound through dietpi-software

#### Extra details:

- ... dietpi-services status shows this however Unbound isn't listed as an installed application
[FAILED] DietPi-Services | ● unbound.service - Unbound DNS server
   Loaded: loaded (/lib/systemd/system/unbound.service; disabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/unbound.service.d
           └─dietpi.conf
   Active: failed (Result: exit-code) since Fri 2021-04-23 19:06:31 BST; 7min ago
     Docs: man:unbound(8)
  Process: 2090 ExecStartPre=/usr/lib/unbound/package-helper chroot_setup (code=exited, status=0/SUCCESS)
  Process: 2093 ExecStartPre=/usr/lib/unbound/package-helper root_trust_anchor_update (code=exited, status=0/SUCCESS)
  Process: 2097 ExecStart=/usr/sbin/unbound -d $DAEMON_OPTS (code=exited, status=1/FAILURE)
 Main PID: 2097 (code=exited, status=1/FAILURE)

Apr 23 19:06:31 PharCloud systemd[1]: Failed to start Unbound DNS server.
Apr 23 19:06:31 PharCloud systemd[1]: unbound.service: Start request repeated too quickly.
Apr 23 19:06:31 PharCloud systemd[1]: unbound.service: Failed with result 'exit-code'.
Apr 23 19:06:31 PharCloud systemd[1]: Failed to start Unbound DNS server.
Apr 23 19:06:31 PharCloud systemd[1]: unbound.service: Start request repeated too quickly.
Apr 23 19:06:31 PharCloud systemd[1]: unbound.service: Failed with result 'exit-code'.
Apr 23 19:06:31 PharCloud systemd[1]: Failed to start Unbound DNS server.
Apr 23 19:06:34 PharCloud systemd[1]: unbound.service: Start request repeated too quickly.
Apr 23 19:06:34 PharCloud systemd[1]: unbound.service: Failed with result 'exit-code'.
Apr 23 19:06:34 PharCloud systemd[1]: Failed to start Unbound DNS server.

#### Additional logs:
```
Warning: The unit file, source configuration file or drop-ins of unbound.service changed on disk. Run 'systemctl daemon-reload' to reload units.
Job for unbound.service failed because the control process exited with error code.
See "systemctl status unbound.service" and "journalctl -xe" for details.
```

ravenclaw900, 2021-04-23:
Hello, could you try running `systemctl daemon-reload` then try again?

Joulinar, 2021-04-23:
As pls share `journalctl -u unbound`

farsider-uk, 2021-04-23:
Tried `systemctl daemon-reload`. The output after trying to restart is:

```
[FAILED] DietPi-Services | ● unbound.service - Unbound DNS server
   Loaded: loaded (/lib/systemd/system/unbound.service; disabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/unbound.service.d
           └─dietpi.conf
   Active: failed (Result: exit-code) since Fri 2021-04-23 19:06:31 BST; 17min ago
     Docs: man:unbound(8)
 Main PID: 2097 (code=exited, status=1/FAILURE)

Apr 23 19:06:31 PharCloud systemd[1]: Failed to start Unbound DNS server.
Apr 23 19:06:31 PharCloud systemd[1]: unbound.service: Start request repeated too quickly.
Apr 23 19:06:31 PharCloud systemd[1]: unbound.service: Failed with result 'exit-code'.
Apr 23 19:06:31 PharCloud systemd[1]: Failed to start Unbound DNS server.
Apr 23 19:06:31 PharCloud systemd[1]: unbound.service: Start request repeated too quickly.
Apr 23 19:06:31 PharCloud systemd[1]: unbound.service: Failed with result 'exit-code'.
Apr 23 19:06:31 PharCloud systemd[1]: Failed to start Unbound DNS server.
Apr 23 19:06:34 PharCloud systemd[1]: unbound.service: Start request repeated too quickly.
Apr 23 19:06:34 PharCloud systemd[1]: unbound.service: Failed with result 'exit-code'.
Apr 23 19:06:34 PharCloud systemd[1]: Failed to start Unbound DNS server.
```

`journalctl -u unbound` Output attached
[journalctl.txt](https://github.com/MichaIng/DietPi/files/6367402/journalctl.txt)

Thanks for your help guys!

ravenclaw900, 2021-04-23:
Okay, could you try manually uninstalling Unbound?
```
apt remove unbound
rm -r /etc/unbound
```
See if the install works after that.

Joulinar, 2021-04-23:
this could be a solution as well https://github.com/MichaIng/DietPi/issues/4159#issuecomment-788110283

farsider-uk, 2021-04-23:
> Okay, could you try manually uninstalling Unbound?
> ```
> apt remove unbound
> rm -r /etc/unbound
> ```
> See if the install works after that.

This worked. Many thanks for your support gents.

closed 2021-04-23 by farsider-uk