MichaIng/DietPi #4303: Hotspot issue

symbios24, 2021-04-25:
Hello
I had a hotspot in my RaspberryPi zero W in dietpi 6.35 until it failed these days so i needed to
install it all over in dietpi 702 but i have not set the country code it says
Unspecified
I have set the channel to 11
But somehow i get half the speed in download that i should get.
If i set a country code will a get the full speed in download? 
I also had a different wifi channel set before but i don't remember which one,is the wifi channel also a reason that i get a slower download speed ?
Thanks

MichaIng, 2021-04-25:
The WiFi country code is most importantly a legal case. WiFi devices are not allowed to send on any radio frequency but only on those which are explicitly allowed in your country. But also your WiFi clients have that country code coded/hardcoded, so won't send on any unallowed frequency as long as no custom firmware is used. If non-matching or only partly matching country codes are used on client and hotspot, the limited shared frequencies might theoretically also limit transfer speed.

But why don't you simply try it out?

WiFi channels are only frequency bands inside of the allowed ranges. To know which channel to use, best is to run some WiFi scanner to check which channel is used how much in your neighbourhood and choose the least used channel. Often routers have such a tool included, else mobile phone apps exist, also Windows/macOS/Linux tools, I guess.

Joulinar, 2021-04-25:
`inSSIDer` is quite a good tool. It will give an overview which channel would be best to use and some other recommendation how to improve the wifi.

symbios24, 2021-04-26:
Thanks
Just tried inSSIDer it recommends channel 11 the one i have, while in some other hotspots recommends a different channel.

MichaIng, 2021-05-07:
I'm marking this as closed. Feel free to reopen, if required.

closed 2021-05-07 by MichaIng