MichaIng/DietPi #4304: DietPi Issue updating APT

rugbycubsf, 2021-04-25:
#### Details:
- Date           | Sun Apr 25 12:53:44 PDT 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DenonPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Get:2 https://packages.sury.org/php stretch InRelease [6824 B]
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Err:2 https://packages.sury.org/php stretch InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Fetched 6824 B in 1s (4834 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Failed to fetch https://packages.sury.org/php/dists/stretch/InRelease  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
E: lzma_read: Read error (10)
E: The package lists or status file could not be parsed or opened.
```

ravenclaw900, 2021-04-25:
Hello,
Please see #4219.  For a fix, run
```
apt-key del '95BD4743' # delete the old key
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg # download the new key
apt update # update again the package cache
```

rugbycubsf, 2021-04-25:
Thanks. Will give it a try.

rugbycubsf, 2021-04-25:
root@DenonPi:~# apt-key del '95BD4743' # delete the old key
OK
root@DenonPi:~# curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg 
root@DenonPi:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://packages.sury.org/php stretch InRelease
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Error!
E: lzma_read: Read error (10)
E: The package lists or status file could not be parsed or opened.

Joulinar, 2021-04-25:
pls try to clear cach as well

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

rugbycubsf, 2021-04-25:
Worked! thanks. I am a NOOB so thanks for super quick response!!!

closed 2021-04-25 by rugbycubsf