MichaIng/DietPi #4306: alsactl -g store fails when setting sound card

sturbs, 2021-04-25:
### Creating a bug report/issue
#### Details:
- Date           | Sun Apr 25 19:37:04 BST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3) | Allo Boss | Allo Isolator v.1.2 
- - Power supply used | Allo Shanti LPS with one output to the Rpi and one to the Isolator v1.2 which is feeding Boss DAC
- Kernel version | Linux DietPi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | alsactl -g store
- Exit code      | 22
- Software title | DietPi-Set_hardware
#### Steps to reproduce:

1. open Dietpi-config and select audio to set sound card.  
2. Ive gotten the same result whether setting the sound card to 'AUTO' or 'Allo-boss-dac-pcm512x-audio'
3. I get the error message that the command 'alsactl -g store' failed with the following error: alsactl: get_control:256: Cannot read control '3,0,0,Playback Channel Map,0': Invalid argument
4. I get the same error if I just ssh into the box and enter 'alasactl -g store'
#### Expected behaviour:

No fail message
#### Actual behaviour:

- /boot/config.txt has the correct dtoverlay appended at the end.  I'm fairly new to alsa so I'm not sure exactly what to look for
#### Extra details:

- # alsactl restore and # alsactl nrestore both complete with the following 
alsactl: load_state:1683: Cannot open /var/lib/alsa/asound.state for reading: No such file or directory
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
Found hardware: "BossDAC" "" "" "" ""
Hardware is initialized using a generic method

--but if I run 'alsactl store' after those, I get the same error about not reading control and invalid argument--
#### Additional logs:
```
alsactl: get_control:256: Cannot read control '3,0,0,Playback Channel Map,0': Invalid argument
```

MichaIng, 2021-04-26:
Many thanks for your report.

I though I had already worked around that issue for the upcoming release, but didn't yet: #4210

You can ignore it (just hit exit, all important steps have been done already). A reboot should solve it, i.e. afterwards `alsactl store` should work, as it has to do with a discrepancy between fully loaded audio devices and the already present tunables in /sys /proc. Maybe also the `vc4-hdmi` device (coming from the KMS driver) play a role. Or that the DAC is already detected but the device tree overlay (driver) not yet loaded, as this requires a reboot.

Audio was disabled before, i.e. the ALSA packages were freshly installed when you selected the card first? This would mean that onboard audio is not the culprit. Before you reboot, could you paste the following:
```
aplay -l
amixer scontrols
```
The second command should show where "Playback Channel Map" belongs to.

In the end, I think we simply need to allow alsactl failing here (silently), at least when a sound card was selected that depends on a loaded device tree overlay, which cannot be reliably done without reboot. That means that any mixer/control states that are applied cannot be saved, but there are none applied in those cases anyway.

MichaIng, 2021-04-26:
Workaround applied: https://github.com/MichaIng/DietPi/commit/7cfc338337313e4b0639a4b39b7cbd903d464bdc
This requires further testing, to know in which circumstances/why this fails and if storing controls fails for all sound cards or only for one while for the others control are stored.

closed 2021-04-29 by MichaIng

sturbs, 2021-04-29:
root@DietPi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 [MAI PCM vc4-hdmi-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: BossDAC [BossDAC], device 0: Boss DAC HiFi [Master] pcm512x-hifi-0 [Boss DAC HiFi [Master] pcm512x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@DietPi:~# amixer scontrols
Simple mixer control 'DSP Program',0
Simple mixer control 'Analogue',0
Simple mixer control 'Analogue Playback Boost',0
Simple mixer control 'Auto Mute',0
Simple mixer control 'Auto Mute Mono',0
Simple mixer control 'Auto Mute Time Left',0
Simple mixer control 'Auto Mute Time Right',0
Simple mixer control 'Clock Missing Period',0
Simple mixer control 'Deemphasis',0
Simple mixer control 'Digital',0
Simple mixer control 'Max Overclock DAC',0
Simple mixer control 'Max Overclock DSP',0
Simple mixer control 'Max Overclock PLL',0
Simple mixer control 'Volume Ramp Down Emergency Rate',0
Simple mixer control 'Volume Ramp Down Emergency Step',0
Simple mixer control 'Volume Ramp Down Rate',0
Simple mixer control 'Volume Ramp Down Step',0
Simple mixer control 'Volume Ramp Up Rate',0
Simple mixer control 'Volume Ramp Up Step',0
root@DietPi:~#

sturbs, 2021-04-29:
Sorry for the delay in my reply.  I should have more time to devote to my setup this weekend and will report back with more details if necessary.

Thanks so much for your help!!

MichaIng, 2021-04-30:
Okay, so `Playback Channel Map` is not there, hence it's not part of the BossDAC, which should be the default sound card. Let's check the VC4 card then:
```
amixer -c 0 scontrols
```