MichaIng/DietPi #4307: Bug in MOTD for DietPI beta v7.1.1 on Raspberry Pi OS Bullseye

ChurchOfNoise, 2021-04-26:
### Creating a bug report/issue

#### Required Information
- DietPi version |
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=1
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'
- Distro version | bullseye/sid
- Kernel version | Linux DietPi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux
- SBC model | RPi Zero W (armv6l)
- Power supply used | RPI power supply
- SDcard used | SanDisk ultra 32GB

#### Steps to reproduce
1. Update DietPi from v7.0 to beta (on Raspberry Pi OS Buster, master)
2. Update Raspberry Pi OS to Bullseye (using the commands as shared by @MichaIng here [https://github.com/MichaIng/DietPi/issues/3009](url)

#### Expected behaviour
on ssh login, you should see:
 ─────────────────────────────────────────────────────
 DietPi v7.1.1 (beta) : 17:14 - Mon 04/26/21
 ─────────────────────────────────────────────────────
 - Device model : RPi Zero W (armv6l)
 - LAN IP : REMOVED
 - Freespace (RootFS) : 26G
 - MOTD : Spring in northern hemisphere: Do you know the tulip command?
          Run "ss -tulp" to see which process listens on which TCP/UDP port.
 ─────────────────────────────────────────────────────

#### Actual behaviour
 ─────────────────────────────────────────────────────
 DietPi v7.1.1 (beta) : 17:16 - Mon 04/26/21
 ─────────────────────────────────────────────────────
 - Device model : RPi Zero W (armv6l)
 - LAN IP : REMOVED
 - Freespace (RootFS) : 26G
 - MOTD : **Open Beta v7.1, please help testing the upcoming release:
          https://github.com/MichaIng/DietPi/issues/4294**
 ─────────────────────────────────────────────────────

ravenclaw900, 2021-04-26:
Hello,
That's just the new MOTD, I don't see the problem.  It's generic and downloaded from the website, so it doesn't know if you're already running the beta or not.

Joulinar, 2021-04-26:
@ChurchOfNoise 
Pls can you be more specific what the issue is from your point of view

ChurchOfNoise, 2021-04-26:
> That's just the new MOTD

So that explains it, probably the change in MOTD occurred close to the moment I upgraded.
I thought I had broken something, my bad...

@Joulinar don't have an issue with it, just thought it wasn't correct...

closed 2021-04-26 by ChurchOfNoise

Joulinar, 2021-04-26:
Usually MPTD is updated once a day and if I'm not mistaken on reboot.

MichaIng, 2021-04-26:
> and if I'm not mistaken on reboot.

Jep, it's stored in /run/dietpi/ tmpfs, hence lost on reboot.

ChurchOfNoise, 2021-04-27:
Interesting!! 

I noticed the change in MOTD probably because the MOTD hadn't changed since installing DietPi... And here I read it should change daily? (hence motD of course). It doesn't for me.

So, on the other RPi 0W, I still have 
 ─────────────────────────────────────────────────────
 DietPi v7.1.1 (beta) : 09:38 - Tue 04/27/21
 ─────────────────────────────────────────────────────
 - Device model : RPi Zero W (armv6l)
 - LAN IP : 192.168.1.2 (eth0)
 - Freespace (RootFS) : 26G
 - MOTD : Spring in northern hemisphere: Do you know the tulip command?
          Run "ss -tulp" to see which process listens on which TCP/UDP port.
 ─────────────────────────────────────────────────────

(was a fresh install which was then updated to the Beta branch, and also on the master branch this never changed)

Joulinar, 2021-04-27:
It did not change daily. The update runs once a day and if there is a new one, it will be updated. But we are not releasing new MOTD each day 😉