MichaIng/DietPi #4311: DietPi - Pi Zero W - Crash USB Hub connection

tbabik, 2021-04-27:
Hello,

I would be thankful for some tipps to debug this issue I'm encountering.

Setup:
- DietPi v7.0.2
- Pi Zero W
- newly setup, only installed packages are Python, VLC, some Python Packages
- Zero powered via PC USB Port; USB meter shows about 100mA on idle

Issue:
- Connecting any device to the USB port causes a crash / restart

I've tried two Micro USB hubs; even without any devices attached to the hubs; as soon as I plug them in the Zero restarts
I've tried a USB OTG adapter connected to a regular USB 2.0 hub; as soon as I plugin the hub to the OTG adapter it crashes
I've tried the USB OTG adapter and a USB RFID card reader; crash
USB OTG adapter and USB soundcard; crash

Once the Zero comes up after such a crash the USB hub + devices are recognized successful. The USB soundcard plays sound. The RFID can read cards.

I've tried a "journalctl -f" and "dmesg -wH" while plugging the USB devices in, but nothing was captured.

What can else can I try? Are there any persistent logs that I could check? Any ideas what the issue might be? As said I'm good if I just leave the hub + devices connected afterwards, the crash / restart only happens on the initial plugin, but would like to know why.

Thank You

Joulinar, 2021-04-27:
Hi,

it's just an idea. Do you have a spare SD card and could try running plain Raspberry OS and see how it behave if you plug in the USB hub?

tbabik, 2021-04-27:
Good idea! Have prepared another MicroSD card with Raspberry OS Lite, only done SSH + WPA Supplicant config.
Same issue - I plug the hub in and it does a restart.

Tell me should I get lost with my obviously not-DietPi related issue :)

Actually .. I've been sitting here now for a few minutes and repeated the process of plugging in and out a couple of times, trying both Hub A and Hub B and I think I'm seeing mixed results. Seems that Hub B suddenly does not cause a crash anymore, while Hub A still does, but also not always. From like 10 tries Hub A did on one occassion not cause a crash.

Quite strange, maybe I'm actually looking at a hardware bug.

Will switch to DietPi once again and repeat .. but maybe I should not stress the connector even further and be happy that the device works once connected and restarted....

Joulinar, 2021-04-27:
hmm not sure if that could be an issue but the hub maybe cause an issue on power while getting connected, causing the device to reboot?

tbabik, 2021-04-27:
I have a similar feeling.

Connecting Hub A still continues to cause a boot.
Connecting Hub B now works, all the time it seems, but connecting something to it causes then to boot.
Connecting the OTG adapter works also all the time, but connecting then a single device to the adapter causes again a boot.

Seems as soon as I attach a power needing device.

Anyhow, that seems to be difficult to debug. Thank You for the suggestion trying another OS, that definitily rules out DietPi.
My final setup foresees me powering the Zero via the 5V GPIO pin(s), maybe that changes something on the behaviour. Should it do I'll update the post.

Joulinar, 2021-04-27:
or is there an option to use a powered USB hub? It's always recommended to have external devices externally powered.

tbabik, 2021-04-27:
Yes, that seems to make a difference. Using my powered USB 3.0 hub connected via OTG adapter brought these results.
10 tries (plugging in and out) while it was externally powered - not a single crash
1st try without power - crash

So powered it must be for solid results. Still a bit strange that a single USB2 hub without any devices attached causes such issues. On the other hand, who knows if the follow standard or if I just by luck have two different bad behaving hubs.

Thanks for the help! I'll try to get a powered Micro USB hub (or open mine and solder external powerlines to it).

MichaIng, 2021-04-27:
Sounds very much like a powering issue indeed. While it's still running, `vcgencmd get_throttled` can be used to check if any head of power induced CPU throttling has been done or is done currently.

Also input PSU + cable is something to have a look at. A bad cable can have a massive voltage drop, and cheap PSU, respectively such which are meant for charging instead of powering a device, can have voltage fluctuations. Otherwise I'd also wonder how a hub only can crash the system 🤔.

tbabik, 2021-04-29:
I think we can close this issue. Here my final conclusion.

I got a bit further with my project and I'm powering the Zero now with a 5V/5A (25Watt) power supply via the GPIO pins. The restart / crash still occurs when using a non-powered hub (and only the hub, no device yet connected).

Not sure what I was searching for in Google before I posted here, perhaps I tuned my search terms a bit, anyhow I'm now finding similar reports online about this behavior. Seems the keyword is "inrush current" and from what I've read only the Zero and the old RPi models are affected, just in case somebody else experiences similar issues (i.e. https://www.raspberrypi.org/forums/viewtopic.php?t=25803)

From my point of view we can close this thread. Once I'm finished with my project all the stuff comes into a closed box and thus I'll not hotplug any devices. Thus I can ignore this for now.

Joulinar, 2021-04-29:
Thx for sharing. I will close this one for now.

closed 2021-04-29 by Joulinar