MichaIng/DietPi #4317: New commands (dietpi-ddns, dietpi-vpn) do not start from commandline

maartenlangeveld, 2021-04-29:
New commands (dietpi-ddns, dietpi-vpn) do not start from commandline are not recognised.
They do start via dietpi-launcher menu.

```
root@M-DNS:/# dietpi-ddns
-bash: dietpi-ddns: command not found
```

```
root@M-DNS:/# dietpi-vpn
-bash: dietpi-vpn: command not found
```

Joulinar, 2021-04-29:
Did you reboot your system after update?

maartenlangeveld, 2021-04-29:
No... will try

MichaIng, 2021-04-29:
Alternatively `. /etc/bashrc.d/dietpi.bash` to load the new aliases.

Joulinar, 2021-04-29:
Or logout/login

Joulinar, 2021-04-29:
Is it working now?

maartenlangeveld, 2021-04-29:
Yes! Thanks for the great work.
I have one question. Using pivpn up until now as openvpn server. Does dietpi-vpn openvpn server as well (or only client?)

MichaIng, 2021-04-29:
It's a client only so far. Maybe we can extend it to configure a server as well by times, adding a client config/revocation UI. But first of all we'll extend the client functionality, adding more providers, free port choice, server rotation and such 😃.

Joulinar, 2021-04-29:
Going to close this as the issue is fixed

closed 2021-04-29 by Joulinar