MichaIng/DietPi #4318: NextCloud install stop

Speeedfire, 2021-04-29:
Hi,
i wanna install nextcloud, but stop here.
Odroid xu4, the latest dietpi.

[image]

Joulinar, 2021-04-29:
Hi,

that's quite strange point to stop on. I did a test an it was working without issues

```
[  OK  ] DietPi-Software | systemctl restart mariadb
[  OK  ] DietPi-Software | systemctl restart redis-server
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/nextcloud_data
[  OK  ] DietPi-Software | chown -R www-data:www-data /var/www/nextcloud /mnt/dietpi_userdata/nextcloud_data
[  OK  ] DietPi-Software | mysql -e grant all privileges on *.* to tmp_root@localhost identified by 'wkmOzNRH9q8YkVnK0hc0QLkZ79eqgD' with grant option
[  OK  ] DietPi-Software | Nextcloud ncc install
```

Next step after `chown` would be creating the database. Maybe you can open a 2nd SSH session once stopped and check `htop` if there is something running still. As well to check `journalctl -n 50` for further error messages.

BTW: no need to do screen prints. You can simply copy the content from SSH session into GitHub 😃

Speeedfire, 2021-04-30:
This is the output:

`Apr 30 08:06:46 DietPi redis-server[12509]: 12509:C 30 Apr 2021 08:06:46.527 # Configuration loaded
Apr 30 08:06:46 DietPi redis-server[12509]: oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
Apr 30 08:06:46 DietPi redis-server[12509]: Redis version=5.0.3, bits=32, commit=00000000, modif
ied=0, pid=12509, just started
Apr 30 08:06:46 DietPi redis-server[12509]: Configuration loaded
Apr 30 08:06:46 DietPi systemd[1]: redis-server.service: Can't open PID file /run/redis/redis-server.pid (yet?) after st
art: No such file or directory
Apr 30 08:06:46 DietPi redis[12516]: Warning: 32 bit instance detected but no memory limit set. 
Setting 3 GB maxmemory limit with 'noeviction' policy now.
Apr 30 08:06:46 DietPi redis[12516]: WARNING: The TCP backlog setting of 511 cannot be enforced 
because /proc/sys/net/core/somaxconn is set to the lower value of 128.
Apr 30 08:06:46 DietPi redis[12516]: Server initialized
Apr 30 08:06:46 DietPi redis[12516]: WARNING overcommit_memory is set to 0! Background save may 
fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or
 run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
Apr 30 08:06:46 DietPi systemd[1]: Started Advanced key-value store.
Apr 30 08:06:46 DietPi /etc/mysql/debian-start[12499]: /usr/bin/mysql_upgrade: the '--basedir' o
ption is always ignored
Apr 30 08:06:46 DietPi /etc/mysql/debian-start[12499]: Looking for 'mysql' as: /usr/bin/mysql
Apr 30 08:06:46 DietPi /etc/mysql/debian-start[12499]: Looking for 'mysqlcheck' as: /usr/bin/mys
qlcheck-
Apr 30 08:06:46 DietPi /etc/mysql/debian-start[12499]: This installation of MySQL is already upg
raded to 10.3.27-MariaDB, use --force if you still need to run mysql_upgrade
Apr 30 08:06:46 DietPi /etc/mysql/debian-start[12535]: Checking for insecure root accounts.
Apr 30 08:06:46 DietPi /etc/mysql/debian-start[12539]: Triggering myisam-recover for all MyISAM tables and aria-recover 
for all Aria tables
Apr 30 08:07:19 DietPi sudo[12675]:   dietpi : TTY=pts/1 ; PWD=/home/dietpi ; USER=root ; COMMAN
D=/bin/journalctl -n 50
Apr 30 08:07:19 DietPi sudo[12675]: pam_unix(sudo:session): session opened for user root by dietpi(uid=0)
`

MichaIng, 2021-04-30:
Was this on DietPi v7.1 already?

For the database creation, we now use a true random password instead of the global software password now. This is received from the systems `/dev/random`, which guarantees that true randomness, but is a blocking source, means it can fail or hang if the systems entropy pool is empty (compared to `/dev/urandom` which would then produce pseudo-random characters). It's just a small number of bytes, while on RPi's the `rng-tools` daemon should assure sufficient entropy such such task, other package installs require way more. However it's the only thing I could imaging that hangs, which would indicate an issue with entropy generation.

Check the following:
```
tr -dc '[:alnum:]' < /dev/random | head -c30
cat /proc/sys/kernel/random/entropy_avail
systemctl status rngd
```
The first should not hang, the second should output a number >3072 (`/proc/sys/kernel/random/write_wakeup_threshold`), the third one should show the entropy daemon being active.

_I was reading a bit more, and for this temporary database admin password `/dev/urandom` should be pretty fine, which cannot block. Without being certain about which random source is secure enough, I went the strict way, but it's an overkill here. On the other hand, if `/dev/random` really blocks, then other tasks (out of our control) will hang too, so maybe good to hang at that stage where no database writes have been done yet so that the install can be repeated without issues 🤔._

Speeedfire, 2021-04-30:
The frist i need stop, no respone after 60sec.

```
root@DietPi:~# tr -dc '[:alnum:]' < /dev/random | head -c30
^C
root@DietPi:~# cat /proc/sys/kernel/random/entropy_avail
39
root@DietPi:~# systemctl status rngd
● rngd.service - Start entropy gathering daemon (rngd)
   Loaded: loaded (/lib/systemd/system/rngd.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2021-04-30 08:47:01 BST; 4h 34min ago
     Docs: man:rngd(8)
  Process: 324 ExecStart=/usr/sbin/rngd -f (code=exited, status=0/SUCCESS)
 Main PID: 324 (code=exited, status=0/SUCCESS)

Apr 30 08:47:01 DietPi systemd[1]: Started Start entropy gathering daemon (rngd).
Apr 30 08:47:01 DietPi rngd[324]: read error
Apr 30 08:47:01 DietPi rngd[324]: read error
Apr 30 08:47:01 DietPi rngd[324]: No entropy sources working, exiting rngd
Apr 30 08:47:01 DietPi systemd[1]: rngd.service: Succeeded.
root@DietPi:~# 
```

Speeedfire, 2021-04-30:
But this is the urandom output:

```
root@DietPi:~# tr -dc '[:alnum:]' < /dev/urandom | head -c30
SiJ5cHOm587WqplkfjvRmByqS8Bq6a
root@DietPi:~# 
```

MichaIng, 2021-04-30:
Okay great, at least we found the issue. Strange, I added the Odroid XU4 to the list of SBCs where rng-tools5 has been proven to work.

Does this file exists?
```
ls -l /dev/hwrng
```

And can you show the loaded kernel modules:
```
lsmod
```

And probably related kernel messages:
```
dmesg | grep random
```

Speeedfire, 2021-04-30:
The output: 
```

root@DietPi:~# ls -l /dev/hwrng
crw------- 1 root root 10, 183 Apr 30 08:46 /dev/hwrng
root@DietPi:~# lsmod
Module                  Size  Used by
tun                    32768  2
sd_mod                 49152  2
sg                     32768  0
spidev                 20480  0
evdev                  24576  1
uas                    20480  1
usb_storage            49152  1 uas
scsi_mod              135168  4 sd_mod,usb_storage,uas,sg
spi_s3c64xx            20480  0
w1_gpio                16384  0
gpio_keys              20480  0
wire                   24576  1 w1_gpio
uio_pdrv_genirq        16384  0
extcon_usb_gpio        16384  0
uio                    20480  1 uio_pdrv_genirq
exynos_gpiomem         16384  0
ip_tables              24576  0
ipv6                  376832  42
root@DietPi:~# dmesg | grep random
[    0.000000] random: get_random_bytes called from start_kernel+0x90/0x430 with crng_init=0
[    0.783742] random: fast init done
[    7.321930] random: systemd: uninitialized urandom read (16 bytes read)
[    7.385831] random: systemd: uninitialized urandom read (16 bytes read)
[    7.415979] random: systemd: uninitialized urandom read (16 bytes read)
[    9.062489] random: crng init done
[    9.062493] random: 7 urandom warning(s) missed due to ratelimiting
root@DietPi:~# 
```

Speeedfire, 2021-04-30:
The rng-tools and rng-tools-debian is fail

```
root@DietPi:~# systemctl start rng-tools
Job for rng-tools.service failed because the control process exited with error code.
See "systemctl status rng-tools.service" and "journalctl -xe" for details.
root@DietPi:~# systemctl status rng-tools-debian
● rng-tools-debian.service - LSB: rng-tools (Debian variant)
   Loaded: loaded (/etc/init.d/rng-tools-debian; generated)
   Active: failed (Result: exit-code) since Fri 2021-04-30 19:54:23 B
ST; 41s ago
     Docs: man:systemd-sysv-generator(8)

Apr 30 19:54:23 DietPi systemd[1]: Starting LSB: rng-tools (Debian variant)...
Apr 30 19:54:23 DietPi rng-tools-debian[5758]: error reading from entropy source
:: No such device
Apr 30 19:54:23 DietPi systemd[1]: rng-tools-debian.serv
ice: Control process exited, code=exited, status=1/FAILURE
Apr 30 19:54:23 DietPi systemd[1]: rng-tools-debian.serv
ice: Failed with result 'exit-code'.
Apr 30 19:54:23 DietPi systemd[1]: Failed to start LSB:
rng-tools (Debian variant).
root@DietPi:~# systemctl start rng-tools-debian
Job for rng-tools-debian.service failed because the control process exited with error code.
See "systemctl status rng-tools-debian.service" and "journalctl -xe" for details.
root@DietPi:~# systemctl status rng-tools-debian
● rng-tools-debian.service - LSB: rng-tools (Debian variant)
   Loaded: loaded (/etc/init.d/rng-tools-debian; generated)
   Active: failed (Result: exit-code) since Fri 2021-04-30 19:55:10 B
ST; 2s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 5795 ExecStart=/etc/init.d/rng-tools-debian start (code=exite
d, status=1/FAILURE)

Apr 30 19:55:10 DietPi systemd[1]: Starting LSB: rng-tools (Debian variant)...
Apr 30 19:55:10 DietPi rng-tools-debian[5795]: error reading from entropy source
:: No such device
Apr 30 19:55:10 DietPi systemd[1]: rng-tools-debian.serv
ice: Control process exited, code=exited, status=1/FAILURE
Apr 30 19:55:10 DietPi systemd[1]: rng-tools-debian.serv
ice: Failed with result 'exit-code'.
Apr 30 19:55:10 DietPi systemd[1]: Failed to start LSB:
rng-tools (Debian variant).
root@DietPi:~#

root@DietPi:~# systemctl status haveged
● haveged.service - Entropy daemon using the HAVEGE algorithm
   Loaded: loaded (/lib/systemd/system/haveged.service; enabled; vendor preset:
enabled)
   Active: active (running) since Fri 2021-04-30 19:56:43 BST; 33s ag
o
     Docs: man:haveged(8)
           http://www.issihosts.com/haveged/
 Main PID: 6088 (haveged)
   Memory: 3.1M
   CGroup: /system.slice/haveged.service
           └─6088 /usr/sbin/haveged --Foreground --verbose=1 -w 1024

Apr 30 19:56:43 DietPi systemd[1]: Started Entropy daemon using the HAVEGE algor
ithm.
Apr 30 19:56:47 DietPi haveged[6088]: haveged: ver: 1.9.1; arch: generic; vend:
; build: (gcc 8.3.0 CTV); collect: 128K
Apr 30 19:56:47 DietPi haveged[6088]: haveged: cpu: (VC); data: 16K (D); inst: 1
6K (D); idx: 12/40; sz: 15012/57848
Apr 30 19:56:47 DietPi haveged[6088]: haveged: tot tests(BA8): A:1/1 B:1/1 conti
nuous tests(B):  last entropy estimate 8.00222
Apr 30 19:56:47 DietPi haveged[6088]: haveged: fills: 0, generated: 0
root@DietPi:~#

```

Speeedfire, 2021-04-30:
```
dietpi@DietPi:~$ tr -dc '[:alnum:]' < /dev/random | head -c30
C1bybViiMXURCoZ7fn3ypqMW4YjJ9sdietpi@DietPi:~$
dietpi@DietPi:~$ cat /proc/sys/kernel/random/entropy_avail
2430
dietpi@DietPi:~$ cat /proc/sys/kernel/random/write_wakeup_threshold
1024
dietpi@DietPi:~$

```

Speeedfire, 2021-04-30:
I deleted all.

```
root@DietPi:~# apt purge --autoremove rng-tools-debian rng-tools rng-tools5
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'rng-tools' is not installed, so not removed
Package 'rng-tools-debian' is not installed, so not removed
Package 'rng-tools5' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi:~#
```

Speeedfire, 2021-04-30:
Yes, thank you guys. 
Nextcloud is running now. :)

Joulinar, 2021-05-02:
another user on the forum https://dietpi.com/phpbb/viewtopic.php?f=11&t=8969

MichaIng, 2021-05-02:
New XU4 images will ship with haveged: https://github.com/MichaIng/DietPi/commit/e95fe08561a89cc925947cbbe322d3b61e801755
With DietPi v7.2, the change is applied to all XU4 systems: https://github.com/MichaIng/DietPi/commit/b67a932b985c73c22e034abc504365670c9a4e17
Changelog: https://github.com/MichaIng/DietPi/commit/36dd6477d974a4a19b154593b270f610dd0983fb

closed 2021-05-18 by MichaIng