MichaIng/DietPi #4319: update v6.34.3 >>> don't works

miarchimi, 2021-04-29:
#### Details:
- Date           | Thu Apr 29 21:54:40 CEST 2021
- Bug report     | e0875196-5ed4-4d8d-934a-99dff373d5e3
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-defaults/php-common_81+0~20210223.34+debian9~1.gbpf52eb0_all.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-apcu/php-apcu_5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-apcu/php7.3-apcu_5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

ravenclaw900, 2021-04-29:
See #4219 for a fix.
```
apt-key del '95BD4743' # delete the old key
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg # download the new key
apt update # update again the package cache
```

MichaIng, 2021-04-29:
That was been done as part of the pre-patches already, actually :thinking:. Looks like indeed clearing the cache is required in some cases.

So if with repeating the above the issue is not solved, please do:
```
/boot/dietpi/func/dietpi-set_software apt-cache clean
```
Then repeat the update.

miarchimi, 2021-04-30:
> Hello,

Thanks a lot for help, did work. Have a nice weekend. Best greetings Michel :-)

Joulinar, 2021-04-30:
Going to close this issue. Feel free to reopen if needed.

closed 2021-04-30 by Joulinar