MichaIng/DietPi #4320: Dietpi-DDNS - change of menu wording

danmoo117, 2021-04-29:
Hi guys, DuckDNS seems to be working nicely with the new dietpi-ddns feature so thanks for adding this in, however one of the menu pages could cause confusion as it asks for a "password" when it actually wants confirmation of the duckdns token. See screenshots.

This is the first duckdns token page which is fine.
[image]

However when progressing this is the token confirmation page but it asks for a password. This confused me for a minute so it will likely confuse others.
[image]

Thanks.

Joulinar, 2021-04-29:
Hi,

yes this is known and can be changed for the time being https://github.com/MichaIng/DietPi/pull/4299#issuecomment-826379864

danmoo117, 2021-04-29:
No worries, didn't see it logged anywhere so thought I'd raise it. 👍🏻

Joulinar, 2021-04-29:
no problem, that's totally fine. If you are ok I'm going to close this one as it's working as designed at the moment. Maybe it will change in future.

@MichaIng 
or do you like to keep it open as reminder 😉

MichaIng, 2021-04-30:
To not make it too complicated, what do you think about making the wording more generic, like:
```
Please confirm your input:
```
or
```
Please enter the input again:
```
or "value" instead of "input"?

MichaIng, 2021-05-01:
Changed it now to these hardcoded but generic messages around the password box:
```
[FAILED] No input made, please try again...
Please retype and confirm your input:
[FAILED] Inputs do not match, please try again...
```

closed 2021-05-18 by MichaIng