MichaIng/DietPi #4321: After update 7.1.2 my Rasperry Pi 3 won't boot

Arghh, 2021-04-29:
### Creating a bug report/issue
After the update what did seems to be go fine I rebooted my Rasperry Pi 3 and now it won't start. Just black. I'm sorry but I have no other useful information. 
Anyone has any tips how can I debug it? Get it working again. 

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model |  (RPi3)
- Power supply used | ( 5V 1A RAVpower)
- SDcard used | (SanDisk )

MichaIng, 2021-04-30:
Many thanks for your report.

Does SSH access work, or can you even ping the Pi?

And what was the version you upgraded from, or did you regularly upgrade your APT packages? Package upgrades and especially kernel upgrades can be a trigger for apparent filesystem corruption e.g.

And for debugging, if SSH access is not possible either, do you have another Linux/UNIX system where you can plug the SD card and access the ext4 root partition?
On Windows you can at least mount the boot partition, check the content (bootloader, kernel and so) and run a filesystem check as well.

Arghh, 2021-04-30:
When I try to access the FAT-partion of the SD-Card my explorer crashes. Need to remove the SD-Card for it to come back.
I managed to get the dietpi-update.log. The boot and preboot are not there.

Yes I have Ramlog enabled and var/log ist empty.

The update-log seems to be fine. I'm closing the issue as it is not related to dietpi update. I got a new SD-Card running and had backup so not much harm done. Just wanted to report the issue, if it was, in fact, updated related.

```
[0m
[J
 [38;5;154mDietPi-Update[0m
[90m─────────────────────────────────────────────────────
 Phase:[0m Applying pre-patches

[0m[0m
[J[90m[[0m      [90m][0m [90mDietPi-Update |[0m Downloading pre-patches[0m[2G[1;33m.     [0m[C[0m
[J[90m[[0m[32m  OK  [90m][0m [90mDietPi-Update |[0m Downloading pre-patches
[0m[0m
[J[90m[[0m      [90m][0m [90mDietPi-Update |[0m Applying execute permission[0m[2G[1;33m.     [0m[C[0m
[J[90m[[0m[32m  OK  [90m][0m [90mDietPi-Update |[0m Applying execute permission
[0m[0m
[J[90m[[0m[32m  OK  [90m][0m [90mDietPi-Update |[0m Successfully applied pre-patches
[0m[0m
[J
 [38;5;154mDietPi-Update[0m
[90m─────────────────────────────────────────────────────
 Phase:[0m Upgrading APT packages

[0m[0m
[J[90m[[0m INFO [90m][0m [90mDietPi-Update |[0m [90m[0mAPT update, please wait...
[0m[90mGet:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [32.9 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 https://archive.raspberrypi.org/debian buster/main armhf Packages [372 kB]
Fetched 13.4 MB in 29s (465 kB/s)
Reading package lists...
[0m[0m
[J[90m[[0m[32m  OK  [90m][0m [90mDietPi-Update |[0m [0mAPT update
[0m[0m
[J[90m[[0m INFO [90m][0m [90mDietPi-Update |[0m [90m[0mAPT upgrade, please wait...
[0m[90mdebconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
(Reading database ... 60%
(Reading database ... 65%
(Reading database ... 70%
(Reading database ... 75%
(Reading database ... 80%
(Reading database ... 85%
(Reading database ... 90%
(Reading database ... 95%
(Reading database ... 100%
(Reading database ... 27177 files and directories currently installed.)
Preparing to unpack .../libapt-pkg5.0_1.8.2.3_armhf.deb ...
Unpacking libapt-pkg5.0:armhf (1.8.2.3) over (1.8.2.2) ...
Setting up libapt-pkg5.0:armhf (1.8.2.3) ...
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
[... 59 lines cut ...]
[J[90m[[0m[32m  OK  [90m][0m verify_dietpi.txt  | Completed
[0m[0m
[J
 [38;5;154mDietPi-Update[0m
[90m─────────────────────────────────────────────────────
 Phase:[0m Applying incremental patches

[0m[0m
[J[90m[[0m INFO [90m][0m [90mDietPi-Update |[0m [90mCurrent version : v7.0.2
[0m[0m
[J[90m[[0m INFO [90m][0m [90mDietPi-Update |[0m [90mLatest version  : v7.1.2
[0m[0m
[J[90m[[0m      [90m][0m [90mDietPi-Patch |[0m rm /boot/dietpi/misc/dietpi-nordvpn[0m[2G[1;33m.     [0m[C[0m
[J[90m[[0m[32m  OK  [90m][0m [90mDietPi-Patch |[0m rm /boot/dietpi/misc/dietpi-nordvpn
[0m[0m
[J[90m[[0m      [90m][0m [90mDietPi-Patch |[0m sed -Ei /^[[:blank:]]*aSOFTWARE_INSTALL_STATE\[(67|171)\]=/d /boot/dietpi/.installed[0m[2G[1;33m.     [0m[C[0m
[J[90m[[0m[32m  OK  [90m][0m [90mDietPi-Patch |[0m sed -Ei /^[[:blank:]]*aSOFTWARE_INSTALL_STATE\[(67|171)\]=/d /boot/dietpi/.installed
[0m[0m
[J[90m[[0m INFO [90m][0m [90mDietPi-Update |[0m [90m[0mAPT autopurge, please wait...
[0m[90m[0m[0m
[J[90m[[0m[32m  OK  [90m][0m [90mDietPi-Update |[0m [0mAPT autopurge
[0m[0m
[J[90m[[0m[32m  OK  [90m][0m [90mDietPi-Update |[0m Incremental patching to v7.1.2 completed
[0m
```

closed 2021-04-30 by Arghh

Joulinar, 2021-04-30:
yes indeed looks like the SD card died. If you have another system running, you could try to connect the faulty SD card using an USB connector and perform some file systems check

Arghh, 2021-05-01:
FYI: I installed the 7.1.2 update on my 2 other PI 3-s running DietPi without any problems. Similar configuration and software. No problems.