MichaIng/DietPi #4322: Rock Pi 4: X Higher Res than 1080p60 does not work on HDMI Port

Keridos, 2021-04-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version` 7.1.2
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version` 10.9
- Kernel version | `uname -a` 5.10.21-rockchip64
- SBC model | `echo $G_HW_MODEL_NAME` Rock Pi 4
- Power supply used | (EG: 5V 1A RAVpower) USB Hub with external Power supply 
- SDcard used | (EG: SanDisk ultra) SanDisk Ultra 32GB A1 U/V1

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) Xserver/xrandr
- Was the software title installed freshly or updated/migrated? freshly installed
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1.  Install DietPI on the RockPi 4
2.a  Install XFCE, start with lightdm with a user with video,audio,render permissions
2.b Or just use autostart with dietpi user and not LightDM
3. connect a monitor/tv with more than 1080p resolution.
4. Try to set higher resolution via cvt/xrandr

```
$ cvt 2560 1440 60
# 2560x1440 59.96 Hz (CVT 3.69M9) hsync: 89.52 kHz; pclk: 312.25 MHz
Modeline "2560x1440_60.00"  312.25  2560 2752 3024 3488  1440 1443 1448 1493 -hsync +vsync
$ xrandr --newmode "wqhd60" 312.25  2560 2752 3024 3488  1440 1443 1448 1493 -hsync +vsync
$ xrandr --addmode HDMI-1 "wqhd60"
$ xrandr --output HDMI-1 --mode "wqhd60"
xrandr: Configure crtc 0 failed

```
#### Expected behaviour

- Either the highest resolution supported by the display should work out of the box or xrandr should not fail.

#### Actual behaviour

- The highest reported resolution for my 2 displays I tested (4k Samsung Smart TV Q82R and WQHD(1440p) Asus VG27AQL1A) was 1080p60. xrandr failed with the following message: "Configure crtc 0 failed"

MichaIng, 2021-05-01:
Many thanks for your report.

I cannot see an obvious reason either, no device tree overlay or boot configuration is available to affect this.

Does `xrandr --verbose ...` flag give any additional helpful output? Also did you try the desktops GUI display settings, or is that what you mean with "highest reported resolution"?

Probably the Rockchip Mali driver package causes some issues (which was actually meant for Stretch images). Can you try the following:
```
apt purge libmali-rk-midgard-t86x-r14p0
rm /etc/X11/xorg.conf
rm /usr/lib/aarch64-linux-gnu/lib{EGL,GLES}*
apt install --reinstall libegl1 libgles2
```

Keridos, 2021-05-01:
Logs: https://gist.github.com/Keridos/8b83154e8a5eafe725227a435b32cc68

> Does `xrandr --verbose ...` flag give any additional helpful output? Also did you try the desktops GUI display settings, or is that what you mean with "highest reported resolution"?

Maybe, see the gist I linked. It does not work in the desktops GUI settings either. showS  only up to 1080p, until I --addmode with xrandr. Screen flickers when I try to set mode to new mode but resets to 1080p afterwards. 

> Probably the Rockchip Mali driver package causes some issues (which was actually meant for Stretch images). Can you try the following:
> 
> ```
> apt purge libmali-rk-midgard-t86x-r14p0
> rm /etc/X11/xorg.conf
> rm /usr/lib/aarch64-linux-gnu/lib{EGL,GLES}*
> apt install --reinstall libegl1 libgles2
> ```

This did not change anything regarding the outputs as far as I could tell.

MichaIng, 2021-05-02:
Okay, I'm out of ideas then. Would be interesting if this is a general issue of specific to your monitor where communicating supported >1080p resolutions seems to not work.

I had a look into the Armbian forum, but cannot find a related issue: https://forum.armbian.com/forum/33-rockchip-3399/

If anyone has more ideas or a ROCK Pi 4 to verify that this is a general issue, would be great. Any other RK3399-based SBC would be good as well, as they all share the same kernel.

MichaIng, 2021-11-11:
Is this still an issue with the latest kernel?

MichaIng, 2021-12-08:
I'll mark this as closed. Needs to be verified with latest image, Debian Bullseye and latest kernel.

closed 2021-12-08 by MichaIng