MichaIng/DietPi #4323: DietPi-Software | CouchPotato: Evaluating development status

MichaIng, 2021-05-01:
CouchPotato currently requires Python 2 and pip to install additional required modules. This makes it difficult to keep up support with Debian Bullseye, where Python 2 has mostly been removed. And of course it is officially deprecated, pip prints large warnings etc.

Our implementation is quite old, runs as root user, installs to /etc and a few other things which require alignment with current implementation and security standards we aim for. But if it won't run anymore on Bullseye anyway, it's not reasonable to invest much further efforts.

CouchPotato forum is locked since 2018, issues disabled on the GitHub repo, which shows still sporadic commits at least. I emailed the primary maintainer to check back about Python 3 support plans and the development status in general.

In comparison to the only concurrent Radarr, CouchPotato btw runs exceptionally good, requires much less memory, faster and more fluent web interface, is my impression. I think it's Python vs Mono, which makes a notable difference. But I haven't compared features and up-to-date downloader + tracker support, where Radarr is surely better suited, being developed very actively.

Roughly ~600 active users through DietPi, based on DietPi-Survey: https://dietpi.com/survey/
Not overwhelming but not bad either. Radarr has ~5 times more, I guess mostly a result of popularity in combination with Sonarr and probably not so much based on actual software quality.

It's about whether to keep or drop (and when) support for CouchPotato, so any input and opinions on this matter are highly welcome.

Joulinar, 2021-05-02:
hmm personally I'm no in favour to support outdated and not maintained software. As well it still require Python 2 which is not maintained as well. Sooner or later this might introduce security risk as the system will run outdated software.

MichaIng, 2021-05-02:
Last commit a bit more than a year ago: https://github.com/CouchPotato/CouchPotatoServer
PRs are however constantly created, interest in the project seems to be there, with reason from what I can say after trying it out. But yes, as long as there is no drastic change in development rate expected, I'm also for dropping support instead of wasting time on a zombie 😄.

MichaIng, 2021-09-25:
The repository has just been official archived, so the project is dead: https://github.com/CouchPotato/CouchPotatoServer
We gonna remove it with next release.