MichaIng/DietPi #4327: Webmin restarting not working properly

Keridos, 2021-05-02:
#### Required Information
- DietPi version | `cat /boot/dietpi/.version` 7.1.2
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version` 10.9
- Kernel version | `uname -a` 5.10.21-rockchip64
- SBC model | `echo $G_HW_MODEL_NAME` Rock Pi 4
- Power supply used | (EG: 5V 1A RAVpower) USB Hub with external Power supply and tried official Rock Pi 4 power adapter
- SDcard used | (EG: SanDisk ultra) SanDisk Ultra 32GB A1 U/V1

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)Webmin
- Was the software title installed freshly or updated/migrated? freshly installed
- Can this issue be replicated on a fresh installation of DietPi? dunno

#### Steps to reproduce

1.  Install DietPI on the RockPi 4
2. Install Webmin
3. do any config setting within webmin that causes a restart

#### Expected behaviour

- The service should just restart

#### Actual behaviour

- Seems it is restarting properly but stopped after that. Something like 
- `systemctl restart webmin && systemctl stop webmin`

Joulinar, 2021-05-02:
Hi,

This is a know behaviour of webmin. You can have a look to this forum post for a workaround https://dietpi.com/phpbb/viewtopic.php?p=33550#p33550

MichaIng, 2021-05-02:
Overdue issue. I hope there is a specific exit code or signal to track when a regular restart is performed, as I'm no fan of blind "Restart=always" with the loop it implies, spamming the logs in case of actual failures.

MichaIng, 2021-05-02:
Found a better method: `GuessMainPID=no`

My original guess was wrong: The systemd service's main process does not run in foreground, so that it would exit once that foreground process does. The underlying perl script vanishes into background internally, hence we do and must run the service in forking mode. Newer systemd versions (at least since Buster) by default try to guess the main process then. Once this main process exits, systemd assumes the whole service (control group) do be exited and kills remaining child processes.

The above changes the default so that systemd does not guess and use a main process anymore, and instead sees the whole service active as long as any process of the control group still runs. The internal restart is done via child process that forks a new main webmin instance, hence this works reliable. However, some testers would be good to be sure.

Keridos, 2021-05-03:
> Found a better method: `GuessMainPID=no`
> 
Works for me!

MichaIng, 2021-05-03:
Many thanks for testing.

> And it is much quicker than this workaround.

Mostly because the main service is changed instead of a drop-in config created 🙂. I prefer it especially since I'm no fan of automated restart loops. In very most cases, when a process crashes of fails to start in the first place, retrying it e.g. 5 times won't make it succeed but might make underlying issues even worse and harden debugging in case. So where possible, I try to avoid them, or make them react only to specific exit codes or signals. But here neither exit code nor signal can be tracked, since the webmin perl script runs in background (relative to the wrapping systemd service).

closed 2021-05-29 by MichaIng