MichaIng/DietPi #4328: DietPi-Software | Bitwarden_RS reinstall fails

Hannlora, 2021-05-02:
#### Details:
- Date           | Sun May  2 09:08:18 CEST 2021
- DietPi version | v7.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux HomePi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | mv bitwarden_rs-1.21.0 /opt/bitwarden_rs
- Exit code      | 1
- Software title | DietPi-Software
- 
#### Steps to reproduce:
1. sudo dietpi-software reinstall 183

#### Expected behaviour:
- Bitwarden_RS is updated to the latest version

#### Actual behaviour:
- reinstall fails and a backup had to be restored

#### Additional logs:
```
mv: cannot stat 'bitwarden_rs-1.21.0': No such file or directory
```

#### Extra details:
- I can only guess, but maybe it have something to do with the renaming to vaultwarden https://github.com/MichaIng/DietPi/issues/4325

Joulinar, 2021-05-02:
Probably, will do some testing.

MichaIng, 2021-05-02:
It's definitely related as the name of directories and binaries has changed. Our "Bitwarden_RS" install option is broken by this for sure 😉. I'm marking this issue as closed, let's track the case is the linked issue.

closed 2021-05-02 by MichaIng