MichaIng/DietPi #4330: sonarr no connection with rTorrent after reinstall

bbsixzz, 2021-05-02:
I can no longer make a connection to rTorrent after reinstalling to take advantage of the new improvements.

What settings should be used?

[image]

`- DietPi-Software | rTorrent: For performance reasons, and to avoid conflicts with Shairport Sync, the default SCGI/RPC access has been changed from TCP port 5000 to UNIX socket. This change is applied on fresh installs and reinstalls of rTorrent, as it requires a change on both all sides, rTorrent, ruTorrent and the webserver. HTTP authentication is now enforced for the rTorrent RPC access with all webservers, using the global software password by default, and the password hash algorithm has been hardened with Nginx.`

https://wiki.servarr.com/Sonarr_Settings#Download_Clients

```
rTorrent
Requires rTorrent version 0.9.0 or newer compiled with XML-RPC support, and a SCGI to HTTP proxy (usually Lighttpd, Nginx, or Apache; ruTorrent can also provide this on on the URL plugins/rpc/rpc.php). Look for ruTorrent instructions for how to set up rTorrent.
Website: rTorrent
```

Joulinar, 2021-05-02:
can you share full output of

`ss -tulpn | grep LISTEN`

bbsixzz, 2021-05-02:
```
tcp     LISTEN   0        1024             0.0.0.0:80             0.0.0.0:*      users:(("lighttpd",pid=8885,fd=4))                                             
tcp     LISTEN   0        500              0.0.0.0:8084           0.0.0.0:*      users:(("mono",pid=319,fd=5))                                                  
tcp     LISTEN   0        128              0.0.0.0:22             0.0.0.0:*      users:(("sshd",pid=743,fd=3))                                                  
tcp     LISTEN   0        500              0.0.0.0:8989           0.0.0.0:*      users:(("mono",pid=8907,fd=8))                                                 
tcp     LISTEN   0        1024                [::]:80                [::]:*      users:(("lighttpd",pid=8885,fd=5))                                             
tcp     LISTEN   0        128                    *:21                   *:*      users:(("proftpd",pid=8858,fd=0))                                              
tcp     LISTEN   0        128                 [::]:22                [::]:*      users:(("sshd",pid=743,fd=4))   
```

Joulinar, 2021-05-02:
@bbsixzz 
let's check config file `cat /etc/lighttpd/lighttpd.conf`

@MichaIng 
Question: I see `rtorrent` socket config is added to main `lighttpd.conf` was there a reason for not creating an own config file?

bbsixzz, 2021-05-02:
```
server.modules = (
	"mod_indexfile",
	"mod_access",
	"mod_alias",
 	"mod_redirect",
)

server.dir-listing = "enable"

server.document-root        = "/var/www"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

# strict parsing and normalization of URL for consistency and security
# https://redmine.lighttpd.net/projects/lighttpd/wiki/Server_http-parseoptsDetails
# (might need to explicitly set "url-path-2f-decode" = "disable"
#  if a specific application is encoding URLs inside url-path)
server.http-parseopts = (
  "header-strict"           => "enable",# default
  "host-strict"             => "enable",# default
  "host-normalize"          => "enable",# default
  "url-normalize-unreserved"=> "enable",# recommended highly
  "url-normalize-required"  => "enable",# recommended
  "url-ctrls-reject"        => "enable",# recommended
  "url-path-2f-decode"      => "enable",# recommended highly (unless breaks app)
 #"url-path-2f-reject"      => "enable",
  "url-path-dotseg-remove"  => "enable",# recommended highly (unless breaks app)
 #"url-path-dotseg-reject"  => "enable",
 #"url-query-20-plus"       => "enable",# consistency in query string
)

index-file.names            = ( "index.php", "index.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.conf.pl"
include "/etc/lighttpd/conf-enabled/*.conf"

#server.compat-module-load   = "disable"
server.modules += (
	"mod_compress",
	"mod_dirlisting",
	"mod_staticfile",
)
#RUTORRENT_DIETPI
server.modules += ( "mod_fastcgi" )
server.modules += ( "mod_scgi" )
server.modules += ( "mod_auth" )
auth.debug = 0
auth.backend = "htdigest"
auth.backend.htdigest.userfile = "/etc/.rutorrent-htaccess"

auth.require = ( "/rutorrent/" => (
                    "method"  => "digest",
                    "realm"   => "rtorrent",
                    "require" => "valid-user"
               ))

scgi.server = ( "/RPC2" =>
    ( "127.0.0.1" =>
        (
            "host" => "127.0.0.1",
            "port" => 5000,
            "check-local" => "disable"
        )
    )
)
#RUTORRENT_DIETPI

```

Joulinar, 2021-05-02:
If I'm not mistaken the issue is with the section `scgi.server = ( "/RPC2" =>` as this is still pointing to port `5000` while it's nor running on socket.

You could try to create a backup of `/etc/lighttpd/lighttpd.conf` before playing with it. Once done replace whole section as follow

**remove**

```
#RUTORRENT_DIETPI
> remove all lines
#RUTORRENT_DIETPI
```

**add**
```
#RUTORRENT_DIETPI
server.modules += ( "mod_fastcgi" )
server.modules += ( "mod_scgi" )
server.modules += ( "mod_auth" )
auth.debug = 0
auth.backend = "htdigest"
auth.backend.htdigest.userfile = "/etc/.rutorrent-htaccess"
auth.require = ( "/rutorrent" =>
        (
                "method"  => "digest",
                "realm"   => "rtorrent",
                "require" => "valid-user"
        ),
        "/RPC2" =>
        (
                "method"  => "digest",
                "realm"   => "rtorrent",
                "require" => "valid-user"
        )
)
scgi.server = ( "/RPC2" =>
        ( "rtorrent" =>
                (
                        "socket" => "/mnt/dietpi_userdata/downloads/.session/rpc.socket",
                        "check-local" => "disable"
                )
        )
)
$HTTP["url"] =~ "^/rutorrent/(conf|share)($|/)" {
        url.access-deny = ("")
}
#RUTORRENT_DIETPI
```

restart web server afterwards `systemctl restart lighttpd.servic`

bbsixzz, 2021-05-02:
That works, cheers!

:]

Joulinar, 2021-05-02:
@MichaIng 
how was it intended during `rTorrent` reinstall? It seems web server configuration did not changed

MichaIng, 2021-05-02:
Many thanks for reporting.

The rTorrent section in `lighttpd.conf` is only for historical reasons. Indeed I missed that reinstall to have the section removed and re-added.

Quick fix before reinstalling rTorrent:
```
sed -i '/#RUTORRENT_DIETPI/,/#RUTORRENT_DIETPI/d' /etc/lighttpd/lighttpd.conf
```

Task for next release is to migrate to a dedicated config, hence running the above + writing to `/etc/lighttpd/conf-available/98-dietpi-rtorrent.conf` + `lighty-enable-mod dietpi-rtorrent`.

Joulinar, 2021-05-02:
@MichaIng 
I could try start working on it. At least to rework/change config file creating and config procedure on Lighttpd.

MichaIng, 2021-05-02:
The would be awesome.

MichaIng, 2021-05-18:
Done with: #4332

closed 2021-05-18 by MichaIng