MichaIng/DietPi #4333: Update from 7.0.2 to 7.1.2 fails with TvHeadend

m0nk3h, 2021-05-02:
#### Details:
- Date           | Sun May  2 22:48:41 BST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero (armv6l) (ID=1)
- Kernel version | Linux DietPi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ... sudo dietpi-update
2. ... with TvHeadend installed, the install stops with error below
#### Expected behaviour:

- ... v7.0.2 should upgrade to 7.1.2
#### Actual behaviour:

- ... fails during TvHeadend update
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Err:2 https://dl.bintray.com/tvheadend/deb jessie InRelease
  403  Forbidden [IP: 3.125.168.216 443]
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
E: Failed to fetch https://dl.bintray.com/tvheadend/deb/dists/jessie/InRelease  403  Forbidden [IP: 3.125.168.216 443]
E: The repository 'https://dl.bintray.com/tvheadend/deb jessie InRelease' is no longer signed.
```

Joulinar, 2021-05-02:
Hi,

did you installed TvHeadend from `dietpi-software` ?? Because I don't find it in our software list. As well you are using Debian Jessie repository on a Buster system wich is not recommended as Jessie went out of support in 2018. In meantime TvHeadend is available vie global raspberrypi.org archive

MichaIng, 2021-05-02:
If I'm not mistaken, bintray.com stopped its APT service completely a while ago, so I wonder how it was still possible to update to DietPi v7.0 with it? Search machines show that https://apt.tvheadend.org/ seems to exist or did exist as alternative, but currently Cloudflare throws a 522 (host error).

So yes, currently using the package from the Raspberry Pi repository seems to be the solution 👍.

m0nk3h, 2021-05-02:
Uhmm....  I cannot remember how I installed TvHeadend tbh.....  I set the whole Pi up on DietPi in a day, probably via instructions on the above site which is now giving me a 522 error too.  Please feel free to close this issue, and I'll re-install TvHeadend correctly.  Sorry!

Joulinar, 2021-05-02:
going to close this one. Feel free to reopen if needed

closed 2021-05-02 by Joulinar

MichaIng, 2021-05-03:
No problem, btw. I just had the idea to add TvHeadend backend + Kodi addon to dietpi-software or at least our docs, as coincidentally I yesterday looped through a Kodi PVR addon request. Please report back when you found working install instructions.

To track website status, earlier or later an issue will raise on their GitHub repos, I guess: https://github.com/tvheadend