MichaIng/DietPi #4335: Docker fails to start after update

eyduh, 2021-05-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | `
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `buster`
- Kernel version | `Linux  4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64 GNU/Linux
`
- SBC model | `Odroid HC-4`
- Power supply used | stock
- SDcard used | boot from SSD in one toaster slot, HDD in other

#### Additional Information (if applicable)
- Docker
- Updated to dietpi 7 and did an apt upgrade in the same process

- 6e17f214-f48c-45de-80dd-961f3f5f10c7

#### Steps to reproduce

1. Update and install docker 
2. start docker,
3. error message related to /etc/docker/daemon.json

#### Expected behaviour

- Docker starts

#### Actual behaviour

- docker can't understand daemon.json file

#### Extra details

- I tried uninstalling docker and starting it which worked. Then I restored backup from before update and it complained about the daemon.json file's formatting again.
- I have docker, docker-compose and portainer installed on the system.

Error message:
` unable to configure the Docker daemon with file /etc/docker/daemon.json: invalid character '{' looking for beginning of object key string`

Daemon.json:
```{
    "data-root": "/mnt/dietpi_userdata/docker-data",
    "log-driver": "journald",
    "log-level": "warn",
    "debug": "false",
    {    "features": { "buildkit": true }} 

}```

I've tried running the .json file through various online json validators to find which part is formatted incorrectly but even if the validator says its formatted correctly it doesnt work.

Joulinar, 2021-05-03:
Hi,

Many thanks for your message. For me the file looks like this 

```
{
    "data-root": "/mnt/dietpi_userdata/docker-data",
    "log-driver": "journald",
    "log-level": "warn",
    "debug": false
}
```

Maybe you can save your current file and create a new one

eyduh, 2021-05-03:
> {
>     "data-root": "/mnt/dietpi_userdata/docker-data",
>     "log-driver": "journald",
>     "log-level": "warn",
>     "debug": false
> }

This `/etc/docker/daemon.json` file works. I'm not sure docker compose would work without the 
`{ "features": { "buildkit": true } }` part though? With this setup I don't need the build kit at the moment so would consider this closed but it's weird that would break the setup as it's the same way it is referenced on [docker help pages](https://docs.docker.com/develop/develop-images/build_enhancements/)

Joulinar, 2021-05-03:
if I'm not mistaken, you can use `DOCKER_BUILDKIT=1 docker build` still.

closed 2021-05-03 by eyduh

ravenclaw900, 2021-05-03:
I think that the docker help article was referencing a blank `daemon.json` file, so you should also be able to do:
```json
{
    "data-root": "/mnt/dietpi_userdata/docker-data",
    "log-driver": "journald",
    "log-level": "warn",
    "debug": "false",
    "features": { "buildkit": true }
}
```

Joulinar, 2021-05-03:
yes this I was thinking as well @ravenclaw900

eyduh, 2021-05-03:
> ```json
> {
>     "data-root": "/mnt/dietpi_userdata/docker-data",
>     "log-driver": "journald",
>     "log-level": "warn",
>     "debug": "false",
>     "features": { "buildkit": true }
> }
> ```

I tried this in the different adjustments I did but it throws a:
`unable to configure the Docker daemon with file /etc/docker/daemon.json: json: cannot unmarshal string into Go struct field Config.debug of type bool`

Joulinar, 2021-05-03:
maybe something to be ask on Docker side how the config file should looke like ??

eyduh, 2021-05-03:
> maybe something to be ask on Docker side how the config file should looke like ??

I suppose, but I never touched this file prior to the error though so the weirdly formatted build kit environment variable must've gotten in there through DietPi somehow.

Basically I setup a fresh dietpi, only installed docker, docker-compose and portainer through dietpi-software. had it running for a month and did an update recently which broke it.

Joulinar, 2021-05-03:
so you did not add this line manually? Because on my Docker installation it is not ther and DietPi is not going to add such a line

eyduh, 2021-05-03:
Not that I can remember, perhaps some docker-compose I was trying out through portainer inserted it? 
I was trying out some stuff from linuxserver.io, also built nextcloudpi from scratch as the latest release had a nextcloud version mismatch. It's possible some option during one of these things inserted it I guess? I dunno, if it works now and it wasn't related to DietPi then all g. If it shows up again I'll know try and see what I did immediately prior.