MichaIng/DietPi #4338: Update to DietPi 7.1.2 failed

MTenbult, 2021-05-03:
#### Details:
- Date           | Mon May  3 23:09:20 CEST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | dpkg --remove-architecture armhf
- Exit code      | 2
- Software title | DietPi-Patch
#### Steps to reproduce:
1. Run dietpi-update
#### Expected behaviour:
- Update to 7.1.2
#### Actual behaviour:
- Error related to dpkg
#### Additional logs:
```
dpkg: error: cannot remove architecture 'armhf' currently in use by the database
```

MichaIng, 2021-05-03:
Many thanks for your report.

I just learned that a foreign architecture cannot be removed, if any package of that architecture is currently installed. Can you please paste the result of:
```
dpkg --get-selections | mawk '/:armhf/{print $1}'
```

MichaIng, 2021-05-04:
This fixes it for DietPi v7.2: https://github.com/MichaIng/DietPi/commit/af3270a804fede0b75644a2985bdebd517e50c27
Would still be interesting which 32-bit package has actually been installed.

MTenbult, 2021-05-05:
Sorry for the late response! Running `dpkg --get-selections '*:armhf' | mawk '{print $1}'`
returns:
`gcc-8-base:armhf
libc6:armhf
libgcc1:armhf
wiringpi:armhf`

Seems like most 32-bits package where installed with wiringPi. By purging wiringPi (I don't need it anymore) only leaves
`libc6:armhf`

MichaIng, 2021-05-05:
Great! For next release we updated our WiringPi install option to use a new maintained Git branch, support for new hardware etc: https://github.com/MichaIng/DietPi/pull/4324
It's compiled on your system, so would be arm64 in your case.

C0D3-M4513R, 2021-05-05:
I can reproduce this. Have the exact same issue, with the exact same packages.
Purging wiringpi removes all armhf packages:
```
root@DietPi:/tmp/DietPi-Patch# apt purge wiringpi
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  gcc-8-base:armhf libc6:armhf libgcc1:armhf
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  wiringpi:armhf*```

Joulinar, 2021-05-05:
did you install `wiringpi` yourself before (without `dietpi-software`)?

MichaIng, 2021-05-05:
Good and about time to remove armhf then, to prevent users from potentially installing a parallel 32-bit system 😄. There are still a few other larger armhf-only packages in the RPi repo, e.g. Wolfram Alpha, Kodi and Chromium we already know about: https://archive.raspberrypi.org/debian/pool/main/w/wolfram-engine/

Installing those pulls a smaller or later amount of armhf dependencies, in cases conflicting with the arm64 ones (not all have multi-arch support), so that the option would be then to replace those and potentially remove other arm64 dependants with it. I mean on interactive installs all info and warnings are shown and confirmation required, but I think this should be done well known by the admin, including adding foreign architectures intentionally, when required, not having this as a default.

C0D3-M4513R, 2021-05-05:
> did you install `wiringpi` yourself before (without `dietpi-software`)?

Don't know. I did a dev project, which in I needed the dev files. Shouldn't there then also be a dev wiringpi package? https://github.com/C0D3-M4513R/CD74HCT299

MichaIng, 2021-05-05:
> Shouldn't there then also be a dev wiringpi package?

WiringPi itself is basically a development library. Installing it with their build script as well as the RPi APT package both contain the `gpio` tool, the `libwiringPi` + `libwiringPiDev` (Dev for Device, not Development 😉) libraries and all development headers. Our install option also leaves the source files in place, which contain docs and a few additional examples/libraries one might want to compile or use as a starting point.

closed 2021-05-18 by MichaIng