MichaIng/DietPi #4339: Dietpi-Drive_Manager | NTFS drives are not mounted on reboot

Joulinar, 2021-05-04:
### Creating a bug report/issue
@MichaIng 
can you have a look to `dietpi-drive_manager`. At the moment we have a couple of user reporting issues using NTFS drives as they are not mounted on reboot. I could replicate the behaviour on my test system.

adjusting `/etc/fstab` from
`UUID=xxx /mnt/ntfs fuseblk noatime,lazytime,rw,nofail,noauto,x-systemd.automount`

to
`UUID=xxx /mnt/ntfs ntfs rw,nofail,noauto,x-systemd.automount,user,fmask=133,dmask=022,uid=1000,gid=1000 0 0`

fixed it for one user

**Reference** https://dietpi.com/phpbb/viewtopic.php?f=9&t=8971

MichaIng, 2021-05-04:
Many thanks.

I think I know where this comes from:
```
# lsblk -no FSTYPE /dev/sdb1
ntfs
# blkid -s TYPE -o value /dev/sdb1
ntfs
# findmnt -no FSTYPE /dev/sdb1
fuseblk
# mount | grep /dev/sdb1
/dev/sdb1 on /mnt/0A5EC1981E18D273 type fuseblk (rw,relatime,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096)
# df -T /dev/sdb1
Filesystem     Type    1K-blocks  Used Available Use% Mounted on
/dev/sdb1      fuseblk   1046524  5712   1040812   1% /mnt/0A5EC1981E18D273
# mount -t fuseblk /dev/sdb1 /mnt/0A5EC1981E18D273
mount: /mnt/0A5EC1981E18D273: wrong fs type, bad option, bad superblock on /dev/sdb1, missing codepage or helper program, or other error.
# mount -t ntfs /dev/sdb1 /mnt/0A5EC1981E18D273
#
```
Looks like in case of NTFS, there is a difference between the filesystem that you are telling "mount" to mount and the filesystem reported back afterwards. I changed the drive detection loop so that info about mounted drive is consequently obtained via `findmnt` and info about non-mounted drives via `lsblk`. Previously we had `blkid` as well here and there but it uses some cache, which can cause issues and it wakes up drives every time, even that it would not be required for the single value that we check for, and even the man page recommends to use `lsblk` instead for unmounted drives. But as well I recognised that inside containers, `lsblk` does not detect filesystem, so needed to revert to `blkid` for unmounted drives. Now it seems the same is true for mounted drives 😢.

At least I found a way to disable the problematic cache.

cocoflan, 2021-05-04:
i have the same problem both on ntfs and ext4 disks on a system, have to rename my disks after reboot with my old names, to see them back on my network.

Joulinar, 2021-05-04:
But you have seen them, their are just not mounted on reboot?

MichaIng, 2021-05-04:
> ext4 disks

At least this particular issue does not exist with ext4, which is named "ext4" with all the above tools. Can you show which "name" you exactly changed, or the `/etc/fstab` content in general after using `dietpi-drive_manager`?

ravenclaw900, 2021-05-04:
I've had the same problem as @cocoflan, sometimes it creates a malformed entry in `/etc/fstab`.  It adds newlines where there shouldn't be any.

Joulinar, 2021-05-04:
I guess we need to be careful. There might be more and different challenges. This one was particular for NTFS as it was not auto mount on reboot

BillyCorgan1, 2021-05-05:
> 
> 
> > ext4 disks
> 
> At least this particular issue does not exist with ext4, which is named "ext4" with all the above tools. Can you show which "name" you exactly changed, or the `/etc/fstab` content in general after using `dietpi-drive_manager`?
> 
> **EDIT: Same for @ravenclaw900, as this should not be related to the filesystem type then.**

This issue does exist for EXT4 and for F2FS. I am running an Odroid N2 and have done a clean install. Using **dietpi-drive_manager** I cannot get a drive that is formatted to EXT4 or F2FS to be mounted on reboot. There seems to be something seriously wrong with dietpi-drive_manager at the moment.

Joulinar, 2021-05-05:
Does it mount if you try to access the file system on os level using `cd /mnt/something`

BillyCorgan1, 2021-05-05:
> 
> 
> Whatever symptoms you face guys, the only relevant thing we can work with is:
> 
> ```
> cat /etc/fstab
> ```

I currently cannot get the latest version of DietPi to install on my Odroid N2 at all (I will open a bug report for this as a separate issue.) I will install the old version I got working to double check but when I put in  **cat /etc/fstab** after setting up the drives in drive_manager the only drives showing were my root/boot partition drive. There were no entries for my Micro SD card and USB Flash Drive.

Joulinar, 2021-05-05:
I did a test on 7.1.2 and `ext4` was working without issues. Drive got mounted correctly upon access

```
May 05 20:18:51 DietPi3 systemd[1]: mnt-demo.automount: Got automount request for /mnt/demo, triggered by 534 (bash)
May 05 20:18:51 DietPi3 systemd[1]: Mounting /mnt/demo...
May 05 20:18:51 DietPi3 kernel: EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)
May 05 20:18:51 DietPi3 systemd[1]: Mounted /mnt/demo.
May 05 20:18:53 DietPi3 systemd[1]: systemd-fsckd.service: Succeeded.
```

BillyCorgan1, 2021-05-05:
> 
> 
> Does it mount if you try to access the file system on os level using `cd /mnt/something`

I have made a folder on the drive using the following command:

`mkdir -p /mnt/MicroSD/Plex/Media/Music`

I have the rebooted the system and used the following command:

`cd /mnt/MicroSD/Plex/Media/Music`

I am able to go to the directory without errors.

Joulinar, 2021-05-05:
The drives are not mounted directly on boot. This is as expected an intended. They are mounted automatically upon access. That's why drive manager shows them unmounted on first place if you did not access them before.

Btw it doesn't matter which version you use to flash you SD card initially. On first boot you will be updated to latest version, always.

MichaIng, 2021-05-05:
The `/etc/fstab` look as expected, so mounting on boot, respectively on first access of the mount point, should work fine.

But I have an idea what the issue with `dietpi-drive_manager` here is, not detecting the external drives as intended: As long as nothing is accessing the mount points (`/mnt/MicroSD` and `/mnt/USBFlash`), they are not actually mounted, but waiting for access. But `dietpi-drive_manager` detects only actually mounted drives as such, via `df`, while unmounted drives are shown with what would be the default mount point (with UUID). That was never different, but is a bug indeed, especially since the fstab entries are in turn removed.

One solution could be to loop through all `/mnt/*` directories before scanning the drives, so trigger the automount. `ls -d /mnt/*/. &> /dev/null` or similar.

And the SD card entry looks not as intended:
```
│                                                 ●─ mmcblk1 ─────────────────────────────────────────────── 
│       /mnt/                                     : /dev/mmcblk1 | No filesystem / format required
│       /mnt/eb0d1343-0e53-4834-bc1f-774554a0c04b : /dev/mmcblk1p1 | ext4 | Not mounted
```
`/dev/mmcblk1` should not show up when `/dev/mmcblk1p1` is present. I'll have a look into this.

Joulinar, 2021-05-05:
>  But dietpi-drive_manager detects only actually mounted drives as such, via df, while unmounted drives are shown with what would be the default mount point (with UUID)

Yes this was the case always, usually hitting the refresh button fixed the view. But indeed not what a user might expect.

MichaIng, 2021-05-05:
> usually hitting the refresh button fixed the view.

Ah that is true. I wonder which step actually triggers the mount.

MichaIng, 2021-05-05:
`df -a` itself triggers the mount.

Now I also found a line break in `/etc/fstab`:
```
├─/mnt/external                       systemd-1  autofs     rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=9585
│ └─/mnt/external                     /dev/sdb1  ext4       rw,noatime,lazytime
```
`findmnt` prints a separate line for the automount entry, which of course has no UUID...

MichaIng, 2021-05-05:
This PR solves both issues for me. Would be great if you could test it as well, as the automount timing might not be the same on each system: https://github.com/MichaIng/DietPi/pull/4349/files

MichaIng, 2021-05-06:
Did further testing and mount some other issues which should now be fixed. `lsblk` first of all cannot be used for UUID, PARTUUID or filesystem type, as it won't show anything in containers. It might not be relevant for most users, but it breaks our testing and plans to create container images in general. `blkid` and `lsblk` cannot be used for all mounted drives, since it works only for physical block devices, not for network drives. Luckily I fount two command line options for `findmnt` which in combination assure that the correct single value is printed, to solve the the issue that generally `findmnt` can have multiple matches when automount units are active, for bind mounts etc. These options are now always used.

Furthermore, the script contained a few fragile assumptions, like that drives without UUID are network drives and such with are physical drives. While this is mostly true, I added a doubled check before adding fstab entries, first checking for the `/sys/block/<name>` device handler to identify it as physical block device, then checking again the UUID, else print an error message. Also the detection prints now show whether a drive has been detected as network drive or physical drives. That all together should make things more failsafe and allow better debugging as well.

And indeed when unmounting a drive, removing the mount point directory can fail as long as the automount unit is still active. Sometimes it stops by itself immediately or quickly, but not always, showing the directory as "busy". It is now disabled actively, before attempting to remove the directory.

All these automount issues seem to have been slipped through testing, as they become relevant only after a reboot, while I tested everything intensively only within one session. I'll add now a little automount bind mount to my testing systems, to include those in general testing.

BillyCorgan1, 2021-05-06:
> 
> 
> This PR solves both issues for me. Would be great if you could test it as well, as the automount timing might not be the same on each system: https://github.com/MichaIng/DietPi/pull/4349/files

I will test the pull request as soon as I can, it should be the next day or so. I am using the instructions from this link to enable dev mode:

[](https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md)

I hope this is the correct way to test it, if not could you please give me some pointers so I can test it correctly. I still cannot do a fresh install of DietPi with the DietPi 7.02 image on my OdroidN2 (I will open this as an issue separately), I need to use v6.28.0 to get it to install. It is just going to take a little time but I will feedback as soon as I can after testing. Thanks.

MichaIng, 2021-05-06:
Many thanks. Instead of `dev` branch, the `drive_manager` needs to be used until it has been merged:
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=drive_manager' /boot/dietpi.txt
dietpi-update 1
```

BillyCorgan1, 2021-05-07:
I apologise that I messed up the link in my earlier comment, it was a link to the github page for enabling dev mode in the boot folder. Thanks for the command to use. I have clean reinstalled DietPi and used your command. I apologise for the long post but I thought it was best to use the same method as I used when reporting the bug so you can compare the two if so needed.

```
─────────────────────────────────────────────────────
 DietPi v7.1.2 : 23:49 - Fri 07/05/21
 ─────────────────────────────────────────────────────
 - Device model : Odroid N2 (aarch64)
 - CPU temp : 34'C : 93'F (Cool runnings)
 - LAN IP : 192.168.1.4 (eth0)
 - MOTD : Did you know that you can run "dietpi-banner" to change its content?
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image by        : DietPi Core Team (pre-image: Meveric)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Contribute      : https://dietpi.com/contribute.html
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

```
**cat /etc/fstab** before installation:

```
root@OdroidN2:~# cat /etc/fstab
# Please use "dietpi-drive_manager" to setup mounts
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1856M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAPFILE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
UUID=e139ce78-9841-40fe-8823-96a304a09859 / ext4 noatime,lazytime,rw 0 1
UUID=EC51-067C /boot vfat noatime,lazytime,rw 0 2
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

```
I entered the first command and then ran the second:

```
root@OdroidN2:~# G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=drive_manager' /boot/dietpi.txt
[  OK  ] G_CONFIG_INJECT | Setting in /boot/dietpi.txt adjusted: DEV_GITBRANCH=drive_manager
root@OdroidN2:~# dietpi-update 1

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Getting repository version: https://raw.githubusercontent.com/MichaIng/DietPi/drive_manager/.update/version
[  OK  ] DietPi-Update | Got valid repository version
[  OK  ] DietPi-Update | Update available:
[... 243 lines cut ...]
```
┌───────────────────────────────────────────────┤ DietPi-Drive_Manager ├───────────────────────────────────────────────┐
│ Please select a drive to see available options.                                                                      │
│  - User data location: RootFS (/mnt/dietpi_userdata)                                                                 │
│                                                                                                                      │
│                                     ●─ mmcblk0 ──────────────────────────────────────────────────●                   │
│                   /                 : /dev/mmcblk0p2 | ext4 | Capacity: 28.3G | Used: 866.7M (3%)                    │
│                   /boot             : /dev/mmcblk0p1 | vfat | Capacity: 255.7M | Used: 95.9M (38%)                   │
│                                     ●─ mmcblk1 ──────────────────────────────────────────────────●                   │
│                   /mnt/MicroSD      : /dev/mmcblk1p1 | ext4 | Capacity: 58.4G | Used: 52M (0%)                       │
│                                     ●─ sda ──────────────────────────────────────────────────────●                   │
│                   /mnt/USBFlash     : /dev/sda1 | ext4 | Capacity: 234.3G | Used: 60M (0%)                           │
│                                     ●─ Global Options ───────────────────────────────────────────●                   │
│                   Idle Spindown     : Set a global idle duration, before drives power down                           │
│                                     ●─ Add / Refresh Drives ─────────────────────────────────────●                   │
│                   Add network drive : Select to mount networked drives                                               │
│                   Refresh           : Scan for recently added/removed drives                                         │
│                                                                                                                      │
│                                                                                                                      │
[... cut ...] │
│                                                                                                                      │
└──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```
I hope this has been helpful. I again apologise for the length of the post but I would rather give you too much information than not have enough. Please let me know if you need me to test anything more or if I have made an error. Thanks for your time.

MichaIng, 2021-05-07:
Many thanks, so all works now as expected, right? Also the doubled entry below `mmcblk1` is gone, that is good. I'll merge the pull request into `dev` branch then, so it is used on regular development and testing until next release.

BillyCorgan1, 2021-05-08:
Yes, for me it works. I go into dietpi_drive_manager and everything appears as expected. I am no longer trying to format a drive and thinking it is broken because it kept appearing that it wasn't mounted. I obviously can't speak for anyone else but I think the change has worked. I will try F2FS next and then try mounting them in Samba. I will of course feedback if I get any problems. Thanks again.

MichaIng, 2021-05-08:
Many thanks for testing 👍.

closed 2021-05-18 by MichaIng