MichaIng/DietPi #4346: dietpi-vpn how to uninstall?

maartenlangeveld, 2021-05-05:
I wanted to try dietpi-vpn, but decided not to use it. On first run it installs some packages. How can I uninstall these?

Since I want to keep my SBCs as clean and lightweight as possible, it would help me if I can revert dietpi-vpn.

Thanks !
Maarten

MichaIng, 2021-05-05:
It has a reset option to de-configure and remove any VPN configuration and provider files. But indeed OpenVPN is not uninstalled, might might be used independently. Probably we can ask whether this shall be removed as well when selecting the reset option.
```
apt purge --autoremove openvpn
```

Joulinar, 2021-05-05:
What about to mark OpenVPN install option as installed? This way it can be installed using `dietpi-software`

MichaIng, 2021-05-05:
The install option does a server setup, creates key pairs via easy-rsa, client configs, enables IP forwarding for clients etc, so that is not the same. Probably the naming it "OpenVPN Server" would make sense to avoid confusion. While the APT package is the same, client and server setup is a fundamental difference.

Long-term, I want to have client and server setups both in `dietpi-vpn`, but that's another topic 🙂.

MichaIng, 2021-05-05:
Done with: #4347

maartenlangeveld, 2021-05-05:
Great, thanks !!

closed 2021-05-18 by MichaIng