MichaIng/DietPi #4350: Failed to install Radarr white installing Kodi and Lidarr

Takerman, 2021-05-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | USB into my Router
- SDcard used | SanDisk 32 GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
1. Download and install the version of DietPi 7.1.2
2. Go to the software section and Choose to install: Kodi, Radarr, Lidarr, Bazarr
3. Press the Install button and wait for the installation to finish
4. You will receive an exception while installing

#### Expected behaviour
- Should not show exception and should be installed successfully

#### Actual behaviour
- While installing showed exception

#### Extra details
- The screenshot is attached down
[image]

Joulinar, 2021-05-06:
Hi,

Thx for information. I guess installation still finished? Because I see backup URL was used.

MichaIng, 2021-05-06:
Many thanks, issue fixed for DietPi v7.2. Changelog: https://github.com/MichaIng/DietPi/commit/9d72451b2d16ae372847a22b84c13f615b9de089

closed 2021-05-18 by MichaIng