MichaIng/DietPi #4352: Update netdata to include Cloud support

C0D3-M4513R, 2021-05-06:
Right now the netdata package is downloaded and installed from the dietpi website somewhere:
https://github.com/MichaIng/DietPi/blob/86c5e377c607c2e1d8e97138c7cb9d377120a55d/dietpi/dietpi-software#L5679
The current version, that is installed is 1.12.0 . 
The [current version of netdata is 1.30.1](https://learn.netdata.cloud/docs/agent/changelog/).
That means, that the version of netdata hasn't been updated in over a Year.

The *new* Cloud feature allows one to create composite views with other nodes. 
This can be very useful to pi clusters (not that I have one), and for monitoring multiple pc's at once.

One possible solution would be to simply update the existing package on the website. 

Another possibility would be to automate netdata's kikstarter script?
```shell
bash <(curl -Ss https://my-netdata.io/kickstart.sh)
```

Joulinar, 2021-05-06:
this is correct for Debian Stretch. For Buster system, the global `apt` package is installed

https://github.com/MichaIng/DietPi/blob/86c5e377c607c2e1d8e97138c7cb9d377120a55d/dietpi/dietpi-software#L5688

C0D3-M4513R, 2021-05-06:
Ok, so this would mean, that debian has to approve further versions, or the Installing mechanism would have to change?

Would it be advisable to simply not use dietpi-software for netdata then, if one wanted to have the extra functionality?

MichaIng, 2021-05-06:
Jep, I just expanded the linked code as far as possible where it can be seen that those old versions are only installed for Stretch, where we're not going to invest further time into (it will become oldoldstable in 2-3 months).

Joulinar, 2021-05-06:
@C0D3-M4513R 
you are free to install whatever you want in DietPi. There is no strickt need to use `dietpi-software`. It's an offer for user who might not have that much experience.

C0D3-M4513R, 2021-05-06:
Should I close this then, since this is a upstream issue?

MichaIng, 2021-05-06:
The Debian Buster version is 1.12 as well: https://packages.debian.org/buster/netdata
That won't change, as Debian does not merge major version updates into stable releases.

Bullseye is up to date: https://packages.debian.org/bullseye/netdata

I had a look into the kickstart script, which pulls the GitHub source archive, which contains another installer that is called: https://github.com/netdata/netdata/blob/master/netdata-installer.sh
That is a monster, I'll have a look into it whether we can use it to reliably build Netdata on DietPi.

Also I remember another issue and case where I had a little talk with an unofficial maintainer of newer Netdata packages. I'll have a look.

Also we could request Debian buster-backports for the newer Netdata version shipped with Bullseye.

Joulinar, 2021-05-06:
currently I have running `bash <(curl -Ss https://my-netdata.io/kickstart.sh) --no-updates --stable-channel --disable-telemetry` on my RPi3B+ 32bit

MichaIng, 2021-05-06:
For reference: https://github.com/MichaIng/DietPi/issues/3622
Still x86_64 only: https://packagecloud.io/netdata/netdata/
I still don't like the this is extremely overblown instead of making use of the modular nature of Netdata and offering a slim base package and additional modules via separate packages.

I guess the installer can be configured to add/skip parts?

Joulinar, 2021-05-06:
after around 10 minutes compiling

```
 --- We are done! ---

  ^
  |.-.   .-.   .-.   .-.   .-.   .  netdata                          .-.   .-
  |   '-'   '-'   '-'   '-'   '-'   is installed and running now!  -'   '-'
  +----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+--->

  enjoy real-time performance and health monitoring...

 OK
```

MichaIng, 2021-05-06:
Not that long, that is good. Also it can be tuned a little, using system libraries (instead of compiling and bundling them) and disabling some plugins. But especially this build contains a large number of build dependencies and as always, security patches are not applied automatically (via APT) but one needs to recompile manually.

Let's keep this issue open for discussion. I'll first try getting backports for Debian Buster, if this fails, we can think about switching to a source build, or providing packages ourselves (if easy to accomplish).

MichaIng, 2021-05-27:
Backports request sent: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=989178

Joulinar, 2021-10-10:
I guess this issue can be closed as there will be no `buster-backports` for `netdata` package. In meantime Debian Bullseye has been released as well, shipping version `1.29.3`