MichaIng/DietPi #4353: nextcloud webfinger/nodeinfo 7.1.2

droogi, 2021-05-06:
sorry i cannot code, only report...

my nextcloud 21.0.1 installation on dietpi 7.1.2 says its not well configured because of missing webnode and nodeinfo configuration. (nginx)

i added /etc/nginx/sites-dietpi/dietpi-nextcloud.conf in line 4-5 with 

location ~ ^/\.well-known/(?:webfinger|nodeinfo) {
	return 301 /nextcloud/index.php$request_uri;

as you write in /.conf/dps_114/nginx.nextcloud.conf
after restarting service nginx i still have same error message in nextcloud

(in forum https://dietpi.com/phpbb/viewtopic.php?p=33770#p33770)

when i copy the complete code from dps_114 in a new nginx.nextcloud.conf i get a third safety hint:
Strict-Transport-Security" is not min. 15552000 Seconds, activation of hsts is recommended.

Do you have any advice?

Regards, droogi

Joulinar, 2021-05-06:
did you checked on the nextcloud forum as I suggested on the forum?

MichaIng, 2021-05-06:
Many thanks for your report. The HSTS warning is expected as long as you didn't enable it intentionally via e.g. `dietpi-letsencrypt` or which whichever method you enabled HTTPS. Just note that you should only do this when you don't see a reason to allow plain HTTP connections any longer for any of your clients, as they will store the information and won't connect via HTTP any longer (which is the goal of HSTS).

The webfinger and nodeinfo warnings should actually be solved. I'll have a look.

droogi, 2021-05-06:
@joulinar, i tried to understand the nextcloud documentation for nginx (thats not the forum), but i get mixed up since dietpi uses not the same way how to configure nginx - sites enabled. I do not understand the syntax  and do not have knowledge and experience to figure out and just tried copy and paste in hopefully the correct file.

MichaIng, 2021-05-06:
Just in case: "uncomment" means to remove the leading hash `#` sign from:
```
#add_header Strict-Transport-Security "max-age=15768000; includeSubDomains" always;
```
So it changes from a config code comment into an actual directive.

While the config has it commented by default, to work with plain HTTP setups or with HTTPS but no HSTS, we uncomment it when running `dietpi-letsencrypt` with HSTS option enabled.

droogi, 2021-05-06:
@MichaIng, when using the dps_114 original complete code i was able to uncomment this line, but still have the webfinger/nodeinfo issue. thanks for explanation. (_offtopic: i just renewed letsencrypt certificate, but i had to open port 80  on router which is usually not forwarded, so no automatic renewals possible?_)

Joulinar, 2021-05-06:
`letsencrypt` requires port 80 to be open to obtain the certificate

MichaIng, 2021-05-07:
Yes. With HTTPS redirect enabled there is also no reason to keep port 80 closed. Read the statement of Let's Encrypt about this: https://letsencrypt.org/docs/allow-port-80/

MichaIng, 2021-05-07:
Strange, I cannot replicate the issue. Installed Nextcloud + Nginx on a clean DietPi and the redirects solve the warnings as expected.

droogi, 2021-05-07:
when i run the the command i get this result:
[webfingertest.log](https://github.com/MichaIng/DietPi/files/6444180/webfingertest.log)
when i change to curl -IL https://your.domain.org/nextcloud/.well-known/webfinger/test the second result:
[webfingertest2.log](https://github.com/MichaIng/DietPi/files/6444182/webfingertest2.log)

MichaIng, 2021-05-07:
So the redirect is done as intended. Of course the final access fails, as it you cannot authenticate via curl, only the first redirect is relevant.

Strange that the Nextcloud admin panel shows it otherwise. Did you try to restart the webserver and PHP and clear the browser cache as well to rule that all out?
```
systemctl restart nginx
systemctl restart php7.3-fpm
```
And `CTRL+F5` to reload the page in browser.

droogi, 2021-05-07:
i always restarted nginx, (also restarted dietpi),  now i also restarted php7.3-fpm, cleared cache in firefox and chrome, reloaded page, still get the warning.
So you think its an nextcloud warning error?

MichaIng, 2021-06-02:
Is this still an issue?

droogi, 2021-06-02:
I updated to 21.0.2 stable about 2 weeks ago, nextcloud still displays webfinger / nodeinfo error message.
I updated dietpi from 7.1.2 to 7.2.3 today, nextcloud still displays webfinger / nodeinfo error message.

Since i am using Nextcloud only private for me and my family, and the installation and connected devices work well, its no problem for me. I do not have any disadvantages when the message is displayed. 

If you were interested to solve, i could give you an account/ssh, but i am fine when its not solved.
Thanks for your support.

MichaIng, 2021-09-01:
Looping through old open issues. Another attempt: In `/var/www/nextcloud/config/config.php` you have
```php
'overwrite.cli.url' => 'http://localhost/nextcloud',
```
right?

I never though about that a HTTPS redirect means this is redirected to `https://localhost/nextcloud`, which fails as `localhost` is no valid domain for the certificate. So this needs to be changed to contain `https://your.domain.org/nextcloud` for CLI calls to succeed, which likely implies the webfinger/nodeinfo calls.

I'll add this to `dietpi-letsencrypt`, to update ownCloud and Nextcloud configs regarding this setting.

droogi, 2021-09-02:
dietpi 7.5.2 & nextcloud 22.1.1
I changed <code>http://localhost/nextcloud </code> (http) to <code>https://mydomain.org/nextcloud</code>(http**[s]
[diepi-nextcloud.zip](https://github.com/MichaIng/DietPi/files/7096709/diepi-nextcloud.zip)
**) in <code>/var/www/nextcloud/config/config.php</code>

The warning in nextcloud is still there.

I attached my current <code>/etc/nginx/sites-dietpi/dietpi-nextcloud.conf</code>. Line 2-4 is not included in line 8<code>location ^~ /nextcloud {</code> block.

(in windows a system32/config/host has to be configured to redirect the domain to localhost, where is this done here?)

MichaIng, 2021-09-02:
Did you restart PHP to assure possible caches are cleared?
```sh
systemctl restart php7.3-fpm
```

The config is as expected:
- The redirect is from `https://mydomain.org/.well-known/webfinger` to `https://mydomain.org/nextcloud/index.php/.well-known/webfinger`, so within the `/nextcloud` block the rule would never apply 😉.
- `location ~ ^/.well-known` syntax means a location matching the "regular expression" (`~`), which starts (`^`) with `/.well-known`.
- `location ^~ /nextcloud` syntax means a location which starts "literally" with `/nextcloud` and that no further location directive should be searched for (`^~`).

MichaIng, 2021-09-02:
Okay I tested it the other way round, and indeed `overwrite.cli.url` doesn't have an effect on those tests but only on the Cron job and `occ` (`ncc`) command.

Another thing you could try is to update the `.htaccess to rule out any possibly old conflicting rewrite:
```sh
ncc maintenance:update:htaccess
```

droogi, 2021-09-05:
Thanks for the detailed explanation. I updated the `.htaccess, without success.
It seems I am the only one with this config message in nextcloud, so for me it's ok to archive/close this topic. 

Next big aim for me it's to upgrade to 64 bit to get OpenOffice in nextcloud (when wireguard works) and import all certificates.

MichaIng, 2021-09-05:
Okay, let me keep it open to have the `overwrite.cli.url` change implemented into `dietpi-letsencrypt`, which is required to allow Cron jobs and occ commands accessing Nextcloud via HTTPS.

MichaIng, 2021-09-09:
Done: https://github.com/MichaIng/DietPi/commit/9909a4f750c1ed50719d74b3dbc995fc4abee992

closed 2021-09-12 by MichaIng