MichaIng/DietPi #4356: dietpi-update script fails

Vladimiromacedo, 2021-05-07:
#### Details:
- Date           | Fri May  7 23:41:15 BST 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. run dietpi-update
2. ...
#### Expected behaviour:

- dietpi should upgrade
#### Actual behaviour:

- update script times out while trying to run apt-get -qq upgrade and exits to error screen
#### Extra details:

- ...
#### Additional logs:
```
debconf: delaying package configuration, since apt-utils is not installed
dpkg: unrecoverable fatal error, aborting:
 unknown system group 'smmsp' in statoverride file; the system group got removed
before the override, which is most probably a packaging bug, to recover you
can remove the override manually with dpkg-statoverride
E: Sub-process /usr/bin/dpkg returned an error code (2)
```

MichaIng, 2021-05-07:
Many thanks for your report. This is a bug in the sendmail-bin Debian package, which has been solved for Bullseye already, but the fix has not been merged into Buster 😞: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=939885

The following should fix it:
```
sed -i '/smmsp/d' /var/lib/dpkg/statoverride
apt upgrade
```

Vladimiromacedo, 2021-05-08:
Workaround worked perfectly. 
Many thanks!

closed 2021-05-08 by Vladimiromacedo