MichaIng/DietPi #4358: Update available not

deckspace, 2021-05-08:
#### Details:
- Date           | Sat May  8 15:19:28 BST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux pi4diet 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. dietpi-update
2. ...
#### Expected behaviour:

- ... update?
#### Actual behaviour:

- ...
- we see the quoted message;  ─────────────────────────────────────────────────────
 DietPi v7.0.2 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (aarch64)
 - CPU temp : 36'C : 96'F (Cool runnings)
 - LAN IP : 192.168.1.126 (eth0)
 - MOTD : Did you know that you can run "dietpi-banner" to change its content?
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image           : DietPi Core Team (pre-image: Raspberry Pi OS 64-bit)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Donate          : https://dietpi.com/#donate
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-update   : Run now to update DietPi from v7.0.2 to v7.1.2.

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@pi4diet:~# dietpi update
-bash: dietpi: command not found
root@pi4diet:~# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Getting repository version: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid repository version
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v7.0.2
[ INFO ] DietPi-Update | Latest version  : v7.1.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=50778 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | ncc maintenance:mode --on
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : rpimonitor
[  OK  ] DietPi-Services | stop : motioneye
[  OK  ] DietPi-Services | stop : mineos
[  OK  ] DietPi-Services | stop : mympd
[  OK  ] DietPi-Services | stop : darkice
[  OK  ] DietPi-Services | stop : icecast2
[  OK  ] DietPi-Services | stop : shairport-sync
[  OK  ] DietPi-Services | stop : mpd
[  OK  ] DietPi-Services | stop : coturn
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : proftpd
[  OK  ] DietPi-Services | stop : avahi-daemon

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian buster InRelease
Get:2 https://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Get:4 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Hit:5 https://archive.raspberrypi.org/debian buster InRelease
Fetched 98.6 kB in 5s (18.3 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
E: Packages were downgraded and -y was used without --allow-downgrades.
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade
 - 
#### Extra details:

- ...
#### Additional logs:
```
E: Packages were downgraded and -y was used without --allow-downgrades.
```

Joulinar, 2021-05-08:
Can you try running`apt update && apt upgrade` and post the full output

deckspace, 2021-05-08:
same result when attempting dietpi-update on pi3b+ 

root@berry3diet:~# apt update
Hit:1 https://download.docker.com/linux/debian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Hit:3 https://deb.debian.org/debian buster InRelease
Get:4 https://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Hit:5 https://deb.debian.org/debian-security buster/updates InRelease
Get:6 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Fetched 98.6 kB in 3s (36.1 kB/s)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
root@berry3diet:~# apt update && apt upgrade
Hit:1 https://download.docker.com/linux/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Hit:4 https://deb.debian.org/debian buster-updates InRelease
Hit:5 https://deb.debian.org/debian-security buster/updates InRelease
Hit:6 https://deb.debian.org/debian buster-backports InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

deckspace, 2021-05-08:
Ok the pi3B+ returns ; - Command: dpkg --remove-architecture armhf                                      │
│  - Exit code: 2                                                                   │
│  - DietPi version: v7.0.2 (MichaIng/master) | HW_MODEL: 3 | HW_ARCH: 3 | DISTRO:  │
│ 5                                                                                 │
│  - Image creator: DietPi Core Team                                                │
│  - Pre-image: Raspberry Pi OS 64-bit                                              │
│  - Error log:                                                                     │
│                                                                                   │
│        Retry          : Re-run the last command that failed                
 
and the Pi4 returns
│ APT upgrade                                                                  │
│  - Command: apt-get -qq upgrade                                              │
│  - Exit code: 100                                                            │
│  - DietPi version: v7.0.2 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 |     │
│ DISTRO: 5                                                                    │
│  - Image creator: DietPi Core Team                                           │
│  - Pre-image: Raspberry Pi OS 64-bit                                         │
│  - Error log:                                                                │
│ E: Packages were downgraded and -y was used without --allow-downgrades.      │
│                                                                              │
│     Retry          : Re-run the last command that failed

Joulinar, 2021-05-08:
You have 2 device? Both have different issues. Can you post `apt update && apt upgrade` for the RPi4 and post the output as requested above. 

Do you have installed `wiringpi` on the RPi3?

MichaIng, 2021-05-08:
To check which package(s) block the armhf removal on RPi3B+:
```
dpkg --get-selections '*:armhf' | mawk '{print $1}'
```
Underlying issue fixed btw: #4338

C0D3-M4513R, 2021-05-09:
I had the same issue. For me it was `cmake` and some other cmake variant. 
Those packages were installing the SAME version. The downgrade consisted of installing the same version, that was already installed.
Had to wipe my pi install, cause no tty suddenly spawned.

deckspace, 2021-05-09:
everything was going so well..!  this from the Pi4

root@pi4diet:~# apt update && apt upgrade
Hit:1 https://deb.debian.org/debian buster InRelease
Get:2 https://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Hit:4 https://deb.debian.org/debian-security buster/updates InRelease
Get:5 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Fetched 98.6 kB in 1s (66.3 kB/s) 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be DOWNGRADED:
  cmake cmake-data
0 upgraded, 0 newly installed, 2 downgraded, 0 to remove and 0 not upgraded.
Need to get 4644 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://archive.raspberrypi.org/debian buster/main arm64 cmake-data all 3.16.3-3~bpo10+1 [1628 kB]
Get:2 https://archive.raspberrypi.org/debian buster/main arm64 cmake arm64 3.16.3-3~bpo10+1 [3016 kB]
Fetched 4644 kB in 1s (5776 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 52778 files and directories currently installed.)
Preparing to unpack .../cmake-data_3.16.3-3~bpo10+1_all.deb ...
Unpacking cmake-data (3.16.3-3~bpo10+1) over (3.16.3-3~bpo10+1) ...
Preparing to unpack .../cmake_3.16.3-3~bpo10+1_arm64.deb ...
Unpacking cmake (3.16.3-3~bpo10+1) over (3.16.3-3~bpo10+1) ...
Setting up cmake-data (3.16.3-3~bpo10+1) ...
Setting up cmake (3.16.3-3~bpo10+1) ...

deckspace, 2021-05-09:
and if I try the update again after i get 
APT upgrade                                                                  │
│  - Command: apt-get -qq upgrade                                              │
│  - Exit code: 100                                                            │
│  - DietPi version: v7.0.2 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 |     │
│ DISTRO: 5                                                                    │
│  - Image creator: DietPi Core Team                                           │
│  - Pre-image: Raspberry Pi OS 64-bit                                         │
│  - Error log:                                                                │
│ E: Packages were downgraded and -y was used without --allow-downgrades.      │
│                                                                              │
│     Retry          : Re-run the last command that failed

Joulinar, 2021-05-09:
on the RPi4 you should be able to perform `dietpi-update` now.

Joulinar, 2021-05-09:
I guess running `apt update && apt upgrade` will show same packages `cmake cmake-data` again?

MichaIng, 2021-05-09:
You see the same error again after successfully confirming the "downgrade"? I wonder why it is shown as this at all as both buster-backports and the RPi repo ship that exact same cmake version:
- https://deb.debian.org/debian/pool/main/c/cmake/
- https://archive.raspberrypi.org/debian/pool/main/c/cmake/

Joulinar, 2021-05-09:
could this be an issue in the Debian package? Workaround might be to purge both packages for the time of the update?

MichaIng, 2021-05-09:
> Workaround might be to purge both packages for the time of the update?

👍 `apt purge --autoremove cmake cmake-data`

I'll check it with the two repos and cmake. It's a bit strange when having the packages from repo A installed and the system sees the ones from repo B as downgrade, why would it try to "upgrade" them in the first place? This should only be the case when the upgrade of another package requires the downgrade. Even if you have a newer package manually installed, which is only present as lower version in a repo, apt-get upgrade will simply ignore it until the repo ships a newer version.

MichaIng, 2021-05-09:
Bug in RPi repo confirmed. The package is pulled from the RPi repo and then it sees itself as upgrade which requires a downgrade confirmation 😕. Not sure how this is possible:
```
Get:1 https://archive.raspberrypi.org/debian buster/main arm64 cmake-data all 3.16.3-3~bpo10+1 [1628 kB]
Get:2 https://deb.debian.org/debian buster/main arm64 libicu63 arm64 63.1-6+deb10u1 [8151 kB]
Get:3 https://archive.raspberrypi.org/debian buster/main arm64 cmake arm64 3.16.3-3~bpo10+1 [3016 kB]
Get:4 https://deb.debian.org/debian buster/main arm64 libxml2 arm64 2.9.4+dfsg1-7+deb10u1 [625 kB]
Get:5 https://deb.debian.org/debian buster/main arm64 libarchive13 arm64 3.3.3-4+deb10u1 [291 kB]
Get:6 https://deb.debian.org/debian buster/main arm64 libjsoncpp1 arm64 1.7.4-3 [71.0 kB]
Get:7 https://deb.debian.org/debian buster/main arm64 librhash0 arm64 1.3.8-1 [121 kB]
Get:8 https://deb.debian.org/debian buster/main arm64 libuv1 arm64 1.24.1-1 [105 kB]
...
# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be DOWNGRADED:
  cmake cmake-data
0 upgraded, 0 newly installed, 2 downgraded, 0 to remove and 0 not upgraded.
Need to get 4644 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://archive.raspberrypi.org/debian buster/main arm64 cmake-data all 3.16.3-3~bpo10+1 [1628 kB]
Get:2 https://archive.raspberrypi.org/debian buster/main arm64 cmake arm64 3.16.3-3~bpo10+1 [3016 kB]
Fetched 4644 kB in 4s (1215 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 16047 files and directories currently installed.)
Preparing to unpack .../cmake-data_3.16.3-3~bpo10+1_all.deb ...
Unpacking cmake-data (3.16.3-3~bpo10+1) over (3.16.3-3~bpo10+1) ...
Preparing to unpack .../cmake_3.16.3-3~bpo10+1_arm64.deb ...
Unpacking cmake (3.16.3-3~bpo10+1) over (3.16.3-3~bpo10+1) ...
Setting up cmake-data (3.16.3-3~bpo10+1) ...
Setting up cmake (3.16.3-3~bpo10+1) ...
# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be DOWNGRADED:
  cmake cmake-data
0 upgraded, 0 newly installed, 2 downgraded, 0 to remove and 0 not upgraded.
Need to get 4644 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://archive.raspberrypi.org/debian buster/main arm64 cmake-data all 3.16.3-3~bpo10+1 [1628 kB]
Get:2 https://archive.raspberrypi.org/debian buster/main arm64 cmake arm64 3.16.3-3~bpo10+1 [3016 kB]
Fetched 4644 kB in 4s (1260 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 16047 files and directories currently installed.)
Preparing to unpack .../cmake-data_3.16.3-3~bpo10+1_all.deb ...
Unpacking cmake-data (3.16.3-3~bpo10+1) over (3.16.3-3~bpo10+1) ...
Preparing to unpack .../cmake_3.16.3-3~bpo10+1_arm64.deb ...
Unpacking cmake (3.16.3-3~bpo10+1) over (3.16.3-3~bpo10+1) ...
Setting up cmake-data (3.16.3-3~bpo10+1) ...
Setting up cmake (3.16.3-3~bpo10+1) ...
```

MichaIng, 2021-05-09:
Reported: https://github.com/RPi-Distro/repo/issues/245

deckspace, 2021-05-09:
thanks for looking into it and reporting the apt bug!

Joulinar, 2021-05-09:
As stated above, as a workaround you could remove the package in question to be able to continue the update

`apt purge --autoremove cmake cmake-data`

Phil1988, 2021-05-13:
Does this also apply to me, as an x86 VM user?

I am getting this:
[image]

and before I destroy it, I though aking first would be better :)

Joulinar, 2021-05-13:
This is a complete different issues. Just have a look to https://github.com/MichaIng/DietPi/issues/4219

Phil1988, 2021-05-13:
Thanks for the fast response @Joulinar 👍 

So I will just copy&paste the solution and hope for the best :D

Joulinar, 2021-05-13:
well, it was working for other users as well. usually `hope` is not needed but it doesn't hurt 😉

Phil1988, 2021-05-13:
Well... I guess I have to open a new issue then because NC isnt availible any more
[image]

:(

Joulinar, 2021-05-13:
yes pls.

deckspace, 2021-05-19:
Thanks all, just ran

apt purge --autoremove cmake cmake-data
then dietpi-update but am getting the same error..

#### Details:
- Date           | Wed May 19 10:36:09 BST 2021
- Bug report     | af1a55c7-0884-4c0d-b574-057037c22ccf
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | Linux berry3diet 5.10.17-v8+ #1414 SMP PREEMPT Fri Apr 30 13:23:25 BST 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | dpkg --remove-architecture armhf
- Exit code      | 2
- Software title | DietPi-Patch
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
dpkg: error: cannot remove architecture 'armhf' currently in use by the database
```
please advise

Joulinar, 2021-05-19:
Can you try running `apt update && apt upgrade` and post the full output

C0D3-M4513R, 2021-05-19:
@deckspace  ``` [ INFO ] DietPi-Update | APT upgrade, please wait...
E: Packages were downgraded and -y was used without --allow-downgrades.```

Is not the same error as:

```dpkg: error: cannot remove architecture 'armhf' currently in use by the database```

MichaIng, 2021-05-19:
Yes there are two different issues mixed here:
- Original issue @deckspace, please run: `dpkg --get-selections '*:armhf' | mawk '{print $1}` to list all installed armhf packages. Fixed our side with: https://github.com/MichaIng/DietPi/commit/af3270a804fede0b75644a2985bdebd517e50c27
- Second unrelated issue with the cmake packages: `apt purge --autoremove cmake cmake-data` to remove them for now. Reported and waiting for investigation: https://github.com/RPi-Distro/repo/issues/245

deckspace, 2021-05-19:
sorry for confusion. This from the Pi3B+ install of Dietpi

dpkg --get-selections '*:armhf' | mawk '{print $1} doesn't seem to return anything but 
>

so I ran 

root@berry3diet:~# dpkg --get-selections '*:armhf' and got this response

chromium-browser:armhf				install
chromium-codecs-ffmpeg-extra:armhf		install
gcc-8-base:armhf				install
glib-networking:armhf				install
libasound2:armhf				install
libatk-bridge2.0-0:armhf			install
libatk1.0-0:armhf				install
libatomic1:armhf				install
libatspi2.0-0:armhf				install
libavahi-client3:armhf				install
libavahi-common-data:armhf			install
libavahi-common3:armhf				install
libblkid1:armhf					install
libbsd0:armhf					install
libc6:armhf					install
libcairo-gobject2:armhf				install
libcairo2:armhf					install
libcolord2:armhf				install
libcom-err2:armhf				install
libcups2:armhf					install
libdatrie1:armhf				install
libdbus-1-3:armhf				install
libdrm-amdgpu1:armhf				install
libdrm-nouveau2:armhf				install
libdrm-radeon1:armhf				install
libdrm2:armhf					install
libedit2:armhf					install
libelf1:armhf					install
libepoxy0:armhf					install
libexpat1:armhf					install
libffi6:armhf					install
libfontconfig1:armhf				install
libfreetype6:armhf				install
libfribidi0:armhf				install
libgbm1:armhf					install
libgcc1:armhf					install
libgcrypt20:armhf				install
libgdk-pixbuf2.0-0:armhf			install
libgl1-mesa-dri:armhf				install
libglapi-mesa:armhf				install
libglib2.0-0:armhf				install
libgmp10:armhf					install
libgnutls30:armhf				install
libgpg-error0:armhf				install
libgraphite2-3:armhf				install
libgssapi-krb5-2:armhf				install
libgtk-3-0:armhf				install
libharfbuzz0b:armhf				install
libhogweed4:armhf				install
libicu63:armhf					install
libidn2-0:armhf					install
libjbig0:armhf					install
libjpeg62-turbo:armhf				install
libjson-glib-1.0-0:armhf			install
libk5crypto3:armhf				install
libkeyutils1:armhf				install
libkrb5-3:armhf					install
libkrb5support0:armhf				install
liblcms2-2:armhf				install
libllvm9:armhf					install
liblz4-1:armhf					install
liblzma5:armhf					install
libmount1:armhf					install
libnettle6:armhf				install
libnspr4:armhf					install
libnss3:armhf					install
libp11-kit0:armhf				install
libpango-1.0-0:armhf				install
libpangocairo-1.0-0:armhf			install
libpangoft2-1.0-0:armhf				install
libpcre3:armhf					install
libpixman-1-0:armhf				install
libpng16-16:armhf				install
libproxy1v5:armhf				install
libpsl5:armhf					install
libraspberrypi0:armhf				install
librest-0.7-0:armhf				install
libselinux1:armhf				install
libsensors5:armhf				install
libsoup-gnome2.4-1:armhf			install
libsoup2.4-1:armhf				install
libsqlite3-0:armhf				install
libssl1.1:armhf					install
libstdc++6:armhf				install
libsystemd0:armhf				install
libtasn1-6:armhf				install
libthai0:armhf					install
libtiff5:armhf					install
libtinfo6:armhf					install
libudev1:armhf					install
libunistring2:armhf				install
libuuid1:armhf					install
libwayland-client0:armhf			install
libwayland-cursor0:armhf			install
libwayland-egl1:armhf				install
libwayland-server0:armhf			install
libwebp6:armhf					install
libx11-6:armhf					install
libxau6:armhf					install
libxcb-render0:armhf				install
libxcb-shm0:armhf				install
libxcb1:armhf					install
libxcomposite1:armhf				install
libxcursor1:armhf				install
libxdamage1:armhf				install
libxdmcp6:armhf					install
libxext6:armhf					install
libxfixes3:armhf				install
libxi6:armhf					install
libxinerama1:armhf				install
libxkbcommon0:armhf				install
libxml2:armhf					install
libxrandr2:armhf				install
libxrender1:armhf				install
libxtst6:armhf					install
libzstd1:armhf					install
raspberrypi-bootloader:armhf			install
zlib1g:armhf					install

deckspace, 2021-05-19:
just to confirm success with the Pi4 instance of DietPi that by running apt purge --autoremove cmake cmake-data then the dietpi-update it's now running 7.12 fine thanks.

Joulinar, 2021-05-19:
Uhh on the RPi3 you installed `chromium-browser:armhf` package which is not available on 64bit. Still a missing point by RPi Foundation. Anyway this pulled a hell of dependencies. Would recommend to remove the package and all dependencies. On 64bt you would need to use pure `chromium` package.

MichaIng, 2021-05-19:
> dpkg --get-selections '*:armhf' | mawk '{print $1} doesn't seem to return anything

Whoops, there was a trailing apostrophe `'` missing. However, output would have been the same, just the without the "installed".

Okay you have a lot of armhf packages installed, likely because of the `chromium-browser` that is only available as 32-bit package, while on 64-bit images one would need to use the `chromium` package instead.

**EDIT: As of @Joulinar's post: Does that Chromium browser actually work fine? If so, you don't strictly need to switch to the 64-but package. You can also leave the armhf architecture in place by selecting "Change command" from the error prompt and enter "true" or ":".**

deckspace, 2021-05-22:
OK I have removed chrome gui which as you mention doesn't yet work for 64 bit  dietpi, t's not required in this case anyway. hmm now it's not booting.. baa looks like a reinstall is on the cards.. will get back to you once I have time to rerun the process/

Joulinar, 2021-05-22:
> doesn't yet work for 64 bit dietpi

just to avoid confusion, it's not 64 bit DietPi. It's Raspberry OS 64bit. DietPi is just some scripting on top.

MichaIng, 2021-10-17:
As Debian backports ship cmake v3.18.4 now, the issue is solved.

closed 2021-10-17 by MichaIng