MichaIng/DietPi #4359: Upgrade to Latest Version Fails On Raspberry Pi Zero

breadman100, 2021-05-08:
#### Details:
- Date           | Sat May  8 11:53:17 EDT 2021
- Bug report     | 6e914ae0-eb24-4dc7-a5cd-2069137f8169
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux DietPi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:
Update doesn't complete
- ...
#### Actual behaviour:
Fails during upgrade process, below error.
- ...
#### Extra details:
- ...
#### Additional logs:
```
(Reading database ... 87097 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1.20210430-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20210430-1) over (1.20210303-1) ...
dpkg: error processing archive /var/cache/apt/archives/raspberrypi-kernel-headers_1.20210430-1_armhf.deb (--unpack):
 unable to securely remove '/usr/src/linux-headers-5.10.17+/include/acpi/platform/acgcc.h.dpkg-new': Structure needs cleaning
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/raspberrypi-kernel-headers_1.20210430-1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

ravenclaw900, 2021-05-08:
Hello,
That error can be indicative of file system corruption.  Could you run a check:
```
> /forcefsck
reboot
# after reboot
journalctl -t systemd-fsck
```

Joulinar, 2021-05-29:
I'm going to close this one as we didn't get any feedback. Feel free to reopen if needed.

closed 2021-05-29 by Joulinar

breadman100, 2021-05-29:
Thanks I wasnt able to connect via console, I ended up replacing microsd with the new image.Sent from my Verizon, Samsung Galaxy smartphone