MichaIng/DietPi #4361: Install unbound failed

tiesto4ever, 2021-05-09:
#### Details:
- Date           | Sun May  9 13:11:36 BST 2021
- DietPi version | v7.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPi-1913 5.10.17-v8+ #1414 SMP PREEMPT Fri Apr 30 13:23:25 BST 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | systemctl enable --now unbound
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. Install Unbound using the Dietpi-software
2. ...
#### Expected behaviour:

- unbound should be installed correctly
#### Actual behaviour:

- It failed 
[image]

#### Extra details:

- ...
#### Additional logs:
```
Synchronizing state of unbound.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable unbound
Job for unbound.service failed because the control process exited with error code.
See "systemctl status unbound.service" and "journalctl -xe" for details.
```

tiesto4ever, 2021-05-09:
[image]

MichaIng, 2021-05-09:
Something already listens on port 53. Can run and paste the output of:
```
ss -tulpn | grep ':53'
```

Joulinar, 2021-05-09:
Maybe Pihole is/was installed in parallel

MichaIng, 2021-05-09:
/etc/unbound/unbound.conf is supported to be missing, as the package by default uses /etc/unbound/unbound.conf.d/*.conf files. I wonder why this warning is printed at all, even as part of manual APT package install and on each service start/restart, or is it omitted when a conf.d config exists?

If Pi-hole was installed in parallel or before, the port would be changed automatically to 5335. But probably it was installed manually (outside of dietpi-software), then indeed this is not the case.

Joulinar, 2021-05-29:
I'm going to close this one as we didn't get any feedback. Feel free to reopen if needed.

closed 2021-05-29 by Joulinar