MichaIng/DietPi #4365: Linux Unified Key Setup LuksFS

cocoflan, 2021-05-10:
Maybe an idea to use on external drives that are not used as a swap disk or userdata location. 

What the ...?
Cryptsetup is a utility used to conveniently set up disk encryption based
on the DMCrypt kernel module.
These include plain dm-crypt volumes, LUKS volumes, loop-AES,
TrueCrypt (including VeraCrypt extension) and BitLocker formats.
The project also includes a veritysetup utility used to conveniently setup
DMVerity block integrity checking kernel module
and integritysetup to setup
DMIntegrity block integrity kernel module.

LUKS Design
LUKS is the standard for Linux hard disk encryption. By providing a standard on-disk-format, it does not
only facilitate compatibility among distributions, but also provides secure management of multiple user passwords.
LUKS stores all necessary setup information in the partition header, enabling to transport or migrate data seamlessly.

https://gitlab.com/cryptsetup/cryptsetup/blob/master/README.md

MichaIng, 2021-05-10:
Many thanks for your request. It's a bit similar to #3377, although it was specifically about allowing to move dietpi_userdata to such an encrypted drive. The first step however would be to add support to dietpi-drive_manager to correctly detect and create such filesystems.

As many relevant information about how to implement this has been gathered there, I renamed the other other to be more generic and mark this issue as closed.

closed 2021-05-10 by MichaIng

cocoflan, 2021-05-10:
Ok good