MichaIng/DietPi #4367: Pi-Hole fails update

Agricolafb, 2021-05-11:
#### Details:
- Date           | Tue May 11 18:53:06 BST 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l) (ID=11)
- Kernel version | Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-defaults/php-common_81+0~20210223.34+debian9~1.gbpf52eb0_all.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Joulinar, 2021-05-11:
Hi,

the issue is not with Pihole. It's more the PGP key of your PHP repository expired https://github.com/MichaIng/DietPi/issues/4219

Solution would be

```
apt-key del '95BD4743'
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
```

MichaIng, 2021-05-11:
Many thanks for your report.

You mean DietPi, not Pi-hole, don't you?

... lol while writing, thanks @Joulinar, that will be it.

Joulinar, 2021-05-29:
I'm going to close this one as we didn't get any feedback. Feel free to reopen if needed.

closed 2021-05-29 by MichaIng