MichaIng/DietPi #4368: Unable to continue, DietPi-Drive_Manager will now terminate

Loedgreem, 2021-05-12:
---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Wed May 12 16:37:28 CEST 2021
- DietPi version | v7.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v8+ #1414 SMP PREEMPT Fri Apr 30 13:23:25 BST 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | mkfs.ext4 -F -m 0 /dev/sdc1
- Exit code      | 1
- Software title | DietPi-Drive_Manager
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
mke2fs 1.44.5 (15-Dec-2018)
The file /dev/sdc1 does not exist and no size was specified.
```
---------------------------------------------------------------------
[FAILED] DietPi-Drive_Manager | Unable to continue, DietPi-Drive_Manager will now terminate.

Joulinar, 2021-05-12:
Hi,

Short question. Do you use an external disk that is extremely powered? Or does it get power from the RPi only?

As well can you check for kernel error messages
`dmesg -l err,crit,alert,emerg
`

Loedgreem, 2021-05-12:
the disk I was trying to cennect was powered by Pi only. But after a reboot, I could successfully add the additional disk and mount it. No further issues. Hardware setup was persistent, no change between both tests.

Joulinar, 2021-05-12:
Issue could be with power. Usually @MichaIng could explain it more in detail 😃 

In general we recommend to use external disk with an own external power supply.

Loedgreem, 2021-05-12:
now its working fine - but thanks anyway

closed 2021-05-12 by Loedgreem