MichaIng/DietPi #4369: Bitwarden_RS failure

yoinx, 2021-05-12:
#### Details:
- Date           | Wed May 12 16:06:41 EDT 2021
- DietPi version | v7.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux rpi4 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | mv bitwarden_rs-1.21.0 /opt/bitwarden_rs
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

Attempted to select and install Bitwarden_RS via dietpi_software
#### Expected behaviour:
Install should succeed
#### Actual behaviour:
Install Fails with the message to report the failure.
#### Extra details:
In addition to this failure, or perhaps related to... BitWarden_RS has rebranded itself as VaultWarden (see: https://github.com/dani-garcia/vaultwarden/discussions/1642 ) which may be causing some trouble. I would imagine that the naming should be changed within the dietpi-software list as well.
#### Additional logs:
```
mv: cannot stat 'bitwarden_rs-1.21.0': No such file or directory
```
---------------------------------------------------------------------

Joulinar, 2021-05-12:
Hi,

this is already known issue https://github.com/MichaIng/DietPi/issues/4325
And will be fixed on next version https://github.com/MichaIng/DietPi/pull/4334

yoinx, 2021-05-12:
Sorry about that, guess I should've read through them before posting. Thanks!

closed 2021-05-12 by yoinx