MichaIng/DietPi #4370: Kernel not found on RPi Model 2B v1.2

menthenchocolat, 2021-05-13:
### bug report/issue

The 64bit version of DietPi does not start on my Raspberry Pi Model 2B v1.2. The status LEDs of the RPi show a repeating sequence of 7 short blinks. According to [documentation here](https://www.raspberrypi.org/documentation/configuration/led_blink_warnings.md) this indicates that the kernel was not found.

On the same hardware (RPi, Power Supply, SDcard) the 32bit version of DietPi does start without any problems 

#### Required Information
- DietPi version | DietPi_RPi-ARMv8-Buster
- Kernel version | standard
- SBC model | RPi 2B v1.2
- Power supply used | 5V 2.4A Varta power converter 100-240V -> USB
- SDcard used | Intenso MircoSD HC Class 10 - 8GB

#### Additional Information (if applicable)

#### Steps to reproduce

write new image of DietPi_RPi-ARMv8-Buster onto SDcard, put in RPi and power up

#### Expected behaviour

DietPi should be starting up in 64bit

#### Actual behaviour

status LED of RPi indicates error "Kernel image not found" (repeating sequence of 7 short flashes)

#### Extra details

I thought that RPI 2B v1.2 is able to process 64bit ARMv8. Maybe it is not?

Joulinar, 2021-05-13:
Can you post following plus

```
cat /sys/firmware/devicetree/base/model
cat /proc/cpuinfo
```

Btw what is the reason to run 64bit? There no real benefit compare to the 32bit version.

MichaIng, 2021-05-13:
> I thought that RPI 2B v1.2 is able to process 64bit ARMv8. Maybe it is not?

It should be able to, the CPU is ARMv8, the same SoC that is used in RPi 3: https://en.wikipedia.org/wiki/Raspberry_Pi#Specifications

What you see is not an exec format error but probably bug in the firmware/bootloader that it does not accept the arm_64bit switch respectively does not accept the 64bit kernel for RPi 2B v1.2.

The `kernel8.img` in the boot FAT partition is the kernel image. The kernel package even ships a dedicated `bcm2710-rpi-2-b.dtb`, the device tree for the RPi 2B v1.2. Please verify that these files are present in your case and yes it would be good to re-assure that it's PCB v1.2 instead of v1.1. In case we need to forward to RPi firmware devs.

Joulinar, 2021-05-29:
Not sure if it is related but sounds similar https://dietpi.com/phpbb/viewtopic.php?f=11&t=9070

MichaIng, 2021-05-29:
Raspberry Pi Zero W does not support 64-bit as it's an ARMv6 SoC. The RPi 2 v1.2 however is. The RPi 2 v1.1 is NOT, so it must be taken care that the correct revision is present and correctly detected.

Joulinar, 2021-05-29:
How could I have missed this 🤣

MichaIng, 2021-06-02:
As of missing response, I'm marking this as closed. Feel free to reopen if issue persists and the RPi model is assured to be ARMv8-compatible, hence RPi 2 v1.2.

closed 2021-06-02 by MichaIng

menthenchocolat, 2021-07-23:
I feel quite stupid now: It was a B+ v1.2...

I thought I had bought a second-hand 2B v1.2 but had got a B+ v1.2. Only saw the "v1.2" on the board and thought "Yes, that's my baby!" and encased it. 
m)

That is, why it never started & never could start in 64bit... m) m) m)

Sorry!!

MichaIng, 2021-07-23:
Ah that explains it of course 😄. I remember a very similar case and hope you can run your application on an RPi 1 with 32-bit image as well 🙂.